realsenseai/realsense-ros #1159: process has died [pid 5825, exit code -11,

jediofgever, 2020-04-09:
some info;
```
* kernel ;5.3.0-46-generic
* ubuntu 18.04
* d435 i
*  RealSense ROS v2.2.13
 * Running with LibRealSense v2.33.1
```

I start my camera node  and log is ; 

```
atas@atas-MS-7B48:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/atas/.ros/log/48d02b2c-7a47-11ea-8b35-309c239ac6ca/roslaunch-atas-MS-7B48-5797.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://atas-MS-7B48:38585/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: 1.2
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 540
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 960
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 848
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 848
[... 113 lines cut ...]
 09/04 17:48:38,682 ERROR [140127401670400] (backend-v4l2.h:196) xioctl(VIDIOC_QBUF) guard failed for fd 17
 09/04 17:48:38,682 WARNING [140127401670400] (backend-v4l2.h:206) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 17
[camera/realsense2_camera_manager-2] process has died [pid 5825, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/atas/.ros/log/48d02b2c-7a47-11ea-8b35-309c239ac6ca/camera-realsense2_camera_manager-2.log].
log file: /home/atas/.ros/log/48d02b2c-7a47-11ea-8b35-309c239ac6ca/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/atas/.ros/log/48d02b2c-7a47-11ea-8b35-309c239ac6ca/camera-realsense2_camera-3*.log
^C[rosout-1] killing on exit
^C^C[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

The error I get is ; 

`process has died `

in the log file it points; 
```
[0m[ INFO] [1586425717.667930769]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:[0m
[0m[ INFO] [1586425717.669832911]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...[0m
[0m[ INFO] [1586425717.690424667]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.[0m
[0m[ INFO] [1586425729.798558019]: Bond broken, exiting[0m
```

MartyG-RealSense, 2020-04-09:
A good rule of thumb with error debugging is to identify the first line of a list of errors.  The first error that occurs may set off a "cascade" chain reaction of errors in systems that are affected by the first failure.  If the first error is resolved then the subsequent errors may disappear.

In your log, the first error is:

**WARNING [140127390017280] (ds5-timestamp.cpp:76) UVC metadata payloads not available.** 

@doronhi the RealSense ROS wrapper developer offers advice on this error in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/932#issuecomment-538720015

jediofgever, 2020-04-09:
That’s not an error , it’s a warning

MartyG-RealSense, 2020-04-09:
It looks like the subsequent backend errors are generated from **backend-v4l2.h**

https://github.com/IntelRealSense/librealsense/blob/master/src/linux/backend-v4l2.h#L183

mikeNike9, 2020-04-09:
No this is an error and is the same error that We have been trying to get  solved for sometime now. 
https://github.com/IntelRealSense/realsense-ros/issues/1142

@poorboy44 @doronhi 

We havent been able to run anything for more than a minute until it crashes. 
Can we combine these two threads and figure this out all.
https://github.com/IntelRealSense/realsense-ros/issues/1142

mikeNike9, 2020-04-09:
I get same log here is my log

[0m[ INFO] [1586448339.349322896]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:[0m
[0m[ INFO] [1586448339.352294633]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...[0m
[0m[ INFO] [1586448339.373216767]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.[0m
[0m[ INFO] [1586448460.862017193]: Bond broken, exiting[0m

mikeNike9, 2020-04-09:
roslaunch realsense2_camera rs_camera.launch
... logging to /home/mike/.ros/log/1921acf8-7a7d-11ea-99ff-a0cec81fb9c9/roslaunch-mike-Blade-6245.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mike-Blade:40965/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: True
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 720
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1280
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 720
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 1280
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 720
 * /camera/realsense2_camera/infra_width: 1280
 * /camera/realsense2_camera/initial_reset: False
[... 59 lines cut ...]
[ INFO] [1586448831.366091801]: Setting Dynamic reconfig parameters.
[ INFO] [1586448831.474921894]: Done Setting Dynamic reconfig parameters.
[ INFO] [1586448831.488168224]: depth stream is enabled - width: 1280, height: 720, fps: 30, Format: Z16
[ INFO] [1586448831.498103813]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1586448831.498186329]: setupPublishers...
[ INFO] [1586448831.500678857]: Expected frequency for depth = 30.00000
[ INFO] [1586448831.504764430]: Expected frequency for color = 30.00000
[ INFO] [1586448831.507119087]: setupStreams...
[ INFO] [1586448831.508005893]: insert Depth to Stereo Module
[ INFO] [1586448831.508063446]: insert Color to RGB Camera
[ INFO] [1586448831.533108680]: SELECTED BASE:Depth, 0
[ INFO] [1586448831.549619070]: RealSense Node Is Up!
 09/04 11:13:51,940 ERROR [140703032538880] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 09/04 11:13:52,229 ERROR [140703032538880] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 09/04 11:13:52,430 ERROR [140703032538880] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 09/04 11:13:52,581 ERROR [140703032538880] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
[camera/realsense2_camera_manager-2] process has died [pid 6273, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/mike/.ros/log/1921acf8-7a7d-11ea-99ff-a0cec81fb9c9/camera-realsense2_camera_manager-2.log].
log file: /home/mike/.ros/log/1921acf8-7a7d-11ea-99ff-a0cec81fb9c9/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/mike/.ros/log/1921acf8-7a7d-11ea-99ff-a0cec81fb9c9/camera-realsense2_camera-3*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2020-04-09:
@mikeNike9  Thanks a lot for adding your discussion link!  

That discussion is likely to be the best place for yourself and @jediofgever  to get assistance with this problem, as it has the attention of two Intel guys (Doronhi and RealSenseCustomerSupport).  I'm still learning about ROS, so they should be able to provide better advice on the subject of these errors right now.

mikeNike9, 2020-04-09:
Honestly @MartyG-RealSense @doronhi @RealSenseCustomerSupport 

Our company will be using this ros wrapper frequently with our new core products 
However we can help please let us know. We are more than happy to test pre release firmware and development packages to make this more seamless.

MartyG-RealSense, 2020-04-09:
@mikeNike9   Thanks so much for your support for RealSense!  If you have code for the wrapper that you think can be of benefit, a "pull request" can be submitted for review at the link below.

https://github.com/IntelRealSense/realsense-ros/pulls

jediofgever, 2020-04-10:
> No this is an error and is the same error that We have been trying to get solved for sometime now.
> #1142
> 
> @poorboy44 @doronhi
> 
> We havent been able to run anything for more than a minute until it crashes.
> Can we combine these two threads and figure this out all.
> #1142

please look carefully to see what I call an error and a "WARNING".

Back to the issue, yes seems like we are having the same issue. I saw the content of the link you provided. Its a bitter experience with realsense, they dont have a stable version of pair of their lib and ROS wrapper, with each update there is a new issue worse than previous one.

MartyG-RealSense, 2020-04-26:
This case will be closed after 7 days from the time of writing this message if there are no further comments.  Thank you.

closed 2020-04-26 by jediofgever

MartyG-RealSense, 2020-04-26:
@jediofgever   I am sorry that the outcome of this case was not to your satisfaction.  It is a problem though that requires developer attention.  As a member of the support team (not the developer team), I do not have the ability to create a solution for the **get_device_time_ms()** error.  A fix for it is awaiting developer approval for merging into the RealSense SDK.