realsenseai/realsense-ros #1160: D415 disconnects when launching rs_camera.launch

FraDePace, 2020-04-10:
Hello everyone,
I cannot make the Intel D415 camera work with ROS.

When I run:
**roslaunch realsense2_camera rs_camera.launch**

the camera disconnects and I cannot see the RGB/Depth video on rviz.
The weird thing is that, despite my pc has only USB 2.0 ports, when I run realsense-viewer, it works properly.

I'm using Ubuntu 18.04 with ROS melodic.
**Camera info:**
Name: Intel RealSense D415
Firmware Version: 05.12.03.00

This is the output of the above command
[log_info.txt](https://github.com/IntelRealSense/realsense-ros/files/4459814/log_info.txt)

Thank you for your help

gagor15, 2020-04-10:
Hello, check if you have the RealSense dependencies installed:
`sudo apt install librealsense2-dev librealsense2-dkms`

FraDePace, 2020-04-10:
@gagor15 Thanks! I missed this package, problem solved :)

closed 2020-04-10 by FraDePace