realsenseai/realsense-ros #1161: Synthetic Stream Error

haydenhager, 2020-04-10:
Camera Model: D415 x 1 via usb3
Firmware: 05.12.03.00
Operating system: Ubuntu 18.04.1
Kernel: 5.3.0-46-generic
Platform: x86_64 desktop intel i7
Language: ROS/realsense-viewer
SDK: 2.34.0
ROS distro: Melodic 2.2.13

I upgraded all of the debian packages with apt-get do to another issue we were having with v412 backend error. Upgrading fixed that issue but now we are seeing this error from realsense-viewer:

 10/04 11:55:39,853 ERROR [140382513903360] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!

Unfortunately the viewer is not wanting to output the logs to a file so I had to copy this from the 'output' from the viewer. 

This is what we are getting as output from our ros rs_camera.launch:

started roslaunch server http://hayden-hp:35323/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: True
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 720
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1280
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 720
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 1280
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 720
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 1280
 * /camera/realsense2_camera/gyro_fps: 400
[... 152 lines cut ...]
 10/04 10:39:54,314 ERROR [140096747824896] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 10/04 10:39:54,414 WARNING [140096747824896] (types.cpp:49) Out of frame resources!
 10/04 10:39:54,414 ERROR [140096747824896] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 10/04 10:39:54,646 WARNING [140096747824896] (types.cpp:49) Out of frame resources!
 10/04 10:39:54,647 ERROR [140096747824896] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 10/04 10:39:54,749 WARNING [140096747824896] (types.cpp:49) Out of frame resources!
 10/04 10:39:54,750 ERROR [140096747824896] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 10/04 10:39:54,980 WARNING [140096747824896] (types.cpp:49) Out of frame resources!
 10/04 10:39:54,980 ERROR [140096747824896] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 10/04 10:39:55,081 WARNING [140096747824896] (types.cpp:49) Out of frame resources!
 10/04 10:39:55,081 ERROR [140096747824896] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 10/04 10:39:55,314 WARNING [140096747824896] (types.cpp:49) Out of frame resources!
 10/04 10:39:55,314 ERROR [140096747824896] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 10/04 10:39:55,414 WARNING [140096747824896] (types.cpp:49) Out of frame resources!
 10/04 10:39:55,415 ERROR [140096747824896] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 10/04 10:39:55,615 WARNING [140096747824896] (types.cpp:49) Out of frame resources!
 10/04 10:39:55,615 ERROR [140096747824896] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 10/04 10:39:55,747 WARNING [140096747824896] (types.cpp:49) Out of frame resources!
 10/04 10:39:55,747 ERROR [140096747824896] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 10/04 10:39:55,812 WARNING [140096747824896] (types.cpp:49) Out of frame resources!
 10/04 10:39:55,812 ERROR [140096747824896] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!

I am still able to use the camera while it is giving this error but my %CPU when I updated went from around 40% to over 100% for the nodelet when trying to stream the image through roslibjs. On lesser pc's it is crashing the nodelet.  

Any advice on how to mitigate this error?

MartyG-RealSense, 2020-04-11:
I located another currently active case that has similarities to yours in terms of the errors generated.

https://github.com/IntelRealSense/realsense-ros/issues/1110

It is very long, so I will pick out some useful quotes from @doronhi the RealSense ROS wrapper developer about a couple of the main warnings that you have experienced.

"The warning message "interrupt event received" can be ignored and indeed will be removed in future versions".

"The warning message "control_transfer returned error" reflects communication error with the device that is resolved by repeating the attempt. These messages are still under investigation but as long as they appear only now and then, say at the beginning of the run, they can safely be ignored for now".

"If the message "messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11" is continuous, that is appears multiple times per second, then it is not good. It suggest a real communication problem between the code and the device and can have effects".

"The "Out of frame resources!" message and the "Exception...processing callback!" could be related to the "IR stream failure" but more often they are related to lack of system resources - the app can't keep up with the incoming flow of images. For that you can reduce frame rate and resolution.
Building with CUDA support should really help in that case on NVidia machines".

I recommend also reading the linked case from the start of course to see if any of the other advice in that discussion is helpful to your own situation.

MartyG-RealSense, 2020-04-26:
@haydenhager    If you are still experiencing these errors, I would recommend investigating the possibility of using a mains electricity powered USB 3 hub to increase USB port stability if your project allows it (for example, it is not a mobile robot that would be unsuited to being tethered to a wall power socket).

Otherwise, the situation may require programming additions from the RealSense developers to reduce the likelihood of errors such as **control_transfer returned** occurring.  So I will close the case, as the developers are aware of the problem and are likely investigating it.

closed 2020-04-26 by MartyG-RealSense