realsenseai/realsense-ros #1162: Cannot transform data showing"control_transfer returned error, index: 768, error: No data available, number: 61" with D435i

DanLiu0809, 2020-04-11:
Operating System | Ubuntu 16.04 LTS
Kernel | 4.15.0-66-generic
ROS | kinetic 1.12.14

I have installed the librealsense2 packages and realsense-ros. Everything works correctly in realsense-viewer, but when I try to run 'roslaunch realsense2_camera rs_camera.launch', it seems that no data is returned. I assume I have installed an incorrect version of something alone the way. Here is my output below(I don't know how to adjust the format). Any suggestions?

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
[... 61 lines cut ...]
[ INFO] [1586601118.745298568]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1586601118.772367251]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1586601118.772537990]: setupPublishers...
[ INFO] [1586601118.775735545]: Expected frequency for depth = 30.00000
[ INFO] [1586601118.802831300]: Expected frequency for infra1 = 30.00000
[ INFO] [1586601118.827301473]: Expected frequency for infra2 = 30.00000
[ INFO] [1586601118.851213205]: Expected frequency for color = 30.00000
[ INFO] [1586601118.877551719]: setupStreams...
[ INFO] [1586601118.928627349]: insert Depth to Stereo Module
[ INFO] [1586601118.928707448]: insert Color to RGB Camera
[ INFO] [1586601118.928740254]: insert Infrared to Stereo Module
[ INFO] [1586601118.928778424]: insert Infrared to Stereo Module
[ INFO] [1586601118.928816208]: insert Gyro to Motion Module
[ INFO] [1586601118.928851182]: insert Accel to Motion Module
[ INFO] [1586601119.187584675]: SELECTED BASE:Depth, 0
[ INFO] [1586601119.234877541]: RealSense Node Is Up!
 11/04 18:31:59,321 WARNING [140115523401472] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/04 18:31:59,402 WARNING [140115523401472] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/04 18:31:59,484 WARNING [140115523401472] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/04 18:31:59,545 WARNING [140115523401472] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/04 18:31:59,626 WARNING [140115523401472] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/04 18:31:59,807 WARNING [140115523401472] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/04 18:39:51,587 WARNING [140115523401472] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 11/04 18:41:16,374 WARNING [140115523401472] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 11/04 18:41:17,455 WARNING [140115523401472] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2020-04-11:
@doronhi the RealSense ROS wrapper developer has the following advice about the meaning of the **control_transfer returned** errors:

"The warning message "control_transfer returned error" reflects communication error with the device that is resolved by repeating the attempt. These messages are still under investigation but as long as they appear only now and then, say at the beginning of the run, they can safely be ignored for now".

"If the message "messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11" is continuous, that is appears multiple times per second, then it is not good. It suggest a real communication problem between the code and the device and can have effects".

What computer hardware are you using please (PC / Mac / Linux / Raspberry Pi / Jetson etc).

DanLiu0809, 2020-04-11:
@MartyG-RealSense I use Linux 16.04 LTS with intel i7

MartyG-RealSense, 2020-04-11:
The errors are likely not related to a low computer specification the if you have an i7 computer.  At 640x480 resolution, you are not putting a heavy processing load on the computer either.

Your RealSense ROS and Librealsense versions (2.2.13 ROS, 2.33.1 Librealsense) are also correctly matched.

Do you know which firmware version your camera is using please?  Firmware version 5.12.3.0 is recommended for the SDK that you are using.  You can update the camera to the latest firmware by going to the 'More' icon at the top of the RealSense Viewer's side-panel and selecting the menu option 'Install Recommended Firmware'.   You do not need a firmware file to do this, as the latest version of the firmware is bundled inside the SDK.

[image]

DanLiu0809, 2020-04-11:
@MartyG-RealSense I update my firmware just like you said as soon as your response, and now the firmware version is : 05.12.03.00, but it has no change launching rs_camera.launch.
I notice that the output shows "IMU Calibration is not available", and others outputs don't have this information, is that the main problem?

MartyG-RealSense, 2020-04-11:
If the IMU on the D435i has not been calibrated by the user then some programs may display this [-21] error.  My understanding is that it should not be a problem if you are not accessing the IMU and its gyro and accelerometer sensors though.

Are you using the official short cable that was supplied with the camera please or are you using a cable of your own choice?  RealSense needs a high quality cable, especially if the cable is 2 meters or more in length, and performance can degrade or the camera disconnect if an own-choice cable is not of sufficient quality.

DanLiu0809, 2020-04-11:
@MartyG-RealSense I use the official cable, now I'm truly comfused:(

MartyG-RealSense, 2020-04-11:
Don't worry, this is a useful process because each new piece of information and answered question narrows down the possible causes for the problem.  So it is very helpful.  Thanks for your patience!

As the camera works fine for you in the RealSense Viewer, it is likely a problem related to the ROS wrapper.  This would also suggest that you are not experiencing problems with power supply instability on your computer's USB ports, otherwise that would affect the camera's performance in the RealSense Viewer.

When you receive the control_transfer returned errors in the log, do they generate continuously or stop shortly after launch?

DanLiu0809, 2020-04-11:
It will generate continuously and after a few moments, output change into "(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11" and then generate continuously.

MartyG-RealSense, 2020-04-11:
Okay, thanks.   It does sound like a more serious problem than just a short "hiccup" on the connection at launch then.

Are you able to use Rviz to check the camera output?  Despite the log messages, if the camera output appears in Rviz then it should mean that the ROS topics are being published correctly.

DanLiu0809, 2020-04-11:
Camera output can appear in proper topic, I can show you screenshots later.

MartyG-RealSense, 2020-04-11:
Ok, thank you very much.  It is a good sign that you are getting the image topic published successfully.

DanLiu0809, 2020-04-11:
[image]
I use RViz to check my camera output and it seems it has no problem about image topic.

ashirsat, 2020-04-11:
Hey,

I read in the ROS documentation for real sense that you need realsense 2
packages and not realsense packages for the camera series that you have.
Might be something that you would have tried earlier. But if not try it.

Aniket

-- 
Regards,
Aniket Shirsat

Mechanical Graduate
Arizona State University| Ira A. Fulton Schools of Engineering
School for Engineering of Matter Transport and Energy
Cell:(480)371-8843
Portfolio <http://seelio.com/aniketshirsat>
<http://www.linkedin.com/in/shirsataniket>
<https://twitter.com/intent/follow?original_referer=https%3A%2F%2Fabout.twitter.com%2Fresources%2Fbuttons&region=follow_link&screen_name=AniketRShirsat&tw_p=followbutton>

MartyG-RealSense, 2020-04-11:
Thanks for the RViz image!

@doronhi the ROS wrapper developer may be the best person to advise about how to deal with your log messages, based on the facts established about your situation earlier in this discussion.  I have linked him into this discussion.  I hope that the messages are not disruptive to your use of ROS in the meantime.

MartyG-RealSense, 2020-04-11:
@ashirsat   Thanks for your input!  @DanLiu0809 is using realsense2_camera, which is the appropriate ROS package for the D435i camera (part of the 400 Series cameras).

[image]

ashirsat, 2020-04-11:
Oops sorry about that. Missed that part.

-- 
Regards,
Aniket Shirsat

Mechanical Graduate
Arizona State University| Ira A. Fulton Schools of Engineering
School for Engineering of Matter Transport and Energy
Cell:(480)371-8843
Portfolio <http://seelio.com/aniketshirsat>
<http://www.linkedin.com/in/shirsataniket>
<https://twitter.com/intent/follow?original_referer=https%3A%2F%2Fabout.twitter.com%2Fresources%2Fbuttons&region=follow_link&screen_name=AniketRShirsat&tw_p=followbutton>

MartyG-RealSense, 2020-04-11:
@ashirsat   No problem at all.  It's great when community members are willing to help each other out!

MartyG-RealSense, 2020-04-26:
@DanLiu0809  Were you able to resolve your problem, please?  This case will be closed after 7 days from the time of writing this message if there are no further comments.  Thank you.

17602357904, 2020-04-26:
> @ashirsat   Thanks for your input!  @DanLiu0809 is using realsense2_camera, which is the appropriate ROS package for the D435i camera (part of the 400 Series cameras).

@MartyG-RealSense Can I run it through D435? I have the same problem as@DanLiu0809

MartyG-RealSense, 2020-04-26:
@17602357904   Using D435 with realsense2_camera is fine.

17602357904, 2020-04-29:
@DanLiu0809 Did you solve the problem? Can you give your way？

DanLiu0809, 2020-05-02:
@17602357904 I actually cannot solve this problem still, but I try avoiding this problem by adding exacting topic to my rviz in terminal. It seems this problem is because the data from D435i  cannot find a proper port to output[my partial opinion].

DanLiu0809, 2020-05-02:
Now I'm trying to change the environment, like I upgrade my OS to Ubuntu 18.04 and also ROS from Kinetic to Melodic, to see whether it has some different changes.

17602357904, 2020-05-02:
@DanLiu0809 Thanks for your answer.How to add strict theme to rviz in terminal to avoid this problem?  If you have made any progress, please notify me. Thank you very much!

doronhi, 2020-05-02:
I am trying to understand your issue @DanLiu0809, although it seems obvious to other people... You mention that using RViz you can see the topics fine so, is the issue the warnings that appear continuously? What did you mean by "adding **exacting** topic to my rviz in terminal"? What exactly is the problem you try to avoid?
Due to the whole covid-19 issue it will probably take me a long while to check on this issue again. My sincere apologies.

DanLiu0809, 2020-05-02:
@doronhi That's OK and thanks for your comment! I use rtabmap to do the SLAM things, and when I start the camera node, I get these warning notice continuously; if I don't stop the terminal, it will repeat this warning. But when I open rviz to check the camera, it can get the pictures rightfully with RGBD and depth graphs; when I launch rtabmap launchfile then, rtabmap can work properly and camera-node has the same warning repeating. At first I think maybe several essential-libs are not installed, but since rviz and rtabmap can both work properly, I'm truly comfusing. If you have any suggestions please contact me. Thank you.

doronhi, 2020-05-05:
I tested the latest version with librealsense [PR#6150](https://github.com/IntelRealSense/librealsense/pull/6150) on my VMWare machine (Ubuntu 18.04). I counted 1277 occurrences of "control_transfer returned error..." in 19 hours run. That's an average of around 1 warning per minute. Is this the approximate rate you also see?
I didn't see any operative problem caused by it.
That's been said, I'll look into it and see how to handle or remove this warning which is annoying in the best scenario and troubling otherwise. Until that time, if the rate the warnings appear for you is something similar to my testings, I believe you can carry on safely with your work.

DanLiu0809, 2020-05-10:
OK it seems that the appearing rate of this warnings is the same of yours, and it has not influenced my work recently. I'll close this issue and hope you can fix this warning notice soon. Thanks again!

closed 2020-05-10 by DanLiu0809

sadransh, 2020-06-04:
I experienced similar issue on d435 ( like what @doronhi told) on firmware 05.09.02.00. 
On d435i(with firmware 05.12.02.100)  this happened and no output e.g. on /d400/color/image_raw : 
```
 04/06 18:14:02,670 WARNING [547161854336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/06 18:14:03,145 WARNING [547161854336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/06 18:14:04,834 WARNING [546903667072] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312445.016228104]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:06,894 WARNING [546903667072] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312447.056337641]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:08,946 WARNING [546937237888] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312449.091244113]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:10,982 WARNING [547203817856] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312451.123730312]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:13,034 WARNING [546937237888] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312453.155249039]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:15,086 WARNING [547203817856] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312455.192474103]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:17,138 WARNING [546903667072] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312457.227373933]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:19,190 WARNING [547203817856] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312459.262550902]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:21,242 WARNING [546937237888] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312461.295108942]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:23,294 WARNING [546903667072] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312463.340762645]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:25,342 WARNING [546937237888] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312465.378150815]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:27,406 WARNING [546903667072] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1591312467.418816322]: Hardware Notification:USB CAM overflow,1.59131e+12,Error,Hardware Error
 04/06 18:14:29,458 WARNING [547203817856] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 13, 

```

MartyG-RealSense, 2020-06-05:
@sadransh   The **USB CAM overflow** error suggests that there is a greater volume of data being processed by the USB system than it can handle.  This may be related to the particular computer hardware that you are using, and the same program may not produce the errors on a different computer.
You may be able to reduce the strain put on the USB by adding to your roslaunch statement the instruction **enable_infra2:=false** to disable the Infra2 channel.  

It saves on USB transportation and kernel work, according to Doronhi the RealSense ROS wrapper developer.  Doronhi adds "It will have no effect on the depth quality.  It only disables the infra2 images' transmission via the USB port.  They are still taken and the depth is calculated using them inside the device.".

sadransh, 2020-06-06:
@MartyG-RealSense Thanks for the answer, yes I am using Jetson TX2, I have already turned off both infras. The point is just switching from d435i(with firmware 05.12.02.100) to  d435 (firmware 05.09.02.00) resolves the USB CAM overflow without any other changes( on d435 just some control_transfer error every minute). And I think it's kind of weird. 

other information for my device : 
kernel version 4.9.1xx-tegra 
ubuntu 18.04
Jetson TX2 on Jetpack 4.3 
liberalsense 2.32.1 ( installed using apt-get)
relasense-ros cloned from development branch a few days ago.

MartyG-RealSense, 2020-06-06:
@sadransh   Doronhi the RealSense ROS developer has said that if you are only receiving a **control_transfer returned** warning at a rate of about one a minute then it is safe to work with the ROS wrapper, and he has been looking at how to handle or remove the warning.

https://github.com/IntelRealSense/realsense-ros/issues/1162#issuecomment-624129592

Regarding the USB CAM Overflow errors: a couple of RealSense users had continuously-generating errors disappear this week by upgrading to the new **5.12.5** firmware.   It may be worth trying that with your D435i.   Then you have still got the D435 and its 05.09.02.00 firmware as backup if the firmware upgrade does not make a difference.

Eslamrady42, 2020-06-21:
I have exact the same error !!

victorll998, 2021-05-03:
@MartyG-RealSense  I have encounter this issue with an average rate of 3 sec.

I have two d455 and t265 running on Intel NUC (i7). I tested with the cables in box and a powered USB hub. When only one d455 and t265 connected, the control_transfer returned error has been refrained after a few second when nodes initialize. But when I tried to connect two d455 and t265, one of the d455 would retain the control transfer error at a high frequency while the other d455 is fine. 

camera 1 continue to give me the error 
```
SUMMARY
========

PARAMETERS
 * /cam_1/realsense2_camera/accel_fps: -1
 * /cam_1/realsense2_camera/accel_frame_id: cam_1_accel_frame
 * /cam_1/realsense2_camera/accel_optical_frame_id: cam_1_accel_optic...
 * /cam_1/realsense2_camera/align_depth: False
 * /cam_1/realsense2_camera/aligned_depth_to_color_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_fisheye_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_infra1_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_infra2_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/allow_no_texture_points: False
 * /cam_1/realsense2_camera/base_frame_id: cam_1_link
 * /cam_1/realsense2_camera/calib_odom_file: 
 * /cam_1/realsense2_camera/clip_distance: -2.0
 * /cam_1/realsense2_camera/color_fps: -1
 * /cam_1/realsense2_camera/color_frame_id: cam_1_color_frame
 * /cam_1/realsense2_camera/color_height: -1
 * /cam_1/realsense2_camera/color_optical_frame_id: cam_1_color_optic...
 * /cam_1/realsense2_camera/color_width: -1
 * /cam_1/realsense2_camera/confidence_fps: -1
 * /cam_1/realsense2_camera/confidence_height: -1
 * /cam_1/realsense2_camera/confidence_width: -1
 * /cam_1/realsense2_camera/depth_fps: -1
 * /cam_1/realsense2_camera/depth_frame_id: cam_1_depth_frame
 * /cam_1/realsense2_camera/depth_height: -1
 * /cam_1/realsense2_camera/depth_optical_frame_id: cam_1_depth_optic...
 * /cam_1/realsense2_camera/depth_width: -1
 * /cam_1/realsense2_camera/device_type: 
 * /cam_1/realsense2_camera/enable_accel: False
 * /cam_1/realsense2_camera/enable_color: True
 * /cam_1/realsense2_camera/enable_confidence: True
 * /cam_1/realsense2_camera/enable_depth: True
 * /cam_1/realsense2_camera/enable_fisheye1: False
 * /cam_1/realsense2_camera/enable_fisheye2: False
 * /cam_1/realsense2_camera/enable_fisheye: False
 * /cam_1/realsense2_camera/enable_gyro: False
 * /cam_1/realsense2_camera/enable_infra1: False
 * /cam_1/realsense2_camera/enable_infra2: False
 * /cam_1/realsense2_camera/enable_infra: False
 * /cam_1/realsense2_camera/enable_pointcloud: False
 * /cam_1/realsense2_camera/enable_pose: False
 * /cam_1/realsense2_camera/enable_sync: False
 * /cam_1/realsense2_camera/filters: spatial,temporal,...
 * /cam_1/realsense2_camera/fisheye1_frame_id: cam_1_fisheye1_frame
 * /cam_1/realsense2_camera/fisheye1_optical_frame_id: cam_1_fisheye1_op...
 * /cam_1/realsense2_camera/fisheye2_frame_id: cam_1_fisheye2_frame
 * /cam_1/realsense2_camera/fisheye2_optical_frame_id: cam_1_fisheye2_op...
 * /cam_1/realsense2_camera/fisheye_fps: -1
 * /cam_1/realsense2_camera/fisheye_frame_id: cam_1_fisheye_frame
 * /cam_1/realsense2_camera/fisheye_height: -1
 * /cam_1/realsense2_camera/fisheye_optical_frame_id: cam_1_fisheye_opt...
 * /cam_1/realsense2_camera/fisheye_width: -1
 * /cam_1/realsense2_camera/gyro_fps: -1
 * /cam_1/realsense2_camera/gyro_frame_id: cam_1_gyro_frame
 * /cam_1/realsense2_camera/gyro_optical_frame_id: cam_1_gyro_optica...
 * /cam_1/realsense2_camera/imu_optical_frame_id: cam_1_imu_optical...
 * /cam_1/realsense2_camera/infra1_frame_id: cam_1_infra1_frame
 * /cam_1/realsense2_camera/infra1_optical_frame_id: cam_1_infra1_opti...
 * /cam_1/realsense2_camera/infra2_frame_id: cam_1_infra2_frame
 * /cam_1/realsense2_camera/infra2_optical_frame_id: cam_1_infra2_opti...
[... 267 lines cut ...]
[ INFO] [1620023267.370824259]: Expected frequency for depth = 30.00000
[ INFO] [1620023267.386364551]: Expected frequency for color = 30.00000
[ INFO] [1620023267.396254492]: setupStreams...
[ INFO] [1620023267.417498629]: insert Depth to Stereo Module
[ INFO] [1620023267.417546521]: insert Color to RGB Camera
[ INFO] [1620023267.466236248]: SELECTED BASE:Depth, 0
[ INFO] [1620023267.471884995]: RealSense Node Is Up!
 03/05 16:27:47,550 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[ WARN] [1620023267.680723885]: 
 03/05 16:27:47,681 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:47,731 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:47,782 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:47,832 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:47,883 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[... cut ...]
 03/05 16:27:55,345 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:56,396 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:56,446 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:58,498 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```

MartyG-RealSense, 2021-05-03:
Hi @victorll998  I am not involved in cases related to T265.  Could you repost your question as a new case with T265 included in the message title using the green **New Issue** button on the front page of the forum please so that the appropriate RealSense team member can handle the case?  Thanks!

https://github.com/IntelRealSense/realsense-ros/issues