realsenseai/realsense-ros #1163: Sharing my hack: Convert depth image to laser scan

dschnabel, 2020-04-12:
I thought I'll share my latest hack on [converting the depth image provided by the D435 to laser scan](https://github.com/dschnabel/realsense-ros/compare/development...dschnabel:laserscan).

I basically took the code from http://wiki.ros.org/depthimage_to_laserscan and modified it to directly take an `cv::Mat` image.

This hack publishes `sensor_msgs::LaserScan` messages to the `/camera/scan` topic directly from _realsense-ros_. I needed this conversion because [move_base](http://wiki.ros.org/move_base) doesn't take depth images. I could have used the pointcloud that's already published by _realsense-ros_ but I'm running on a Raspberry Pi4 so I'm low on CPU resources and pointclouds use a lot of CPU.

Alternatively you could use the standalone 'depthimage_to_laserscan' ROS package which takes an `sensor_msgs::Image` message and converts it to LaserScan, but this is less efficient since there's one extra conversion:

**realsense-ros + standalone depthimage_to_laserscan (without my hack):**
```cv::Mat --> sensor_msgs::Image --> sensor_msgs::LaserScan```

**my realsense-ros hack:**
```cv::Mat --> sensor_msgs::LaserScan```

----

I've ported over four of the [config options](http://wiki.ros.org/depthimage_to_laserscan#Parameters):
```
~scan_height (int, default: 1 pixel)
    The number of pixel rows to use to generate the laserscan. For each column, the scan will return the minimum value for those pixels centered vertically in the image. 
~range_min (double, default: 0.45m)
    The minimum ranges to return in meters. Ranges less than this will be output as -Inf. 
~range_max (double, default: 10.0m)
    The maximum ranges to return in meters. Ranges greater than this will be output as +Inf.
~output_frame_id (str, default: d435_frame)
    The frame id of the laser scan. For point clouds coming from an "optical" frame with Z forward, this value should be set to the corresponding frame with X forward and Z up.  
```

I added one additional option:
```
~scan_tilt (int, default: 0 pixel)
    The vertical section of the image to use (in pixel). 0 means use image center y. A positive number moves the center down, a negative number moves the center up. Just like tilting a camera down or up.
```

These options can be set in your D435 launch file, e.g.
```
<launch>
   <group ns="d435">
      <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">

          [... other configs ...]

         <arg name="scan_range_min"       value="0.1"/>
         <arg name="scan_range_max"       value="3.0"/>
         <arg name="scan_height"          value="70"/>
         <arg name="scan_tilt"            value="14"/>
         <arg name="scan_output_frame"    value="my_frame"/>
      </include>
   </group>
</launch>
```

Maybe this is going to be useful for others as well.

MartyG-RealSense, 2020-04-12:
Thanks so much for your contribution!  I have shared it on another of Intel's RealSense forums.  :)

https://support.intelrealsense.com/hc/en-us/community/posts/360043450734-Highlighting-a-user-contributed-ROS-program-for-converting-D435-laser-image-to-depth-scan

MartyG-RealSense, 2020-04-26:
Thanks again for your great contribution!  I will close this case now, as your contribution will be findable through internet searches in the future.

closed 2020-04-26 by MartyG-RealSense

jpdevicente, 2022-04-08:
@dschnabel Hi! I am trying to use your hack of realsense-ros using your branch laserscan with my d435, but when I run your lanchers I dont see the scan topic being published anywhare. Could you provide some instructions in how to use it?

MartyG-RealSense, 2022-04-08:
Hi @jpdevicente   Are you using librealsense SDK version **2.33.1** as recommended by the documentation of the forked wrapper by @dschnabel  please?

https://github.com/dschnabel/realsense-ros

A specific ROS wrapper version should ideally be matched with a specific librealsense SDK version that is listed in that wrapper version's release notes.  I assume that the forked  @dschnabel wrapper is based on ROS wrapper 2.2.13 that is the correct match for librealsense 2.33.1.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.13

Another complicating factor in using this fork is that if you are using librealsense 2.33.1 then the maximum camera firmware driver version that you should be using is **5.12.7.100**.  This is because the next firmware after that, 5.12.8.200, requires a minimum librealsense version of **2.39.0** due to internal changes in the firmware from that point onwards.

jpdevicente, 2022-04-08:
No, I'm using the most recent librealsense SDK as the most recent d435 firmware, so that must be the problem. That being the case, I will just continue to use the ros package deptimage_to_laserscan as a middle step. Thank you for your reply!

MartyG-RealSense, 2022-04-08:
You are very welcome, @jpdevicente - I'm pleased that I could help.  :)