realsenseai/realsense-ros #1164: Jetson nano with D435i using ros melodic (resource temporarily unavailable)

Ahmedkaramakka, 2020-04-12:
I have been facing a problem with D435i working on jetson nano using the latest image provided by NVIDIA, I will try to give a detailed recap about what I had done so far.
I installed ROS melodic following steps provided by ( https://www.jetsonhacks.com/2019/10/23/install-ros-on-jetson-nano/ ) , also I had done the same installing ( Librealsense) following this link 
( https://www.jetsonhacks.com/2019/05/16/jetson-nano-realsense-depth-camera/ ) 
then I suffered a lot with the realsense ros wrapper with error considering the USB and by searching i found issue #1088 
discussing the USB parsing so i updated my files and wrote the following instructions ( from installing realsense ros wrapper guide on github)

catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc

the problem of USB part appeared to be solved but another problem appears, many time i got the same error exactly as #1161 ( Synthetic Stream Error )
sometimes i got an error saying i am an outframe , that appears when trying to launch rs_camera launch file ( although i made another trial modifying the frame rate to be 15 instead of 30 ) 

my end goal if camera worked probably is drawing  a map using depth to image laser scan package, gmapping, rtabmap , tf ) seeing all of these on rviz while drawing the map 

this is a copy of my terminal considering the error when launching the rs_camera launch file 

 roslaunch realsense2_camera rs_camera.launch filters:=pointcloud unite_imu_method:=linear_interpolation  align_depth:=true
... logging to /home/swarm/.ros/log/8ec3d74e-7cba-11ea-a4d6-00044be6a537/roslaunch-swarm-desktop-18985.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.2:44127/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
[... 160 lines cut ...]
 12/04 14:46:05,250 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 12/04 14:46:05,714 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:05,715 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 12/04 14:46:05,848 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:05,848 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 12/04 14:46:05,983 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:05,983 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 12/04 14:46:06,450 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:06,450 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 12/04 14:46:06,584 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:06,585 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 12/04 14:46:06,715 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:06,715 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 12/04 14:46:07,183 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:07,184 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 12/04 14:46:07,315 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:07,322 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 12/04 14:46:07,449 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:07,450 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 12/04 14:46:07,983 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:07,983 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 12/04 14:46:08,115 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:08,121 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
^C 12/04 14:46:08,648 WARNING [545720885632] (types.cpp:49) Out of frame resources!
 12/04 14:46:08,649 ERROR [545720885632] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!

MartyG-RealSense, 2020-04-13:
Hi @Ahmedkaramakka  I just wanted to acknowledge that your case has been seen.  As you mention, your case is similar to the 'Synthetic Stream Error' case and also resembles other cases involving these errors.  I would recommend adding a comment to the Synthetic Stream Error case to make sure that your problem is also highlighted to the RealSense ROS wrapper developer there as well.

Other than that, all the advice that I have on this subject is covered in the 'Synthetic Streams' case.  Thanks so much for your patience!

Ahmedkaramakka, 2020-04-14:
Thanks, @MartyG-RealSense  for your comment and time , so do you advise me to put my whole problem as comment in the issue I tagged ? , especially I had mentioned some details that make the case differs from the previous cases like changing the frame rate in the launch file and still got the errors shown

MartyG-RealSense, 2020-04-14:
@Ahmedkaramakka   It is always very helpful if additional details can be provided that will help the diagnostic process.  Thank you very much!

Ahmedkaramakka, 2020-04-16:
@MartyG-RealSense  sorry for asking too many questions, but do you suggest a certain solution for my problem ?

MartyG-RealSense, 2020-04-16:
There is no limit on the number of questions that you can ask, so do not worry.   :)

Do you know whether your project can operate correctly on more powerful hardware than the Jetson Nano, please?

Ahmedkaramakka, 2020-04-16:
yes, it worked on my laptop but I was using ROS kinetic not melodic

MartyG-RealSense, 2020-04-16:
Installing Librealsense on Nano is a trickier process than on a standard PC such as a laptop, as I'm sure you have found!

I re-read this case from the start.  It loos as though you installed ROS on the Nano with one guide and Librealsense on the Nano with a different guide.   you may have a smoother process if you clean the Nano out and start from the beginning with a single guide for installing both Librealsense and ROS on Nano:

https://github.com/JetsonHacksNano/installRealSenseROS

jose-fernando-UB, 2020-04-17:
I have the same issue, and the same setup as OP.

I do not want to clean the Nano out for many reasons.

I do not think this is a compatibility issue, it seems like nvidia does not care to provide a solution to this problem.

Do not advertise yourself as multi-platform if I cannot make a simple robotic application with your new/expensive products

MartyG-RealSense, 2020-04-17:
@jose-fernando-UB   Intel tries very hard to make RealSense work on as many software and hardware platforms as possible.  The camera hardware is extremely flexible in terms of the computing devices it will work with, and the SDK software does not include a lot of dependencies that would negatively influence its compatibility with other software.

However, there are a small number of computing device platforms that have more compatibility challenges than others, and this is outside of the RealSense developers' control.  Intel tests a lot and encourages developers to publish and contribute their own solutions.    In the case of the Nvidia Jetson range, the website 'JetsonHacks' are an especially valuable source for installation guides for the Jetson range (Nano, TX, Xavier).  

Problems tend to get ironed out over time as users contribute knowledge, features are added to the RealSense SDK and reference guides emerge.  I can appreciate that this may not be of comfort to users who are having problems before such assistance is available.  I promise that we always try our best to help community members find solutions though.

Ahmedkaramakka, 2020-04-18:
@MartyG-RealSense  I tried to install realsense ros from same source as you said ( jetson hacks) but the problem still there this is the copy of the log 

roslaunch realsense2_camera rs_camera.launch filters:=pointcloud unite_imu_method:=linear_interpolation  align_depth:=true
... logging to /home/swarm/.ros/log/aa64f88e-81ba-11ea-8dbe-00044be6a537/roslaunch-swarm-desktop-11737.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://swarm-desktop:36009/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 309 lines cut ...]
 18/04 23:25:57,197 WARNING [547599933824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 23:25:57,257 WARNING [547599933824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 23:25:57,266 WARNING [547599933824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 23:25:57,325 WARNING [547599933824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 23:25:57,335 WARNING [547599933824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 23:25:57,394 WARNING [547599933824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 23:25:57,404 WARNING [547599933824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 23:25:57,463 WARNING [547599933824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 23:25:57,494 WARNING [547599933824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
[... cut ...]
 18/04 23:25:57,914 ERROR [547574755712] (error-handling.cpp:68) Error during polling error handler: get_xu(id=7) failed! Last Error: Success
 18/04 23:25:57,924 WARNING [547691860352] (types.cpp:49) set_xu(ctrl=1) failed! Last Error: Success
 18/04 23:25:57,925 WARNING [547599933824] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 135 error: Device or resource busy, number: 16
 18/04 23:25:57,943 ERROR [547691860352] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Success
terminate called without an active exception

MartyG-RealSense, 2020-04-18:
@Ahmedkaramakka   At the moment, the standard advice that I am giving for the various cases involving **control_transfer returned** errors is to try using the camera with a mains electricity powered USB 3 hub if you have one or can obtain one.  

Otherwise, the only other course of action is to await a response from the RealSense ROS wrapper developer, as I am on the RealSense GitHub support team and not the developer team, and so regrettably have no means to provide a solution through programming.  I truly understand the frustration that you must be experiencing and appreciate your patience.

Ahmedkaramakka, 2020-04-18:
@MartyG-RealSense it is really frustrating, as I changed the files and tried many suggestions provided in many issues comments, I really hope to get a reply from the developer team as soon as possible
but I want to assure that I do appreciate your help and your time

Ahmedkaramakka, 2020-04-18:
@jose-fernando-UB did you try to contact NVIDIA for that issue ?

jose-fernando-UB, 2020-04-21:
@Ahmedkaramakka No I haven't, but maybe someone should given how common this problem is..

jose-fernando-UB, 2020-04-21:
@doronhi

jose-fernando-UB, 2020-04-21:
@doronhi

Caveats, 2020-04-28:
Hey - so i've hit this issue too (and thanks @jose-fernando-UB all for your sleuthing and @MartyG-RealSense for the updates). I guess all we can ask is for this to be expedited.  

I've seen elsewhere some points along the same lines: e.g. https://github.com/IntelRealSense/realsense-ros/issues/1161 and #1110 and #2045

**Hardware consideration**
> If you are still experiencing these errors, I would recommend investigating the possibility of using a mains electricity powered USB 3 hub to increase USB port stability if your project allows it (for example, it is not a mobile robot that would be unsuited to being tethered to a wall power socket).
> 
> Otherwise, the situation may require programming additions from the RealSense developers to reduce the likelihood of errors such as control_transfer returned occurring. So I will close the case, as the developers are aware of the problem and are likely investigating it.

On the hardware front just a word of caution -  

> is to try using the camera with a mains electricity powered USB 3 hub if you have one or can obtain one.
> 

I'm using a realsense 435i and have had similar issues with  "Exception...processing callback!
Whilst i'm planning on flashing and trying a fresh CUDA install and Kernel update - as well as getting some of the refresh issues running on a Jetson Nano B01 

But it looks like the USB may be a factor: 

1) Its basically using the entire bandwidth / data capacity of the USB C - 

https://github.com/IntelRealSense/librealsense/issues/2045

2) USB C powered may well help - but this ties into a lot of experience from other Devs. 
https://software.intel.com/en-us/forums/realsense/topic/557823

I'd been hoping to be able to run a cable to my window from the Jetson Nano to use the outdoor space to map against a static view. However it looks like a longer cable - in part due to quality and that the USB-C is running at both power and data capacity - that this is advised against. 

So I suspect for £15-30 GBP this will solve some of the issues (will report back). 
I already bridged the Nano power supply for this - but its a known issue for the Realsesne and the power limitations of the Nano (basically the Nano can happily use all it's PSU input - 

(Couldn't find source - but is in my developer / install notes) - "I am currently using the Micro USB connection powered with a 5v 2.1 amp plug, which I believe is adequate for running the system. I find the nano slows to a crawl when running “roslaunch realsense2_camera rs_camera.launch” and when trying to run realsense-viewer it crashes entirely after about 20 seconds. The camera (D415) is being powered exclusively from the usb 3.0 cable connecting it to the Nano."

"The Jetson alone can easily use 2A @ 5V if you are in 10W mode. There isn’t enough power to run both the Jetson and the RealSense simultaneously from the power supply that you are using."

3) Don't use a long cable !  Instead bridge the node (sensor) to the processor.... 

4) Install CUDA and make processing load / lightweight. 

5) Hope for a Realsense fix. 

6) Sacrifice a first born for a simple fix script via JetsonHacks. 

I'm going to order some USB - C powered mains - and can advise on this - but this seems to be a mix of Hardware / Software issues but given this impacts the raw input (into USB-C > Device - there's not too much we can do but test and wait). 

For reference this also helps to explain USB-C in more detail. 
computerworld.com/article/2488194/usb-c-explained-how-to-get-the-most-from-it-and-why-its-great.html

I've previously had issues with USB-C hardware not being up to scratch (Custom built PC) due to power and the USB interface (computer side). I've also seen some other threads based on the interface on the computer side not being able to support the data rate on other Realsense Cam + Single board comp. setups. 

TLDR - the pipe is full and the device may not have enough power. Add power - Wait for fix (its a well known issue with the Jetson use case (but seriously fix this ASAP please!) or if you can't there seem to be some workarounds that reduce the overhead on the data - but at the expense of data - e.g. fewer frames - etc.  

Finally there are some posts about reducing the frame rate of the Realsense Cam itself - but this is beyond me. (for now and also RealSense have fairly stated that due to the need to test all of these settings if it's a default config - e.g. 24fps) it creates a really big UAT bottleneck and testing overhead. 

It might be the way they go - but engineers are going to engineer. 

Hope that helps - happy for this post to be sanitised where relevant.

Caveats, 2020-04-28:
Potentially off-topic - but I was able to get ROS Melodic and RVIZ running RTAB map on a B01 instance. 

What's really odd is that on one run - it chewed up all the resources - and reduced to 7FPS. 
On the second run - it worked for a time then crashed.  

I've run the same SDK via Win10 on a surface pro3 as well as via a Core i7 with a boat load of resources. On the Win10 sp3 - low power mode / or any other apps - had a tendency to crash the SDK. On the main machine - with a secondary power source to the USB 3 ports - no issues. 

So my best guess is try the power - avoid the cable - and look at a custom / install from binary melodic install as per Jetson hacks (fewest problems before I managed to break the kernel). 

Also I'm on my third Flashing of the Jetson - the issue seems to lie somewhat with ROS implementation of RealSense - in particular the ROS2 release. Whereas the viewer happily streams data. It may be possible to use the Realsense Viewer / SDK2.0 to generate a BAG file and run that way (not ideal - but gives you a raw input file to playback and tinker with). 

This is the direction I'm considering taking as a stopgap - use a 10GB ROS Bag file and push it into ROS for playback - as if it was the sensor input. Not ideal but removes the blocker. 

Hope this is all helpful.

Caveats, 2020-04-28:
Finally I also get what appears to be a memory leak - where the ROS node for Realsense gradually consumes all RAM on the Jetson Nano. 

source /opt/ros/dashing/setup.bash
source ~/ros2_ws/install/local_setup.bash
# console #1 launch rviz2
ros2 run rviz2 rviz2 -d realsense_ros2_camera/rviz/ros2.rviz
# console #2 launch realsense_ros2_camera
ros2 run realsense_ros2_camera realsense_ros2_camera
aku@aku-desktop:~$ ros2 run realsense_ros2_camera realsense_ros2_camera

[INFO] [RealSenseCameraNode]: RealSense Node Is Up!

Approx ten mins running (despite no data being published into nodes due to crash / above issue - or sporadic via ROS1 Melodic instance) 

So it could be as _"simple"_as the ROS Wrapper misbehaving. 
Happy to issue some reproducible steps or assist with validation if helpful.

MartyG-RealSense, 2020-04-29:
@Caveats   Thanks so much for your extensive and detailed feedback and suggestions!

Cabling distances of 100 meters can be achieved with the right equipment.  For example, a USB cable with 'repeater' technology built into it can go up to a recommended maximum length of 20 m.  100 m can be achieved with fiber optic cabling.  The video below by the RealSense supporting industrial-grade USB equipment supplier **Newnex** illustrates this.

https://youtube.com/watch?v=GLQgR1jT04M

Alternatively, distance can be extended beyond 30 m with ethernet cabling.  An example is the 'EtherSense' project:

https://github.com/krejov100/EtherSense

Caveats, 2020-04-29:
@MartyG-RealSense - This looks incredible! Thank you so much.

MartyG-RealSense, 2020-05-11:
Hi @Caveats   Is it okay to close this case now if you have no further problems?  Thanks!

closed 2020-05-17 by MartyG-RealSense