realsenseai/realsense-ros #1165: Add USB plug/cable to model

JStech, 2020-04-13:
I recently broke a USB cable plugged into a D415 as a result of a self-collision. I've updated our model to include a plug with a short length of cable to prevent this, but it'd be a nice feature to have the plug in the URDF with a parameter controlling whether it is included.

MartyG-RealSense, 2020-04-13:
Thanks so much for your suggestion @JStech - I have labelled this message with an "Enhancement" (Feature Request) tag so that it can be tracked by @doronhi the RealSense ROS wrapper developer.

RealSenseCustomerSupport, 2020-04-17:
----------------------------------------------

Hi JStech,

Are you saying you physically "broke" the USB cable? ... by cutting it?

Thanks

JStech, 2020-04-17:
I bent it
[image]

RealSenseSupport, 2020-04-17:
Oh, even the locking cable plug is bent? ... that looks hit very hard.

MartyG-RealSense, 2020-07-18:
Hi @JStech    Your pull request still has some conflicts that need to be resolved in order for the contribution to be merged into the SDK.   Are you still working on this pull please?   Thanks!

https://github.com/IntelRealSense/realsense-ros/pull/1173

doronhi, 2020-07-19:
> Hi @JStech Your pull request still has some conflicts that need to be resolved in order for the contribution to be merged into the SDK. Are you still working on this pull please? Thanks!
> 
> #1173

Thanks for the reminder @MartyG-RealSense . @JStech , it looks like a nice addition. It would be great if you could update your PR and we'll try to merge it by the next release.

JStech, 2020-07-20:
Thanks for the ping. Should be good to go.

doronhi, 2020-07-21:
Thanks for the PR. It's merged now.

MartyG-RealSense, 2020-07-26:
As the PR has been merged, I will close this case now.  Thanks everyone!

closed 2020-07-26 by MartyG-RealSense