realsenseai/realsense-ros #1167: How to publish disparity map?

jorgemia, 2020-04-14:
I’ve seen there is a disparity filter that can be turned on, but see no difference when visualising the depth stream through rviz. Where can I see the changes when applying this filter? Is it possible to publish a topic with the disparity map and another topic with the depth (in metres) already processed simultaneously?

MartyG-RealSense, 2020-04-15:
A filter type can be selected in the **roslaunch** instruction.

https://github.com/IntelRealSense/realsense-ros/issues/742#issuecomment-487034341

You can then define the specific settings of the filter in a 'dynamic reconfigure' **rosrun** command immediately after the roslaunch instruction.

https://github.com/IntelRealSense/realsense-ros/issues/583

I hope that applying filter settings with a rosrun instruction will give you the results you need when the stream is visualised in RViz.

jorgemia, 2020-04-15:
@MartyG-RealSense yes I’ve been able to turn on the filter by adding it to the roslaunch command however don’t know what topic the filter is being applied to? 
Is it the depth topic? When visualising in rviz I saw no difference between turning the filter on or off. Maybe I’m visualising the wrong topic/data stream?

MartyG-RealSense, 2020-04-15:
In Intel's D435i ROS tutorial for a SLAM point cloud, the topic used to render the cloud in RViz is  /**camera/color/image_raw**

MartyG-RealSense, 2020-04-15:
A list of published topics is here:

https://github.com/IntelRealSense/realsense-ros#published-topics

**camera/depth/image_rect_raw** seems like a good candidate for a choice of topic.

jorgemia, 2020-04-15:
In realsense-viewer, if you turn off the _Disparity to Depth_ filter in the Depth Stream, I believe you get to see the Disparity map (instead of Depth Map), as the colors invert (This is logical as _Disparity = (baseline * focal length)/depth_ is the inverse of depth). 

Yes, I am checking **camera/depth/image_rect_raw** in RViz, yet I am seeing no change in the data even if I turn the disparity filter on. I have also checked individual pixel values when applying the filter and they still give me depth data instead of disparity data.

jorgemia, 2020-04-15:
When turning the filter on, the following topics appear:

```
/camera/disparity_end/parameters_descriptions
/camera/disparity_end/parameters_updates
/camera/disparity_start/parameters_descriptions
/camera/disparity_start/parameters_updates
```

These don't seem to publish the disparity data as their data is of the _dynamic_reconfigure_ type. When running `rqt_reconfigure`, however, no options appear for those topics.

jorgemia, 2020-04-15:
As per the comment in #724 :

`disparity - convert depth to disparity before applying other filters and back.`

could it be that the disparity filter changes to disparity and then back to depth again? So I can't actually see the disparity map?

MartyG-RealSense, 2020-04-15:
I wondered too about the "and back" part when I saw it.  I agree that tagging Doronhi into the discussion was a good idea, as I researched this problem deeply and could not find a solution.  I do apologize.

doronhi, 2020-04-17:
Some of the filters give better results when applied over the disparity image then over the depth image. A recommended scheme for using filters is mentioned in [post-processing-filters.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#using-filters-in-application-code)
The "disparity" filter in the ROS wrapper enables the depth2disparity transform before some of the other filters and than the disparity2depth transform to produce the depth image again.
Therefor, when applied, it won't make any difference by itself but may  improve the performance of other filters.
I hope that helps to explain what you see and apologize for the confusion and misunderstanding caused by the "and back" part. I will happily accept any description improvements.

jorgemia, 2020-04-17:
@doronhi yeah understood! Thanks! 

Is there no way of publishing the disparity map instead of the depth one then?

MartyG-RealSense, 2020-05-17:
Hi @jorgemia   Do you still require assistance with this case, please?  Thanks!

jorgemia, 2020-05-17:
I never got an answer so I used an alternative method and no longer need a disparity map...but publishing the disparity map as a topic could be a useful feature to have. Don’t think this should be a hard thing to implement as the camera already produces this and then transforms it into depth. So would be good to have an option which you can turn on.

MartyG-RealSense, 2020-05-17:
The link below has a suggestion about how to visualize a disparity map from **stereo_image_proc** in RViz or rqt:

https://gitee.com/null_937_5399/avoidance#local-planner

MartyG-RealSense, 2020-05-30:
@jorgemia   Do you require further assistance please or can this case be closed?   Thanks!

closed 2020-06-05 by MartyG-RealSense