realsenseai/realsense-ros #1170: Realsense D435i gazebo 9 plugin

ashirsat, 2020-04-16:
Hi,

I am currently trying to use real sense d435i camera for gazebo9 running on Ubuntu 18.04 with Ros-melodic. I have not been successful in finding a plugin for the d400 series.
Is there an already existing one ?
If not, is it planned to be released soon? 
Due to current pandemic situation real world experiments are increasingly proving difficult to perform, and this would really help in keeping the testing and usability of Intel realsense at least in Software in the loop for aerial robots.
@RealSenseCustomerSupport, @doronhi 

Thanks,
Aniket

MartyG-RealSense, 2020-04-16:
The most recent project I know of involving the RealSense D435 in robotics that can make use of Gazebo is the 'Erwhi Hedgehog' project.

https://gbr1.github.io/erwhi_hedgehog.html

[image]

Before that, there was SyrianSpock's Gazebo plugin for ROS and the RealSense R200 camera, which itself was an evolution of an official Intel Gazebo plugin for R200.

https://github.com/SyrianSpock/realsense_gazebo_plugin

ashirsat, 2020-04-16:
Thanks for the info. @MartyG-RealSense 

I had reviewed that project before posting the question. It does not provide a gazebo plugin for the software in the loop testing for d435i. They just use the rviz visualizer for rendering the map from actual realsense d435 camera.

How different would be the plugin for d435 from the one that is there for r200. Would i be able to build d435i plugin it using the r200 plugin?

Thanks,
Aniket

MartyG-RealSense, 2020-04-16:
It looks like a .dae model for the D435 would be needed to substitute in for the realsense,dae model of the R200 in the project.

Intel provide CAD models of the RealSense cameras in STEP model file format here:

https://dev.intelrealsense.com/docs/cad-files

Conceivably, you could then use a STEP to dae convertor to convert the model to dae format.  Googling for '**convert step to dae**' provides some examples of conversion tools for doing so.

Regarding what other changes would need to be made to the values in the **model.sdf** file, I'm not clear.

I'm off-shift now as it is near midnight in my time zone.  I will be back in 7 hours from the time of writing this though and will be happy to continue the conversation if you need further advice.   Good luck!

ashirsat, 2020-04-17:
@MartyG-RealSense  I was more concerned about the actual realsense plugin code it would be different from r200 one because the optics changed and it does not have an imu and also i am not sure how to encode the infrared pattern

MartyG-RealSense, 2020-04-17:
The R300 is the original ancestor of the D435i (they are both Stereo cameras), so they are similar in that respect, though of course they have a different field of view size and the D435i supports higher resolutions and frames per second speeds.  It is true that lack of simulation for the IMU would be another complicating factor.

The RealSense SDK provides a system called **software device** to simulate cameras in software without having the physical camera, though it probably is not as ideal as Gazebo in terms of simulating robot projects.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

ashirsat, 2020-04-19:
@MartyG-RealSense  Can you point me to the R200 gazebo plugin? I can start building my plugin for d435i based off on that. If I have the geometry of the realsense d435i, I can probably add the imu to the plugin for d435i. I would also need the orientation of the imu in the actual camera

MartyG-RealSense, 2020-04-19:
The link below is the official Intel Gazebo plugin for the R200 that preceded Syrianspock's.

https://github.com/intel/gazebo-realsense

ashirsat, 2020-04-21:
Hi @MartyG-RealSense . Thanks for the link. I will take a look and reach out in case I have any issues or if the plugin works I will keep you in the loop.

takanotume24, 2020-04-26:
Hi, I'm also looking for a way to get the D435 to work on the Gazebo9.
I found a description like this, is it helpful? 
https://github.com/pal-robotics/realsense_gazebo_plugin/issues/7

MartyG-RealSense, 2020-04-26:
@takanotume24   That link looks very interesting - thanks!  I don't have the ability to test it myself right now but I'd love to hear if others have success with it.

takanotume24, 2020-04-27:
@MartyG-RealSense Turtlebot3 waffle pi + realsense d435 now works in gazebo. Follow the README of the following repository.
https://github.com/takanotume24/turtlebot3_simulations
The dependencies should be resolved by rosdep, but if you have any errors, can you write them in the issue in my repository?

The model with the changes can be found in this(https://github.com/takanotume24/turtlebot3) repository.
diff -> https://github.com/ROBOTIS-GIT/turtlebot3/compare/master...takanotume24:master

MartyG-RealSense, 2020-04-27:
@takanotume24   Thanks!  I will create an announcement post on the forum front page that links here so that Gazebo users can try out your repository and feed back.  Thanks for your contribution   :)

ashirsat, 2020-04-28:
@takanotume24  Thanks for linking the repo. @MartyG-RealSense  From a cursory look through the code, I think that repo will do the job for the plugin side. Its still missing the realsense d435 sdf and dae file for gazebo rendering and frame represenations. Is this something that @MartyG-RealSense could provide in the realsense ros repo from Intel?

Thanks,
Aniket

MartyG-RealSense, 2020-04-28:
Hi @ashirsat   I am not involved in development as I am on the support team.  @doronhi is the developer of the RealSense ROS wrapper.

takanotume24, 2020-04-28:
@ashirsat @MartyG-RealSense I'm sorry, I was failing to resolve the dependency. Can you try again following the [turtlebot3_simulations/README.md at with-realsense-d435 · takanotume24/turtlebot3_simulations](https://github.com/takanotume24/turtlebot3_simulations/blob/with-realsense-d435/README.md)? (If possible, please clear out catkin_ws/src or create a new workspace and try again).

The D435 model uses a forked version of https://github.com/pal-robotics-forks/realsense.

ashirsat, 2020-04-29:
@takanotume24 I don't see the realsense d435 plugin or the urdf/sdf in the linked repo. Am I looking at the wrong branch or web hook?

@doronhi Would we be able to get the d435i sdf file in the realsense ros repo itself?

Thanks,
Aniket

ashirsat, 2020-05-06:
@MartyG-RealSense  @doronhi Can anyone provide any update how to incorporate the d435 in gazebo. @takanotume24 has not linked the plugin nor the urdf file for the realsense gazebo simulation in the previous message repo.

Thanks,
Aniket

takanotume24, 2020-05-07:
@ashirsat Is the simulator not working?  What errors do you get in your environment when you follow the ``README`` of the ``with-realsense-d435`` branch?
If you're looking for the URDF file for D435, it's at https://github.com/pal-robotics-forks/realsense/tree/kinetic-devel/realsense2_description/urdf, and this file combined with Turtlebot3 is my ``turtlebot3_simuration`` repository.

MartyG-RealSense, 2020-05-07:
@takanotume24  Thank you for your further help with this discussion!

ashirsat, 2020-05-07:
@takanotume24 I see what i missed. I was on the wrong branch. I will checkout the branch that you suggested and get back to you.

Thanks for the help.

Regards,
Aniket

MartyG-RealSense, 2020-05-17:
Hi @ashirsat   Do you require further assistance with this question please, or is it okay to close this case?   Thanks!

ashirsat, 2020-05-17:
I haven’t yet checked the suggested solution. But you can close the case.
If it does not work I will re-open the issue

-- 
Regards,
Aniket Shirsat

Mechanical Graduate
Arizona State University| Ira A. Fulton Schools of Engineering
School for Engineering of Matter Transport and Energy
Cell:(480)371-8843
Portfolio <http://seelio.com/aniketshirsat>
<http://www.linkedin.com/in/shirsataniket>
<https://twitter.com/intent/follow?original_referer=https%3A%2F%2Fabout.twitter.com%2Fresources%2Fbuttons&region=follow_link&screen_name=AniketRShirsat&tw_p=followbutton>

MartyG-RealSense, 2020-05-17:
Ok, thank you very much!

closed 2020-05-17 by MartyG-RealSense

rickstaa, 2021-12-29:
I will also put this here for future reference. An example of how to use the D435i camera in Gazebo can be found [here](https://github.com/rickstaa/realsense-ros-gazebo), might people need it. A detailed explanation video created by [@issaiass](https://github.com/issaiass) can  be found [here](https://www.youtube.com/watch?v=hpUCG6K5muI&feature=youtu.be).

MartyG-RealSense, 2021-12-29:
Thanks very much @rickstaa for posting your resource links   :)