realsenseai/realsense-ros #1171: D435i ros-wrapper unstable and crash base on jetson nano

lightsalt2011, 2020-04-17:
[ INFO] [1587093001.067961095]: Initializing nodelet with 4 worker threads.
[ INFO] [1587093001.352782130]: RealSense ROS v2.2.11
[ INFO] [1587093001.352886622]: Running with LibRealSense v2.31.0
[ INFO] [1587093001.725060607]:
17/04 11:10:02,277 WARNING [547561193856] (types.cpp:49) hwmon command 0x4f failed. Error type: No data to return (-21).
[ INFO] [1587093002.330385056]: Device with serial number 947122070521 was found.

[ INFO] [1587093002.330522365]: Device with physical ID 2-1.3-6 was found.
[ INFO] [1587093002.330628056]: Device with name Intel RealSense D435I was found.
[ WARN] [1587093002.337918928]: Error extracting usb port from device with physical ID: 2-1.3-6
Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[ INFO] [1587093002.345399826]: getParameters...
[ INFO] [1587093002.444041871]: setupDevice...
[ INFO] [1587093002.444127247]: JSON file is not provided
[ INFO] [1587093002.444185640]: ROS Node Namespace: camera
[ INFO] [1587093002.444246377]: Device Name: Intel RealSense D435I
[ INFO] [1587093002.444293258]: Device Serial No: 947122070521
[ INFO] [1587093002.444337690]: Device physical port: 2-1.3-6
[ INFO] [1587093002.444391447]: Device FW version: 05.11.01.100
[ INFO] [1587093002.444437130]: Device Product ID: 0x0B3A
[ INFO] [1587093002.444501096]: Enable PointCloud: Off
[ INFO] [1587093002.444548915]: Align Depth: Off
[ INFO] [1587093002.444594962]: Sync Mode: Off
[ INFO] [1587093002.445238117]: Device Sensors:
[ INFO] [1587093002.445754693]: Stereo Module was found.
[ INFO] [1587093002.445914609]: RGB Camera was found.
[ INFO] [1587093002.446011861]: Motion Module was found.
[ INFO] [1587093002.446139065]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1587093002.446203396]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1587093002.446253298]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1587093002.446310336]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1587093002.446775656]: num_filters: 0
[ INFO] [1587093002.446836080]: Setting Dynamic reconfig parameters.
[ INFO] [1587093007.733984162]: Done Setting Dynamic reconfig parameters.
[ INFO] [1587093007.788406136]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1587093007.789139975]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1587093007.789736193]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1587093007.841915916]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1587093007.843135960]: setupPublishers...
[ INFO] [1587093007.848657568]: Expected frequency for depth = 30.00000
[ INFO] [1587093007.857428216]: Expected frequency for infra1 = 30.00000
[ INFO] [1587093007.861933659]: Expected frequency for infra2 = 30.00000
[ INFO] [1587093007.866169903]: Expected frequency for color = 30.00000
[ INFO] [1587093007.876578563]: setupStreams...
[ INFO] [1587093007.930999026]: insert Depth to Stereo Module
[ INFO] [1587093007.931538988]: insert Color to RGB Camera
[ INFO] [1587093007.931698903]: insert Infrared to Stereo Module
[ INFO] [1587093007.931797404]: insert Infrared to Stereo Module
[ INFO] [1587093007.931870538]: insert Gyro to Motion Module
[ INFO] [1587093007.931975811]: insert Accel to Motion Module
17/04 11:10:08,048 WARNING [547432145280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1587093008.276969922]: SELECTED BASE:Depth, 0
[ INFO] [1587093008.317713629]: RealSense Node Is Up!
17/04 11:10:08,400 WARNING [547432145280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
17/04 11:10:08,460 WARNING [547432145280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
17/04 11:10:08,542 WARNING [547432145280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
17/04 11:10:08,681 ERROR [547113386368] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
17/04 11:10:08,724 WARNING [547432145280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
17/04 11:10:08,905 WARNING [547432145280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
17/04 11:10:08,986 WARNING [547432145280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
17/04 11:10:09,050 WARNING [547432145280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
17/04 11:10:09,488 WARNING [547432145280] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 23,
[ WARN] [1587093009.755832530]: Hardware Notification:IR stream start failure,1.58709e+12,Error,Hardware Error
[ERROR] [1587093009.756025314]: Hardware Reset is needed. use option: initial_reset:=true
17/04 11:10:09,760 WARNING [547440537984] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 9,
17/04 11:10:10,290 WARNING [547138564480] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 134 error: No such device, number: 19
17/04 11:10:10,290 ERROR [547138564480] (hid-device.cpp:153) failed to submit UVC request
17/04 11:10:10,290 WARNING [547432145280] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 135 error: No such device, number: 19
17/04 11:10:10,291 WARNING [547113386368] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 132 error: No such device, number: 19
17/04 11:10:10,291 ERROR [547113386368] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13
[... 21 lines cut ...]
17/04 11:10:11,422 WARNING [547088208256] (messenger-libusb.cpp:96) usb_request_cancel returned error, endpoint: 130 error: No such device, number: 19
17/04 11:10:11,422 WARNING [547088208256] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 130, error: No such device, number: 19
17/04 11:10:11,432 WARNING [546225090944] (messenger-libusb.cpp:96) usb_request_cancel returned error, endpoint: 131 error: No such device, number: 19
17/04 11:10:11,434 WARNING [546225090944] (messenger-libusb.cpp:96) usb_request_cancel returned error, endpoint: 131 error: No such device, number: 19
17/04 11:10:11,434 WARNING [546225090944] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 131, error: No such device, number: 19
17/04 11:10:11,466 WARNING [547432145280] (messenger-libusb.cpp:96) usb_request_cancel returned error, endpoint: 135 error: No such device, number: 19
17/04 11:10:11,497 WARNING [547113386368] (messenger-libusb.cpp:96) usb_request_cancel returned error, endpoint: 132 error: No such device, number: 19
17/04 11:10:11,497 WARNING [547113386368] (messenger-libusb.cpp:96) usb_request_cancel returned error, endpoint: 132 error: No such device, number: 19
17/04 11:10:11,502 WARNING [547113386368] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 132, error: No such device, number: 19
17/04 11:10:11,556 WARNING [547138564480] (messenger-libusb.cpp:96) usb_request_cancel returned error, endpoint: 134 error: No such device, number: 19
17/04 11:10:11,579 ERROR [548118739440] (handle-libusb.h:51) failed to open usb interface: 5, error: RS2_USB_STATUS_NO_DEVICEL4T 32.3.1
JetPack 4.3
Requires librealsense v2.31.0
Installs RealSense ROS Version = 2.2.11
17/04 11:10:11,579 ERROR [548118739440] (handle-libusb.h:51) failed to open usb interface: 5, error: RS2_USB_STATUS_NO_DEVICE

I have follow https://github.com/JetsonHacksNano/installRealSenseROS step by step , but my D435i ros-wrapper cannot work well, error log as above.
realsense_viewer and rs-xxx cmd works well *

config as below:
R32 (release), REVISION: 3.1, GCID: 18186506, BOARD: t210ref, EABI: aarch64, DATE: Tue Dec 10 06:58:34 UTC 2019
RealSense ROS v2.2.11
LibRealSense v2.31.0

is there someone give some suggestion ?

MartyG-RealSense, 2020-04-17:
I apologise for the delay in responding to your question.

The errors that you have been experiencing have also been experienced by other users.  I will paste below some advice from the ROS wrapper's developer on the meaning of some of the errors in your particular case.
...........................................
"The warning message "interrupt event received" can be ignored and indeed will be removed in future versions".

"The warning message "control_transfer returned error" reflects communication error with the device that is resolved by repeating the attempt.  These messages are still under investigation but as long as they appear only now and then, say at the beginning of the run, they can safely be ignored for now".

"If the message "messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11" is continuous, that is appears multiple times per second, then it is not good.  It suggest a real communication problem between the code and the device and can have effects".
...........................................

Are you able to visualize the camera data in RViz, please?   If so, this would indicate that the data is being correctly published to topics despite the errors in the log.  Thanks!

lightsalt2011, 2020-04-18:
the topics no data published

/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/gyro/imu_info
/camera/gyro/sample
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_rect_raw/compressedDepth/parameter_updates
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/infra2/image_rect_raw/compressed
/camera/infra2/image_rect_raw/compressed/parameter_descriptions
/camera/infra2/image_rect_raw/compressed/parameter_updates
/camera/infra2/image_rect_raw/compressedDepth
/camera/infra2/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra2/image_rect_raw/compressedDepth/parameter_updates
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates

lightsalt2011, 2020-04-18:
update the firmware from 05.11.xxx to 05.12.01.00, this problem is existing still. 

_"The warning message "control_transfer returned error" reflects communication error with the device that is resolved by repeating the attempt. These messages are still under investigation but as long as they appear only now and then, say at the beginning of the run, they can safely be ignored for now"._

**yes, **roslaunch realsense2_camera rs_camera.launch** works well if the  "control_transfer returned error"  just  at the beginning of the run. 
but when the   **roslaunch realsense2_camera rs_camera.launch** works badly, meanwhile the refs output topics no data.  the error log as below is always fire without stopping**

 18/04 14:05:55,687 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/04 14:05:56,138 WARNING [547398607232] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 23, 
[ WARN] [1587189956.221320906]: Hardware Notification:IR stream start failure,1.58719e+12,Error,Hardware Error
[ERROR] [1587189956.221485596]: Hardware Reset is needed. use option: initial_reset:=true
 18/04 14:05:56,225 WARNING [547398607232] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 9, 
 18/04 14:05:57,158 ERROR [546509406592] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 18/04 14:05:57,296 ERROR [546534584704] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 18/04 14:05:57,390 WARNING [546509406592] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 130, error: Device or resource busy, number: 16
 18/04 14:05:57,398 WARNING [546534584704] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 132, error: Device or resource busy, number: 16
 18/04 14:05:57,398 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 14:05:57,407 WARNING [546509406592] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 130 error: Device or resource busy, number: 16
 18/04 14:05:57,409 ERROR [546509406592] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13
 18/04 14:05:57,419 WARNING [546509406592] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 130 error: Device or resource busy, number: 16
 18/04 14:05:57,419 ERROR [546509406592] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13
 18/04 14:05:57,434 WARNING [546257756544] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 131 error: Device or resource busy, number: 16
 18/04 14:05:57,434 ERROR [546257756544] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13
 18/04 14:05:57,436 WARNING [546534584704] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 132 error: Device or resource busy, number: 16
 18/04 14:05:57,436 ERROR [546534584704] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13
 18/04 14:05:57,450 WARNING [547037884800] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 134 error: Device or resource busy, number: 16
 18/04 14:05:57,450 ERROR [547037884800] (hid-device.cpp:153) failed to submit UVC request
 18/04 14:05:57,453 WARNING [547390214528] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 135 error: Device or resource busy, number: 16
 18/04 14:05:57,462 WARNING [546257756544] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 131 error: Device or resource busy, number: 16
 18/04 14:05:57,471 ERROR [546257756544] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13
 18/04 14:05:57,481 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 14:05:57,481 WARNING [546534584704] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 132 error: Device or resource busy, number: 16
 18/04 14:05:57,481 ERROR [546534584704] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13
 18/04 14:05:57,552 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 14:05:57,642 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 14:05:57,678 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 14:05:57,754 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 14:05:57,764 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 14:05:57,823 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 14:05:57,834 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 14:05:57,893 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
[... cut ...]
^C 18/04 14:05:58,509 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
 18/04 14:05:58,531 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16

> I apologise for the delay in responding to your question.
>
> The errors that you have been experiencing have also been experienced by other users. I will paste below some advice from the ROS wrapper's [...]

lightsalt2011, 2020-04-18:
reopen

MartyG-RealSense, 2020-04-18:
If you are able to plug the camera into a mains electricity powered USB 3 hub to improve USB stability then I recommend trying that, as it is one of the the most reliable ways to make USB related errors disappear.

If you do not have a hub and are unable to purchase one, then the best option may be to seek a response from the ROS wrapper's developer @doronhi

lightsalt2011, 2020-04-18:
thanks for ur quick reply. 

1. yes, powered usb3 hub improved the **roslaunch realsense2_camera rs_camera.launch** stability, I have verified.
2. why realsense_viewer and  rs_xx cmd works well without powered usb3 hub ?

MartyG-RealSense, 2020-04-18:
For some users it is the opposite situation (problems with RealSense Viewer but works fine with ROS), like with this case posted today:

https://github.com/IntelRealSense/librealsense/issues/6262

Again, @doronhi the ROS wrapper developer is the best person to comment on the workings of the wrapper.  :)

MartyG-RealSense, 2020-05-17:
Apologies that you did not receive a further response on this question.   Do you still require assistance please or is it okay to close the case?   Thanks!

doronhi, 2020-05-26:
Sorry for joining in so late. There has been some work on done on the libuvc backend and enhancing stability on the Jetson Nano is on the roadmap.
Could you upgrade both librealsense and realsense2_camera and update us with the errors you receive?
In any case, I think we should keep this issue open for a while until I manage further investigation on the Jetson.

MartyG-RealSense, 2020-07-18:
Hi @doronhi   Do you want to keep this case open for loner, please?   Thanks!

doronhi, 2020-07-19:
I didn't get to test on NVidia for a while. Although some work has been done on the RSUSB backend in librealsense2, I imagine the issue is still not fixed. I think we should keep this issue open until we manage to confirm that it is fixed, although, admittedly, this may take a little while.

MartyG-RealSense, 2020-07-19:
Okay, I will keep it open.  Thanks!

AndreV84, 2020-07-19:
what are the suggested steps to resolve the issue? What is the RSUSB?

zhangpeng66, 2020-09-03:
I have also encountered this problem，How to solve it？

AndreV84, 2020-09-03:
@zhangpeng66 which exactly issue do you experience?

zhangpeng66, 2020-09-03:
> @zhangpeng66 which exactly issue do you experience?
the issue：
18/04 14:05:58,139 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
18/04 14:05:58,208 WARNING [547390214528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16

zhangpeng66, 2020-09-03:
18/04 14:05:57,419 ERROR [546509406592] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13

AndreV84, 2020-09-03:
@zhangpeng66 does the issue exist if you run the binary withh sudo?
the device busy also could imply that there is another process accessing the device

MartyG-RealSense, 2020-09-13:
Adding a note to keep this case open for a further period of time.

MartyG-RealSense, 2020-10-08:
Continuing to keep this case open because of the request of doronhi to do so.

https://github.com/IntelRealSense/realsense-ros/issues/1171#issuecomment-660604570

MartyG-RealSense, 2020-10-30:
Hi @doronhi    Do you want this case to be kept open for a further period of time please?   Thanks!

https://github.com/IntelRealSense/realsense-ros/issues/1171#issuecomment-660604570

AndreV84, 2020-10-30:
It was a while since we tried.
but as far as I remember it slowly but worked somehow with ros wrapper on any jetson
on nano it just worked ore slowly than on other jetsons

MartyG-RealSense, 2020-11-07:
Adding a note to keep this case open for a further time period.

doronhi, 2020-11-08:
It's been a while and work was done in librealsense2 with NVidia. I still didn't get around to test it on NVidia so if everybody's ok with that I feel we can close this issue for now.

AndreV84, 2020-11-08:
@doronhi 
Hi. thank you for following uo.
I suggest to assume the issue is resolved, unless there will be compelling reasons to reopen it in the future.
Once I have acces to nano I will try to run the scenario again in irder to see if it works. However the nano might lack resources . so it might be better to use more powerful jetson device like NX or Jetson AGX Xavier.

MartyG-RealSense, 2020-11-08:
Thanks very much @doronhi and @AndreV84   I will close the issue as you are in agreement on this.   Thanks for the discussion!

closed 2020-11-08 by MartyG-RealSense