realsenseai/realsense-ros #1174: realsense-ros on windows

nawara72, 2020-04-20:
Hi 
I have been able to compile realsense-ros on windows. however when I run 
roslaunch realsense2_camera rs_camera.launch

I get the following error below. Would really appreciate any help to resolve it

Thanks

 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://127.0.0.1:11311
process[camera/realsense2_camera_manager-1]: started with pid [14892]
process[camera/realsense2_camera-2]: started with pid [6416]
[ INFO] [1587347732.252159300]: Initializing nodelet with 8 worker threads.
[ERROR] [1587347732.579273000]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Failed to load library C:/tmp/catkin_ws/install\lib/realsense2_camera.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = C:/tmp/catkin_ws/install\lib/realsense2_camera.dll)
[ERROR] [1587347732.588991900]: The error before refreshing the cache was: Failed to load library C:/tmp/catkin_ws/install\lib/realsense2_camera.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = C:/tmp/catkin_ws/install\lib/realsense2_camera.dll)
[FATAL] [1587347732.596705600]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[31m[camera/realsense2_camera-2] process has died [pid 6416, exit code -1, cmd C:\opt\ros\melodic\x64\lib\nodelet\nodelet.exe load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=C:\Users\cv\.ros\log\7b7ea1e1-82a6-11ea-b1b4-8c0f6f78be9a\camera-realsense2_camera-2.log].

nawara72, 2020-04-20:
Sorted it out, by just running devel\setup.bat
If I run install\setup.bat, it doesn't work

MartyG-RealSense, 2020-04-24:
Hi @nawara72   My research suggests that if you are compiling the ROS package from source, you should not have any Librealsense files in your catkin workspace, otherwise there may be a version conflict.

https://github.com/IntelRealSense/realsense-ros/issues/78#issuecomment-244260341

MartyG-RealSense, 2020-05-17:
Hi @nawara72    Do you require further assistance with this question please, or is it okay to close the case?   Thanks!

closed 2020-05-17 by nawara72

ccyinlu, 2021-07-09:
before compile and run `realsense2_camera`, you should first install the realsense sdk and paste the according dll file from the sdk to the ros devel directory.  
you can find `realsense2.dll` in your realsense SDK directory for example:  
`C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64`  
paste the `realsense2.dll` to the `ros_ws/devel` of your own and relaunch your ros code.