realsenseai/realsense-ros #1175: the version of realsense-ros doesn't match

Andrew5c, 2020-04-20:
1. After I installed the realsense followed by this guide [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md), 
And then I run the `git clone https://github.com/IntelRealSense/librealsense.git` to get the SDK,
2. then I run the demo of this [https://github.com/IntelRealSense/librealsense/tree/master/examples/cmake](https://github.com/IntelRealSense/librealsense/tree/master/examples/cmake), which will output the version of librealsense.  and what I got is `librealsense - 2.32.1` , which is **not the least** version of realsense!!  What troubles me is that **there is no a version of realsense-ros matched with my realsense(2.32.1)!**
3. Now I want to install realsense-ros, So, the question is what can I do to get a different librealsense SDK **so that I can install the realsense-ros?**. 
Thanks!!!

MartyG-RealSense, 2020-04-20:
According to the RealSense ROS developer @doronhi, a debian version of SDK 2.33.1 became available a few weeks ago.  

https://github.com/IntelRealSense/realsense-ros/issues/1145#issuecomment-607052295

He said though that 2.32.1 "should suffice for building realsense2_camera version **2.2.13**".

https://github.com/IntelRealSense/realsense-ros/issues/1114#issuecomment-596344620

Andrew5c, 2020-04-20:
ok, thanks for your reminding, i will give a try.
and I am installing the realsense-ros follow this guide [https://github.com/IntelRealSense/realsense-ros](https://github.com/IntelRealSense/realsense-ros). 
hope librealsense SDK 2.32.1 is suffice! 
thanks.

closed 2020-05-17 by MartyG-RealSense