realsenseai/realsense-ros #1176: How to change Realsense D435i IMU orientation in ROS?

KrabasChuliganas, 2020-04-23:
Hi, is there any solution how to change camera D435i IMU orientation in ROS? I need that because i get on linear_acceleration: Y:  -9.845, while it needs to be on Z. Thus, it doesn't work properly on robot_localization package. Any info would help.

ashirsat, 2020-04-23:
Hey,

you can do static transform from the camera frame to the one you want to
get it in . The way the orientation is decided is dependent upon the actual
orientation in the device.

-- 
Regards,
Aniket Shirsat

Mechanical Graduate
Arizona State University| Ira A. Fulton Schools of Engineering
School for Engineering of Matter Transport and Energy
Cell:(480)371-8843
Portfolio <http://seelio.com/aniketshirsat>
<http://www.linkedin.com/in/shirsataniket>
<https://twitter.com/intent/follow?original_referer=https%3A%2F%2Fabout.twitter.com%2Fresources%2Fbuttons&region=follow_link&screen_name=AniketRShirsat&tw_p=followbutton>

KrabasChuliganas, 2020-04-23:
you mean like this? Tried, it doesn't affect message.

<node pkg="tf"
        type="static_transform_publisher"
        name="camera_imu_broadcaster"
        args="0.0 0.0 0.0 0.0 3.14 0.0 camera_link /camera/imu_combined 40" />

KrabasChuliganas, 2020-04-23:
well, I found solution. In base_realsense_node.cpp file i change axis mapping. I know that's wrong way to do it, but at least it works. :)

`  1216 - 1222 lines.
    imu_msg.angular_velocity.x = gyro_data.m_reading.z;
    imu_msg.angular_velocity.y = gyro_data.m_reading.x;
    imu_msg.angular_velocity.z = -gyro_data.m_reading.y;

    imu_msg.linear_acceleration.x = accel_data.m_reading.z;
    imu_msg.linear_acceleration.y = accel_data.m_reading.x;
    imu_msg.linear_acceleration.z = -accel_data.m_reading.y;`

` From 1229-1241 lines
  if (GYRO == stream_index)
    {
        imu_msg.angular_velocity.x = imu_data.m_reading.z;
        imu_msg.angular_velocity.y = imu_data.m_reading.x;
        imu_msg.angular_velocity.z = -imu_data.m_reading.y;
    }
    else if (ACCEL == stream_index)
    {
        imu_msg.linear_acceleration.x = imu_data.m_reading.z;
        imu_msg.linear_acceleration.y = imu_data.m_reading.x;
        imu_msg.linear_acceleration.z = -imu_data.m_reading.y;
    }
    return imu_data.m_time;`

[image]

closed 2020-04-23 by KrabasChuliganas

EhrazImam, 2021-12-07:
Hi @KrabasChuliganas same issue i followed your steps but that doesn't help me