realsenseai/realsense-ros #1177: TX2 Computation and USB Questions

seanxu112, 2020-04-24:
Hello,

I am a big fan of the realsense cameras that I have received. Currently trying to run D435 and T265 on TX2. While there are a lot of other posts about this. I still have a few more questions. In librealsense, I have set BUILD_WITH_TM2=ON, BUILD_WITH_CUDA=ON, and FORCE_RSUSB_BACKEND=ON. SDK version 2.33.1 and realsense-ros 2.2.13. When I only use D435, the camera works ok, only has these on start: `Hardware Notification:IR stream start failure,1.58769e+12,Error,Hardware Error
(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
`
When I tried to connect both D435 and T265 with an unpowered usb3.0 port, the error above shows up consistently with usb scp overflow issue. I wonder if a powered usb port would potentially solve this problem? My Kernel is:4.4.38-tegra

Also, what is the expected cpu usage with D435 if I built with CUDA. Seems like on TX2 it is still using more than one core when I set align_depth=true. 

Sincerely,
Sean

RealSenseCustomerSupport, 2020-04-28:
----------------------------------------------

@seanxu112 Could you please share what launch file you used to run D435 and T265 when get the issue? Thanks!

seanxu112, 2020-04-29:
Just the default rs_d400_with_t265.launch

RealSenseCustomerSupport, 2020-05-15:
----------------------------------------------

@seanxu112 Any problem if you only start T265 with demo_t265.lauch?
https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/demo_t265.launch

RealSenseCustomerSupport, 2020-06-03:
----------------------------------------------

@seanxu112 Could you please update if you see any problem when start T265 with demo_t265.launch? Thanks!

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/demo_t265.launch

RealSenseCustomerSupport, 2020-06-22:
----------------------------------------------

@seanxu112 Could you please update? Please note that the issue will be closed if we don't get your response in 7 days. Thanks!

seanxu112, 2020-06-23:
Running t265 by itself works then. However, our whole system changed a lot since the day of posting, switching to 2.23.0 sdk, and 2.2.6 ros driver works now on tx2. Is there a new release of sdk and ros driver that works on tx2?

RealSenseSupport, 2020-06-30:
@seanxu112 The latest ROS wrapper 2.2.14 and SDK 2.35.2 are published. Could you please have a try? Thanks!

RealSenseSupport, 2020-07-27:
@seanxu112 Did you get chance to try the latest ROS wrapper to see if you still have issue? Thanks!

RealSenseSupport, 2020-08-13:
@seanxu112 Could you please update if you still support on this? Thanks!

darshb34, 2020-08-13:
Hello All,

I am using one t265 and 2 d430s and 2 d435i for my project on jetson tx2 as well same with jetson xavier.

The command `sudo apt-get install ros-melodic-realsense2_camera` installs latest ros wrapper as well as the latest compatible librealsense2.

My question is, _Does it have these libraries with CUDA enabled ? If yes, how can I test if it is really using CUDA?_

Thank you

RealSenseSupport, 2020-08-14:
@darshb34 According to https://github.com/IntelRealSense/librealsense/issues/6302#issuecomment-619531064 Debian packages are already built with CUDA. 

Also you can refer to https://github.com/IntelRealSense/librealsense/issues/6302#issuecomment-619516818 to check CUDA if enabled or not

doronhi, 2020-08-14:
The debian packages stored in the ros-distro are NOT built with CUDA support. remove "ros-melodic-librealsense2" and install librealsense2 according to the installation guide in the [librealsense2 repository](https://github.com/IntelRealSense/librealsense)

RealSenseSupport, 2020-08-17:
@doronhi Thanks for the clarification! 
@darshb34 Please refer to doronhi's comment to set BUILD_WITH_CUDA=true to build with CUDA. Thanks!

doronhi, 2020-08-24:
If I understand you correctly then the answer is yes, the connection between librealsens2 version and realsense2_version will break when realsense2_camera gets updated.
The new version of realsense2_camera will require the latest version of librealsense2.
In librealsense2's installation guide are instructions for the installation of librealsense2-dev debian package from intel's s3 (amazon) repository. I think it is compiled with CUDA and may suite your needs.
You can, alternatively, build from source using the correct building flags.
Whatever methos you use, you will need, to make sure the librealsense2 version matches the realsense2_camera version you have every time the later is being updated.

RealSenseSupport, 2020-09-02:
@seanxu112 @darshb34 Any other questions about this with Doronhi's reply? Looking forward to your update. Thanks!

RealSenseSupport, 2020-09-09:
@seanxu112 @darshb34 Could you please update？Thanks!

darshb34, 2020-09-11:
> The debian packages stored in the ros-distro are NOT built with CUDA support. remove "ros-melodic-librealsense2" and install librealsense2 according to the installation guide in the [librealsense2 repository](https://github.com/IntelRealSense/librealsense)

Hi @RealSenseSupport @dorodnic 

I cannot remove "ros-melodic-librealsense2" without having "ros-melodic-realsense2-camera" removed automatically with it as well. I tried apt-mark hold to hold the "ros-melodic-realsense2-camera" but that made removing "ros-melodic-librealsense2" fail.

Thanks you

RealSenseSupport, 2020-09-14:
@darshb34 You may refer to method 2 to install RealSense ROS wrapper: https://github.com/IntelRealSense/realsense-ros#step-1-install-the-latest-intel-realsense-sdk-20

RealSenseSupport, 2020-09-21:
@darshb34 Any update from your side? Looking forward to your reply. Thanks!

RealSenseSupport, 2020-09-27:
@seanxu112  @darshb34  Any other questions about this? Thanks!

RealSenseSupport, 2020-10-10:
@seanxu112 @darshb34 If no other questions about this, will close it accordingly. Thanks!

closed 2020-10-12 by RealSenseSupport

maxiuw, 2022-11-08:
Two solutions:
1. The best one - Usb 3.0 cable connected to USB 3.0 port
2. Drastically reduce the frame rate, as mentioned in #1751 if you reduce the framerate it will only work if you also define width and height, otherwise, fps won't be changed and you will keep on getting the error. i.e. 
`roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=5`