realsenseai/realsense-ros #1178: Intel Realsense: Time is out of Dual 32-bit range

JacobsDaan, 2020-04-24:
We are trying to get the Intel Realsense D435i to work on our Raspberry Pi with the Raspbian OS and ROS Melodic.

After we configured our Raspberry Pi with Raspbian and installed ROS Melodic on it, we installed the realsense-ros package on our Raspberry Pi. When we connect our Realsense camera to the Raspberry and run the following command:

$ roslaunch realsense2_camera rs_camera.launch

We get the following error:
[image]

We get some ros topics from the camera but they don't publish any data. Also not all of the topics are displayed. When we plug the camera in to a pc it works fine.

We already googled it and tried the following:

full clean and rebuild of catkin workspace
update and upgrade of all ros packages
delete ros_comm package
But non off the above worked for us.

Is there anybody who has any ideas?

Thanks in advance!

MartyG-RealSense, 2020-04-24:
RealSense ROS wrapper @doronhi has investigated cases of this error in the past and thought it was likely that it was unrelated to Librealsense and more to do with ROS.

There was a case on the internet where the error was resolved by completely removing ROS (not just updating it) and then reinstalling it.

Zarbokk, 2020-05-15:
Did you make it work? currently hanging at exactly the same problem. Maybe you found something?

YoshuaNava, 2020-05-19:
@Zarbokk I would suggest that you check that the time on our computer is up to date. Furthermore, make sure that it is not going backwards due to some network time synchronization tool running in the background.

MartyG-RealSense, 2020-05-30:
I will close this case as it is apparently not a Librealsense-caused problem.  This post will remain findable by web search for other who may be having the same issue.

closed 2020-05-30 by MartyG-RealSense

Zarbokk, 2020-06-05:
My solution in the end was to use a different OS away from Buster to RPI4-Ubuntu-18.04.
Together with a stable power connection the camera works.
Thanks

MartyG-RealSense, 2020-06-05:
@Zarbokk   I'm pleased that you found a solution that works for you - thanks for the update!