realsenseai/realsense-ros #1179: which camera that Center point is corresponding with

frinklai, 2020-04-24:
As mentioned in title, which camera that Center point is corresponding with
When I launch rs_rgbd.lunch
I found it seems that the center point of output frame is start from the **Left imager** rather than the **rgb module**.
Is it right?

[image]

[image]

MartyG-RealSense, 2020-04-24:
Yes, the camera uses the center of the left imager as its origin point.

The dynamic calibration guide for the 400 Series says: "Assume left camera is the reference camera and is located at world origin".

The data sheet document for the 400 Series adds: "The left and right imagers capture the scene and sends imager data to the depth imaging processor, which calculates depth values for each pixel in the image by correlating points on the left image to the right image and via shift between a point on the Left image and the Right image".

frinklai, 2020-04-25:
Ok, I got it, thanks for your quick reply.
By the way, would you tell me the difference between rs_camera.launch and rs_rgbd.launch ?
Thank you very much.

MartyG-RealSense, 2020-04-25:
I believe that **rs_rgbd.launch** is used when you want to obtain a point cloud that has depth aligned to color.

https://github.com/IntelRealSense/realsense-ros/issues/277#issuecomment-363117777

frinklai, 2020-04-25:
Thanks for your quick reply.
In fact, I'm not sure the difference for following command?
1. `roslaunch realsense2_camera rs_camera.launch align_depth:=true`
2. `roslaunch realsense2_camera rs_rgbd.launch`

In my application, I want to get XYZ coordinate in **meter** .
So far, my method are as follow:
1. launch `rs_rgbd.launch`
2. Collect RGB and depth msg from `/camera/color/image_raw` and `/camera/aligned_depth_to_color/image_raw`
3. Use the function below to transfer pixel to meter
```
def deproject_pixel2point(self, pixel, depth):
        depth_ = depth
        pixel_x = pixel[0]
        pixel_y = pixel[1]
        intrin_fx  = 612.9651489257812
        intrin_fy  = 613.2086791992188
        intrin_ppx = 324.2138671875
        intrin_ppy = 236.86329650878906
        coeffs = [0, 0, 0, 0, 0]
    
        tmp_x = (pixel_x - intrin_ppx)/intrin_fx
        tmp_y = (pixel_y - intrin_ppy)/intrin_fy

        r2 = (tmp_x*tmp_x) + (tmp_y*tmp_y)
        f = 1 + coeffs[0]*r2 + coeffs[1]*r2*r2 + coeffs[4]*r2*r2*r2
        tmp_x  = tmp_x*f 
        tmp_y  = tmp_y*f

        point_x = depth_*tmp_x
        point_y = depth_*tmp_y
        depth_point = [point_x/1000, point_y/1000, depth_/1000]
        print('real xyz (m) = ', depth_point)
        return depth_point
```
Is my method is right?
Should I change to use `roslaunch realsense2_camera rs_camera.launch align_depth:=true` or there are any better suggestion?
Thanks again for your patient reply.

MartyG-RealSense, 2020-04-25:
According to the documentation for the RealSense ROS wrapper, using **roslaunch realsense2_camera rs_camera.launch align_depth:=true** publishes the aligned depth stream to other available streams such as color or infra-red.  So it would seem to not have much to do with getting the XYZ coordinates.

https://github.com/IntelRealSense/realsense-ros#aligned-depth-frames

I wonder if the script in the link below for returning X, Y and Z in meters may be a useful reference for you:

https://github.com/IntelRealSense/realsense-ros/issues/551#issuecomment-489873418

frinklai, 2020-04-25:
Ok, thanks for your detail information.
So it seems that 
`roslaunch realsense2_camera rs_camera.launch align_depth:=true` 
is equal to 
`roslaunch realsense2_camera rs_rgbd.launch` 
is that right ?

MartyG-RealSense, 2020-04-25:
The way that I interpret the ROS wrapper documentation's description of **roslaunch realsense2_camera rs_camera.launch align_depth:=true** is that adding the term **align_depth:=true** to the launch instruction gives you the option to align the depth stream and then view the depth in RViz aligned to color (rgb), infra1 (left infrared) or infra2 (right infrared).

I would think that if align_depth:=true is not added to the launch instruction then the topics published by **rs_camera.launch** would not be aligned to depth. 

**rs_rgbd.launch** seems to me to be something different from **rs_camera.launch**, and used when you want to generate a 3D point cloud instead of a 2D image.

I apologise for not speaking in certainties.  ROS is not currently one of my specialist subjects (I am still learning it).

frinklai, 2020-04-25:
Ok, I got it, thank you so much for your detail and patient reply.  :)

MartyG-RealSense, 2020-04-25:
You are very welcome.  Good luck with your project!   :)

MartyG-RealSense, 2020-05-24:
Case closed due to no further comments received.   Thanks for your question!

closed 2020-05-24 by MartyG-RealSense

EvertonTrentoJR, 2022-10-04:
And after the alingment, the center continue in the left imager, or its transfered to the center of the camera itself?

MartyG-RealSense, 2022-10-04:
Hi @EvertonTrentoJR  The origin of the coordinate system of the stream that was aligned to will be used.  So if depth to color alignment is performed, the color coordinate system will become the origin instead of the center-line of the left imager.

EvertonTrentoJR, 2022-10-04:
Hi @MartyG-RealSense thanks for your reply. and the color reference is the centerline of the color camera?

EvertonTrentoJR, 2022-10-04:
Ok. thank you again for your reply.