realsenseai/realsense-ros #1180: rosbag_filename does not work

HMellor, 2020-04-24:
Camera Model: D435i x 1 via usb3.2
Firmware: 05.12.03.00
Operating system: Ubuntu 18.04.4
Platform: x86_64 desktop intel i7
Language: ROS/realsense-viewer
SDK: 2.34.0
ROS distro: Melodic

When running `roslaunch realsense2_camera rs_from_file.launch rosbag_filename:="/path/to/bag_files/test1.bag"` I get the error in the log below. realsense2_camera appears to be opening the bag file and enabling the streams just fine, but it crashes just before the `RealSense Node Is Up!` message and I don't know what to make of the error it gives, has there been a change in the way realsense2_camera works recently?

```
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 0
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 0
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 0
 * /camera/realsense2_camera/depth_fps: 0
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 0
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 0
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 0
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 0
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 0
 * /camera/realsense2_camera/gyro_fps: 0
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 0
[... 58 lines cut ...]
[ INFO] [1587746416.688544810]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1587746416.688557070]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1587746416.688582375]: num_filters: 0
[ INFO] [1587746416.688596321]: Setting Dynamic reconfig parameters.
[ INFO] [1587746416.692082282]: Done Setting Dynamic reconfig parameters.
[ WARN] [1587746416.692266905]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Width: 0, Height: 0, FPS: 0
[ INFO] [1587746416.692391775]: infra1 stream is enabled - width: 1280, height: 800, fps: 30, Format: Y8
[ INFO] [1587746416.692467027]: infra2 stream is enabled - width: 1280, height: 800, fps: 30, Format: Y8
[ WARN] [1587746416.692483655]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: 0, Height: 0, FPS: 0
[ INFO] [1587746416.692505950]: setupPublishers...
[ INFO] [1587746416.694399517]: Expected frequency for infra1 = 30.00000
[ INFO] [1587746416.717309431]: Expected frequency for infra2 = 30.00000
[ INFO] [1587746416.733435791]: setupStreams...
[ INFO] [1587746416.733505969]: insert Infrared to Stereo Module
[ INFO] [1587746416.733546468]: insert Infrared to Stereo Module
[ INFO] [1587746416.733578710]: insert Gyro to Motion Module
[ INFO] [1587746416.733596234]: insert Accel to Motion Module
[ INFO] [1587746416.733923841]: SELECTED BASE:Depth, 0
double free or corruption (!prev)
[FATAL] [1587746416.851444972]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-2] process has died [pid 17861, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/user/.ros/log/467f9132-864a-11ea-810a-2c9464002edb/camera-realsense2_camera_manager-2.log].
log file: /home/user/.ros/log/467f9132-864a-11ea-810a-2c9464002edb/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has died [pid 17862, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/user/.ros/log/467f9132-864a-11ea-810a-2c9464002edb/camera-realsense2_camera-3.log].
log file: /home/user/.ros/log/467f9132-864a-11ea-810a-2c9464002edb/camera-realsense2_camera-3*.log
```

MartyG-RealSense, 2020-04-24:
There was a previous case with this error when using the Librealsense ROS wrapper.  In that case, the solution that worked was to delete the workspace build folder and then build again.  

https://github.com/IntelRealSense/realsense-ros/issues/745

HMellor, 2020-04-24:
I am fairly confident that the "does not follow the naming conventions" warning is not the issue.

Everything works just fine when connecting to the camera itself using `roslaunch realsense2_camera rs_camera.launch`.

The error only appears when I specify the `rosbag_filename` argument in my launch file.

MartyG-RealSense, 2020-04-24:
Another RealSense user launching a bag in the same way and getting the Fatal error near the bottom of the log:

[FATAL] [1587746416.851444972]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'

Found that they needed to provide the entire path of the filename to make it work.

https://github.com/IntelRealSense/realsense-ros/issues/769

HMellor, 2020-04-24:
Thank you for another suggestion, but I am already using the full path (i.e. `/home/user/path/to/file.bag`).

As you can see from these lines:
```
[ INFO] [1587746416.596974312]: publish topics from rosbag file: /path/to/bag_files/test1.bag
[ INFO] [1587746416.651457946]: getParameters...
[ INFO] [1587746416.688227008]: setupDevice...
[ INFO] [1587746416.688260335]: JSON file is not provided
[ INFO] [1587746416.688274009]: ROS Node Namespace: camera
[ INFO] [1587746416.688288988]: Device Name: Intel RealSense D435I
[ INFO] [1587746416.688303138]: Device Serial No: 844212071905
[ INFO] [1587746416.688315825]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/video4linux/video0
[ INFO] [1587746416.688327507]: Device FW version: 05.12.03.00
[ INFO] [1587746416.688342408]: Device Product ID: 0x0B3A
```
The bag file is being opened by `realsense2_camera`.

MartyG-RealSense, 2020-04-24:
"publish topics from rosbag file: /path/to/bag_files/test1.bag" suggests to me that ROS is just confirming the location to look for the bag to publish topics from, rather than actually confirming that it has successfully connected to a bag file at that location.

Apologies if this is a silly question, but do you actually have the folder structure **Path > to > bag_files** set up, with a test1.bag file inside it?  The described path sounds like one written in a tutorial to teach readers how to describe the bag-file path in a launch statement.

HMellor, 2020-04-24:
Yes, I had just replaced the paths to make them shorter. I have edited my previous comments to make this more clear. 

Although, given that librealsense is able to find a `Device Name`, `Device Serial No`, etc. my previous comment does confirm that it has successfully connected to the bag file.

Not to mention that there aren't actually any tutorials on how to do this. Just a mention of the `rosbag_filename` argument in the README, which isn't actually in `rs_camera.launch`, you have to add it yourself.

MartyG-RealSense, 2020-04-24:
I researched the problem deeply but could not find further helpful research leads.  So I will tag in @doronhi the RealSense ROS developer to seek his expert input.  I apologise for the delay this will cause in the meantime.

HMellor, 2020-04-24:
I appreciate your help, thank you for tagging a developer. Hopefully we can figure this out.

HMellor, 2020-04-30:
@MartyG-RealSense @doronhi Is there any update on this? Any extras information you need?

MartyG-RealSense, 2020-04-30:
I researched this case from the beginning today to try to find new leads to investigate but came up with nothing, sadly.  I will refer the case to the Intel support team to seek further advice.

RealSenseSupport, 2020-05-06:
Hi @HMellor,

Please try to open test1.bag file using Intel.RealSense.Viewer and let it play until the end of the file.
Does Intel.RealSense.Viewer have an issue to play it/do you see any abnormal behavior?

Thank you!

HMellor, 2020-05-07:
`realsense-viewer` is able to playback the recorded file, but in the output console it is printing this multiple times per frame (i.e. 4 in 2ms and then a gap of ~33ms):
```
 07/05 09:27:11,472 INFO [140577741981440] (ros_file_format.h:87) Failed to convert source: 8UC1 to matching rs2_format
```

RealSenseSupport, 2020-05-07:
@HMellor,

Thank you for the provided information. Please download one of the bag files from here:
https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

Open it using Intel.RealSense.Viewer and let it play until the end of the file.If you don't see the "Failed to convert source" message or any other abnormal behavior,open that file using ROS Wrapper. Please let us know the results of your findings.

RealSenseSupport, 2020-05-14:
@HMellor 

Do you still need help with this question?

HMellor, 2020-05-14:
Sorry for the delay. With the RealSense product not working priorities have shifted. 

I just replayed one of the sample files recorded on the D435i using RealSense Viewer and I get the same error but for `mono16` and `rgb8` instead of `8UC1`

RealSenseSupport, 2020-05-18:
@HMellor

Please share with us the LibRealSense log files (Debug log severity) for both samples from [D435i sample data](http://realsense-hw-public.s3.amazonaws.com/rs-tests/TestData/d435i_sample_data.zip).

Thank you!

doronhi, 2020-05-25:
It seems that your file "test1.bag" does not contain depth information. It contains infra1 and infra2 only. realsense2_camera node expect a depth sensor, even if it is not enabled, so it can compute and publish transformations between sensors.

closed 2020-06-12 by RealSenseSupport