realsenseai/realsense-ros #1181: Subscribe PointCloud2 topic gives error

brsbilgin, 2020-04-25:
When I start to subscribe  /camera/depth/color/points topic it starts give error on bellow. When I stopped read topic it stops to give error.

25/04 22:18:16,033 ERROR [546022039936] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 25/04 22:18:16,100 WARNING [546022039936] (types.cpp:49) Out of frame resources!
 25/04 22:18:16,100 ERROR [546022039936] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 25/04 22:18:16,133 WARNING [546022039936] (types.cpp:49) Out of frame resources!
 25/04 22:18:16,133 ERROR [546022039936] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 25/04 22:18:16,399 WARNING [546022039936] (types.cpp:49) Out of frame resources!
 25/04 22:18:16,400 ERROR [546022039936] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 25/04 22:18:16,433 WARNING [546022039936] (types.cpp:49) Out of frame resources!
 25/04 22:18:16,433 ERROR [546022039936] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 25/04 22:18:16,500 WARNING [546022039936] (types.cpp:49) Out of frame resources

MartyG-RealSense, 2020-04-26:
Hi @brsbilgin  It looks like this case may be related to an issue with synthetic stream errors that does not have a clear solution at the time of writing this.

https://github.com/IntelRealSense/realsense-ros/issues/1161

I will quote some advice from that discussion that was provided by @doronhi the RealSense ROS wrapper developer:

"The "Out of frame resources!" message and the "Exception...processing callback!" could be related to ... lack of system resources - the app can't keep up with the incoming flow of images.  For that you can reduce frame rate and resolution.  Building with CUDA support should really help in that case on NVidia machines".

A RealSense user gave advice on testing whether reducing frame rate would help by adjusting the FPS in the launch file:

https://github.com/IntelRealSense/realsense-ros/issues/1037#issuecomment-573976862

brsbilgin, 2020-05-03:
Hi @MartyG-RealSense . I will check the #1161 .
The realsense is already build with CUDA support. The fps is already decreased to 15.

MartyG-RealSense, 2020-05-03:
@brsbilgin  Okay, thank you very much for the update.   I look forward to hearing from you again after you have read https://github.com/IntelRealSense/realsense-ros/issues/1161

jorgemia, 2020-05-05:
I also get the following when launching with pointcloud on, on a Jetson Nano. I first get the control transfer error when launching the node (I know the devs are aware of this) and then when I visualize the pointcloud in rviz I start getting the callback error (which I imagine has to do with the framerate/resolution). **This callback error, however, doesn't happen all the time, could it be affected by whatever there is in the scene?**

Also, if I kill the node using ctrl+c and then try and launch it again, I start getting uvc device warnings and Hardware Errors. (See second code snippet)

```
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
[... 275 lines cut ...]
[ WARN] [1588669573.802277878]: Hardware Notification:IR stream start failure,1.58867e+12,Error,Hardware Error
 05/05 11:06:14,341 WARNING [547631329664] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 22, 
 05/05 11:06:14,361 WARNING [547465716096] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/05 11:06:14,495 WARNING [547465716096] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 05/05 11:06:14,556 WARNING [547465716096] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/05 11:06:14,780 WARNING [547465716096] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1588669574.823336777]: Hardware Notification:Depth stream start failure,1.58867e+12,Error,Hardware Error
 05/05 11:06:15,301 WARNING [547631329664] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 12, 
 05/05 11:06:15,570 WARNING [547631329664] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 23, 
[... cut ...]
 05/05 11:06:18,603 ERROR [546534584704] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 05/05 11:06:18,670 WARNING [546534584704] (types.cpp:49) Out of frame resources!
 05/05 11:06:18,670 ERROR [546534584704] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 05/05 11:06:18,900 WARNING [546534584704] (types.cpp:49) Out of frame resources!
 05/05 11:06:18,901 ERROR [546534584704] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 05/05 11:06:18,968 WARNING [546534584704] (types.cpp:49) Out of frame resources!

```

MartyG-RealSense, 2020-05-05:
@jorgemia   The link below has advice from @doronhi the RealSense ROs wrapper developer about the set of errors that you are receiving.  Like you are using the Nano, the user in that case similarly had a low-power single board computer (the Up Board).

https://github.com/IntelRealSense/realsense-ros/issues/715#issuecomment-479855525

MartyG-RealSense, 2020-05-17:
@jorgemia    Do you still require further assistance with the errors that you were experiencing, please?

jorgemia, 2020-05-17:
@MartyG-RealSense just waiting for the errors to be fixed.

MartyG-RealSense, 2020-05-17:
You could try disabling the **infra2** channel to save some processing, by using **enable_infra2:=false**

It saves on USB transportation and kernel work, according to Doronhi the RealSense ROS wrapper developer.  Doronhi adds "It will have no effect on the depth quality.   It only disables the infra2 images' transmission via the USB port.  They are still taken and the depth is calculated using them inside the device".

Regarding the out of frame resources errors:  Doronhi has also said:

 "The "Out of frame resources!" message and the "Exception...processing callback!" could be related to ... lack of system resources - the app can't keep up with the incoming flow of images.  For that you can reduce frame rate and resolution.  Building with CUDA support should really help in that case on NVidia machines".

FPS speed could be modified in the launch file:

https://github.com/IntelRealSense/realsense-ros/issues/1037#issuecomment-573976862

If you are able to successfully generate a point cloud but are continuously generating errors, a "dirty" workaround may be to disable on-screen error logging in ROS.

https://github.com/IntelRealSense/librealsense/issues/6347#issuecomment-625149967

doronhi, 2020-05-18:
Regarding the stream start failure on the second run, I encountered something similar and fixed it in #1202 . I didn't manage to check it on Jetson Nano yet. Will be happy to know if it fixed it for you.

jorgemia, 2020-05-18:
@doronhi yes that PR fixed the issue I used to encounter on a second run! Thanks! FYI though, I get the follow libusb warning when shutting down the node now:

```
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
 18/05 10:27:38,984 WARNING [546919031168] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 130 error: Success, number: 0
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

MartyG-RealSense, 2020-06-05:
Case closed due to no further comments received.  @jorgemia  please create a new issue if you are still receiving a libusb warning and require support for that issue.

closed 2020-06-05 by MartyG-RealSense