realsenseai/realsense-ros #1182: Raspberry Pi Buster + D435i results in multiple error on USB bus and restarts the camera atleasts once

echoGee, 2020-04-25:
- ROS Melodic(1.14.5) (compiled from source on Buster)
- RealSense ROS v2.2.13 (compiled from source on Buster)
- LibRealSense v2.34.0 (compiled from source on Buster)

I have run the realsense-viewer which runs fine.

Running 
> roslaunch realsense2_camera demo_pointcloud.launch 

results in rviz crashing. 

Console log:
[demo_launch_25Apr2020.txt](https://github.com/IntelRealSense/realsense-ros/files/4534095/demo_launch_25Apr2020.txt)

So removed rviz from the launch file and ran
> roslaunch realsense2_camera demo_withoutRVIZ_pointcloud.launch 
pointcloud_texture_stream:=RS2_STREAM_COLOR

Console log:
[demo_launch_withoutRviz_25Apr2020.txt](https://github.com/IntelRealSense/realsense-ros/files/4534105/demo_launch_withoutRviz_25Apr2020.txt)

This also results in multiple bus errors.
The images and depth map(Non-colored) topic is published.

MartyG-RealSense, 2020-04-26:
I have researched your case very carefully.  I found a case with a similar setup to yours (Rpi4, Buster, ROS Melodic, SDK 2.34.0) in which the user in that cases was experiencing similar errors to the ones that you did when using RViz.

https://github.com/IntelRealSense/realsense-ros/issues/1149

There is currently no solution available on that linked-to case though.

Could you please try the point cloud demo in the ROS wrapper instructions that uses this launch instruction:

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**

https://github.com/IntelRealSense/realsense-ros#rgbd-point-cloud

Does this launch setting enable you to see a point cloud once you open RViz please?

echoGee, 2020-04-27:
Tried **roslaunch realsense2_camera rs_camera.launch filters:=pointcloud** . But does not output color pixels.

MartyG-RealSense, 2020-04-27:
I found a discussion that sounds like what you are trying to achieve

Please read downward through the comments from the point in the discussion that I have linked to.

https://github.com/IntelRealSense/realsense-ros/issues/277#issuecomment-363117777

If this link is not what you are referring to when you say that it "does not output color pixels", please provide more information about what you are trying to achieve.  Thanks!

echoGee, 2020-04-30:
The particular issue discusses about running 
`roslaunch realsense2_camera rs_rgbd.launch`, I run into this error

> Resource not found: rgbd_launch
> ROS path [0]=/opt/ros/melodic/share/ros
> ROS path [1]=/home/pi/catkin_ws/src
> ROS path [2]=/opt/ros/melodic/share
> The traceback for the exception was written to the log file
> 

Is there a documentation on each of the launch files in this package or which ones to run. 
https://github.com/IntelRealSense/realsense-ros#rgbd-point-cloud shows rs_camera.launch while https://github.com/IntelRealSense/realsense-ros/tree/2.0.2#rgbd-point-cloud shows rs_rgbd.launch

MartyG-RealSense, 2020-04-30:
I researched your question but could not find a reference about each of the launch files.

I did find a case though where someone else had the'resource not found' error when using **roslaunch realsense2_camera rs_rgbd.launch** and fixed it by setting RViz “Fixed Frame” to **camera_color_optical_frame** or **camera link**.

echoGee, 2020-04-30:
Not sure if the previous issue is related. The roslaunch crashes even before the rviz opens up. The log is included for the launch below:

> [roslaunch][INFO] 2020-04-30 08:23:57,090: Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> [roslaunch][INFO] 2020-04-30 08:23:57,106: Done checking log file disk usage. Usage is <1GB.
> [roslaunch][INFO] 2020-04-30 08:23:57,107: roslaunch starting with args ['/opt/ros/melodic/bin/roslaunch', 'realsense2_camera', 'rs_rgbd.launch']
> [roslaunch][INFO] 2020-04-30 08:23:57,107: roslaunch env is {'SSH_CLIENT': '192.168.6.4 54858 22', 'ROS_DISTRO': 'melodic', 'XDG_SESSION_TYPE': 'tty', 'ROS_IP': '192.168.6.8', 'LOGNAME': 'pi', 'USER': 'pi', 'PATH': '/opt/ros/melodic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games', 'HOME': '/home/pi', 'CMAKE_PREFIX_PATH': '/home/pi/catkin_ws/devel:/opt/ros/melodic', 'LD_LIBRARY_PATH': '/home/pi/catkin_ws/devel/lib:/opt/ros/melodic/lib:/usr/local/lib:', 'LANG': 'en_GB.UTF-8', 'TERM': 'xterm-256color', 'SHELL': '/bin/bash', 'SHLVL': '1', 'ROS_LOG_FILENAME': '/home/pi/.ros/log/76931ae6-8add-11ea-bc27-dca63263de10/roslaunch-raspberrypi4-20305.log', 'TEXTDOMAIN': 'Linux-PAM', 'ROS_ETC_DIR': '/opt/ros/melodic/etc/ros', 'XDG_SESSION_CLASS': 'user', 'XDG_RUNTIME_DIR': '/run/user/1000', 'PYTHONPATH': '/home/pi/catkin_ws/devel/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages::/usr/local/lib', 'ROS_ROOT': '/opt/ros/melodic/share/ros', 'PKG_CONFIG_PATH': '/home/pi/catkin_ws/devel/lib/pkgconfig:/opt/ros/melodic/lib/pkgconfig', 'ROS_PACKAGE_PATH': '/home/pi/catkin_ws/src:/opt/ros/melodic/share', 'XDG_SESSION_ID': 'c4', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', '_': '/opt/ros/melodic/bin/roslaunch', 'SSH_CONNECTION': '192.168.6.4 54858 192.168.6.8 22', 'SSH_TTY': '/dev/pts/1', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/pi/catkin_ws/devel/share/common-lisp', 'PWD': '/home/pi', 'ROS_MASTER_URI': 'http://192.168.6.8:11311', 'ROS_PYTHON_VERSION': '2', 'ROS_VERSION': '1', 'MAIL': '/var/mail/pi', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:'}
> [roslaunch][INFO] 2020-04-30 08:23:57,107: starting in server mode
> [roslaunch.parent][INFO] 2020-04-30 08:23:57,108: starting roslaunch parent run
> [roslaunch][INFO] 2020-04-30 08:23:57,108: loading roscore config file /opt/ros/melodic/etc/ros/roscore.xml
> [roslaunch][INFO] 2020-04-30 08:23:58,414: Added core node of type [rosout/rosout] in namespace [/]
> [roslaunch.config][INFO] 2020-04-30 08:23:58,415: loading config file /home/pi/catkin_ws/src/realsense-ros/realsense2_camera/launch/rs_rgbd.launch
> [roslaunch][INFO] 2020-04-30 08:23:58,543: Added node of type [nodelet/nodelet] in namespace [/camera/]
> [roslaunch][INFO] 2020-04-30 08:23:58,563: Added node of type [nodelet/nodelet] in namespace [/camera/]
> [roslaunch][ERROR] 2020-04-30 08:23:58,564: Resource not found: rgbd_launch
> ROS path [0]=/opt/ros/melodic/share/ros
> ROS path [1]=/home/pi/catkin_ws/src
> ROS path [2]=/opt/ros/melodic/share
> [roslaunch][ERROR] 2020-04-30 08:23:58,565: The traceback for the exception was written to the log file
> [roslaunch][ERROR] 2020-04-30 08:23:58,567: Traceback (most recent call last):
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 330, in main
>     p.start()
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 289, in start
>     self._start_infrastructure()
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 238, in _start_infrastructure
>     self._load_config()
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 144, in _load_config
>     roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/config.py", line 460, in load_config_default
>     loader.load(f, config, argv=args, verbose=verbose)
[... cut ...]
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 212, in _find_executable
>     full_path = _get_executable_path(rp.get_path(args[0]), path)
>   File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 207, in get_path
>     raise ResourceNotFound(name, ros_paths=self._ros_paths)
> ResourceNotFound: rgbd_launch
> ROS path [0]=/opt/ros/melodic/share/ros
> ROS path [1]=/home/pi/catkin_ws/src
> ROS path [2]=/opt/ros/melodic/share
> 
> [rospy.core][INFO] 2020-04-30 08:23:58,568: signal_shutdown [atexit]

MartyG-RealSense, 2020-04-30:
Try running this line to make sure rgbd-launch is installed:

**sudo apt install ros-kinetic-rgbd-launch**

nbelanger99, 2020-05-15:
Here's the log form running 
`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

```
... logging to /home/pi/.ros/log/eae6ad8a-96c8-11ea-b683-dca63263de10/roslaunch-raspberrypi4-1756.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi4:43505/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
[... 147 lines cut ...]
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
```
And we can see explicitely that there exists a publisher for the topic: 
`rostopic info /camera/depth/color/points `
```
Type: sensor_msgs/PointCloud2

Publishers: 
 * /camera/realsense2_camera_manager (http://raspberrypi4:43665/)

Subscribers: None
```
Bug again, no luck with color point cloud: 
`rostopic hz /camera/depth/color/points `
```
subscribed to [/camera/depth/color/points]
no new messages
no new messages
no new messages
no new messages
no new messages
```

*I'm working with @echoGee by the way.

MartyG-RealSense, 2020-05-15:
@nbelanger99  Regarding the error **'An error has occurred during frame callback: Error occured during execution of the processing block!**, @doronhi the RealSense ROS wrapper developer has previously said the following:

"From what I understand, the problem is that the filters don't keep up with the frames, due to computer load.  A new image cannot be pushed to the still full queue and therefore it is missing when the filter is called to process it.  Then the exception is issued".

echoGee, 2020-05-15:
We get one image topic, one depth cloud without color. However, we require all the data(second image etc) for our task. 

There is another reference on someone having to fix drivers: https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense
Do you the fixes mentioned in this link is relevant or upto date ?

MartyG-RealSense, 2020-05-15:
@echoGee   The installation script for the Pi 4 Raspbian Buster was most recently updated in SDK version 2.34.0

https://github.com/IntelRealSense/librealsense/pull/6040

echoGee, 2020-05-15:
The v2.34.0 is still giving a
`uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a` on the dmesg command. Is this expected ? 
This page(https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense#Installing_librealsense) mentions changing few kernel level things to solve this issue.

MartyG-RealSense, 2020-05-16:
@echoGee   I am not familiar with the workings of the Raspbian Buster installation script, so I will have to enquire to another RealSense team member.

@dorodnic  Can you advise about the question above about uvcvideo please?

echoGee, 2020-05-18:
It would also be helpful if there is an image for RPi4 with ROS and realsense drivers that we can just use.

MartyG-RealSense, 2020-05-18:
@echoGee  The RealSense developers have done some work recently in SDK version 2.34.0 to ease installation on Pi 4 and Raspbian Buster by updating the installation script for Pi 4 and Raspbian Buster to run cleanly.

https://github.com/IntelRealSense/librealsense/pull/6040

Increased support for Pi 4 / Buster would probably depend on Pi 4 users being vocal with requests so that the developers can gauge the demand for such a feature.

doronhi, 2020-05-26:
The issue with the pointcloud could be related to this error as appears on the log file you published: " get_device_time_ms() took too long (more then 2 mSecs)". This was fixed in librealsense2 v2.34.1. 
Could you upgrade and give it a try? Please disconnect and reconnect the camera before start running to avoid issues related to stopping the device at the end of a run.
I expect lots of warning are sill there but is the pointcloud now published and visible?

echoGee, 2020-05-27:
We tried installing the v2.34.1 on Pi 4 and it doesn't run the demo point cloud anymore. 

This is the dmesg output:
> [18567.485187] usb 2-2: new SuperSpeed Gen 1 USB device number 35 using xhci_hcd
> [18567.516224] usb 2-2: New USB device found, idVendor=8086, idProduct=0b3a, bcdDevice=50.c3
> [18567.516240] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [18567.516254] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 435i
> [18567.516267] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
> [18567.516279] usb 2-2: SerialNumber: 903223051356
> [18567.520994] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
> [18567.526634] uvcvideo: Unable to create debugfs 2-35 directory.
> [18567.527002] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
> [18567.527018] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
> [18567.527033] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
> [18567.527371] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/input/input23
> [18567.528827] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
> [18567.528853] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
> [18567.532875] uvcvideo: Unable to create debugfs 2-35 directory.
> [18567.533259] uvcvideo 2-2:1.3: Entity type for entity Processing 7 was not initialized!
> [18567.533274] uvcvideo 2-2:1.3: Entity type for entity Extension 8 was not initialized!
> [18567.533289] uvcvideo 2-2:1.3: Entity type for entity Camera 6 was not initialized!
> [18567.542709] hid-generic 0003:8086:0B3A.0018: hiddev96,hidraw0: USB HID v1.10 Device [Intel(R) RealSense(TM) Depth Camera 435i Intel(R) RealSense(TM) Depth Camera 435i] on usb-0000:01:00.0-2/input5
> 

We build the librealsense from source by checking out 2.34.1 and 
`cmake . `
`sudo make uninstall && make clean && make && sudo make install`
`sudo ./scripts/setup_udev_rules.sh`

doronhi, 2020-06-01:
Why is it not running the demo point cloud anymore? Can you post the log? 
Did you rebuild realsense2_camera to link with the librealsense2 v2.34.1?

MartyG-RealSense, 2020-06-10:
@echoGee   Do you still require assistance with this case please, or can it be closed?  Thanks!

closed 2020-06-18 by MartyG-RealSense