realsenseai/realsense-ros #1183: Raspbian buster + ROS melodic does not publish /camera/depth/color/points

echoGee, 2020-04-25:
- ROS Melodic(1.14.5) (compiled from source on Buster)
- RealSense ROS v2.2.13 (compiled from source on Buster)
- LibRealSense v2.34.0 (compiled from source on Buster)

I have run the realsense-viewer which runs fine.

Running launch file demo_pointcloud.launch (without rviz since it crashes in Raspberry pi)
> roslaunch realsense2_camera demo_withoutRVIZ_pointcloud.launch 

or 

> roslaunch realsense2_camera demo_withoutRVIZ_pointcloud.launch 
pointcloud_texture_stream:=RS2_STREAM_COLOR

Console log:
[demo_launch_withoutRviz_25Apr2020.txt](https://github.com/IntelRealSense/realsense-ros/files/4534131/demo_launch_withoutRviz_25Apr2020.txt)

does not publish any color depth points in /camera/depth/color/points

Output of command [rs-color](https://github.com/IntelRealSense/realsense-ros/files/4534145/rs-color.txt)
Output of command [rs-enumerate-devices.txt](https://github.com/IntelRealSense/realsense-ros/files/4534146/rs-enumerate-devices.txt)

MartyG-RealSense, 2020-04-26:
In your other case today I suggested using the launch instruction **roslaunch realsense2_camera rs_camera.launch filters:=pointcloud** with your Buster / wrapper 2.2.13 / SDK 2.34.0 setup.

Are color depth points published in **/camera/depth/color/points** if you use this launch instruction, please?

echoGee, 2020-04-27:
Tried this. But it is not publishing depth points to **/camera/depth/color/points**

MartyG-RealSense, 2020-04-27:
Updated comment on other case:

https://github.com/IntelRealSense/realsense-ros/issues/1182#issuecomment-620079215

MartyG-RealSense, 2020-05-17:
Case closed due to the discussion continuing at the other case https://github.com/IntelRealSense/realsense-ros/issues/1182

closed 2020-05-17 by MartyG-RealSense