realsenseai/realsense-ros #1184: Which type USB cable to use for stable work of RealSense D435I and realsense-ros?

amburkoff, 2020-04-28:
I use NVIDIA Jetson Xavier as a computer with two USB ports type-C 3.1 and one USB ports type-A 3.0. I have a lot of problems working with RealSense D435I and D435 through these ports.

Tell me, please, for what speed USB ports and USB cable RealSense D435I is designed for stable operation(5,10 or 20gbps)?
Which type USB cable type-C to type-C 3.0/3.1 Gen 1/3.1 Gen 2 and cable length 1m,1.5m,1.8m?
Which type USB cable type-A to type-C 3.0/3.1 Gen 1/3.1 Gen 2 and cable length 1m,1.5m,1.8m?
Are any other characteristics important?
Please provide links to recommended cables.

Can I use additional usb hubs? which models?

Is it important what type of usb port on the computer and its speed? What is the minimum value for stable operation?

MartyG-RealSense, 2020-04-28:
RealSense cameras transmit a high volume of data through the USB cable when using high resolutions / FPS speeds, so a high quality USB cable is important.  The longer the cable is, the greater the chance of performance degradation or camera disconnections if the cable is not high quality.

The reports I have seen from RealSense users indicate that both Gen 1 and Gen 2 should work.  Here is an example report:

https://github.com/IntelRealSense/librealsense/issues/4603#issuecomment-521448700

My primary recommendation for industrial grade USB cables for use with RealSense is **Newnex**.  They have supported RealSense for a long time, have a dedicated page for RealSense cabling and have also demonstrated on video that their long-length USB cables and equipment work well with RealSense.

**Newnex RealSense web-page:**
https://www.newnex.com/realsense-3d-camera-connectivity.php

**NewNex YouTube long-distance cabling demonstration:**
https://www.youtube.com/watch?v=GLQgR1jT04M

There are of course other suppliers of industrial grade USB 3 cables.  Here is another example:

https://www.amazon.com/Tether-Tools-USB-USB-C-orange/dp/B0794B1SDR/ref=sr_1_7

If it is a high quality cable then the length of it should not matter so much and it should be able to cope with transmitting data over lengths 2 m or more.

The USB standard allows for the chaining together of up to 5 hubs (a principle known as '5 levels deep').

I have seen reports of some USB hubs that work whilst others do not.  For example, a user of an expensive USB 3.2 industrial grade hub had problems but a user of an inexpensive SUB 3.1 hub did not.  When Intel wrote their white-paper document on use of multiple RealSense cameras, they successfully used an AmazonBasics mains-powered USB 3 hub to test the speeds of up to 4 cameras.

The full USB hub bandwidth test result charts are available in Section 2 of the multiple camera paper:

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

In regard to the type of USB port, RealSense 400 Series cameras can work with both USB2 and USB 3.  The camera has limited speeds and modes available on a USB2 connection though due to the limitations of USB2 ports compared to USB 3.

amburkoff, 2020-04-28:
Thanks for the detailed answer!

MartyG-RealSense, 2020-05-17:
Hi @amburkoff   Do you require further assistance with this question please, or can the case be closed?   Thanks!

closed 2020-05-24 by MartyG-RealSense