realsenseai/realsense-ros #1185: How to prevent realsense from show these logs

5730289021-NN, 2020-04-29:
Greeting everyone, may I have some little help here

My launch file is
`<launch>
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="filters"             value="pointcloud"/>
        <arg name="depth_width"         value="640"/>
        <arg name="depth_height"        value="480"/>
        <arg name="enable_depth"        value="true"/>
        <arg name="clip_distance"       value="2.5"/>
	    
        <arg name="color_width"         value="640"/>
        <arg name="color_height"        value="480"/>
        <arg name="enable_color"        value="true"/>

        <arg name="enable_fisheye"      value="false"/>
        <arg name="enable_infra1"       value="false"/>
        <arg name="enable_infra2"       value="false"/>

        <arg name="device_type"         value="d435i"/>

        <arg name="depth_fps"           value="15"/>
        <arg name="color_fps"           value="15"/>
    </include>
</launch>`

and now the launch file emitting logs like

```
 29/04 19:46:08,668 WARNING [140540339263232] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1588164368.882269952]: Hardware Notification:USB CAM overflow,1.58816e+12,Error,Hardware Error
 29/04 19:46:10,784 WARNING [140540339263232] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 13, 
[ WARN] [1588164370.913553904]: Hardware Notification:USB CAM overflow,1.58816e+12,Error,Hardware Error
 29/04 19:46:12,900 WARNING [140540607698688] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 13, 

```
How could I edit my launch file to suppress that?

MartyG-RealSense, 2020-04-29:
You could try adding to your launch file:

**output="log"**

This should send warnings to a log file in **$ROS_HOME/log** instead of the screen.

5730289021-NN, 2020-04-29:
Thank you for the solution. I thought it was normal to get those message since there were tons of them when I ran the launch file.

MartyG-RealSense, 2020-04-29:
No, it's not normal.  It suggests that something is overwhelming the USB port with data.

Someone who had the same problem when using the camera on a USB2 port resolved it by disabling the infrared streams by adding the instructions below to the roslaunch statement.

**enable_infra1:=false enable_infra2:=false**

5730289021-NN, 2020-04-29:
I see, I'll figure out on that.
Furthermore, the **output="log" can only be defined has to be defined within [<node> tag](https://wiki.ros.org/roslaunch/XML/node) like <node name="listener1" output="log" /> but I think from this library structure it's better to manipulate within [<include> tag](https://wiki.ros.org/roslaunch/XML/include) or [<launch> tag](https://wiki.ros.org/roslaunch/XML/launch). I don't think putting inside [here](https://github.com/IntelRealSense/realsense-ros/blob/3ab5077763e980ae72c0c29aec4a697067d43744/realsense2_camera/launch/includes/nodelet.launch.xml#L94) is a good idea

MartyG-RealSense, 2020-04-29:
Good luck!  Please feel welcome to come back to the RealSense ROS forum any time that you need to.

5730289021-NN, 2020-04-30:
> I see, I'll figure out on that.
> Furthermore, the **output="log"** can only be defined within [node tag](https://wiki.ros.org/roslaunch/XML/node) like `<node name="listener1" output="log" />` but I think from this library structure it's better to manipulate within [include tag](https://wiki.ros.org/roslaunch/XML/include) or [launch tag](https://wiki.ros.org/roslaunch/XML/launch). I don't think putting inside [here](https://github.com/IntelRealSense/realsense-ros/blob/3ab5077763e980ae72c0c29aec4a697067d43744/realsense2_camera/launch/includes/nodelet.launch.xml#L94) is a good idea

Any idea on fixing the launch tag without adding “output” attribute to the <node> parameter directly?

MartyG-RealSense, 2020-04-30:
The simplest methods that I could find to disable all logging are in the link below:

http://wiki.ros.org/rosconsole#Disable_all_logging

5730289021-NN, 2020-05-01:
Alright, thank you

closed 2020-05-01 by 5730289021-NN