realsenseai/realsense-ros #1186: D435 RGB Motion Blur

smtabatabaie, 2020-05-01:
Hi,
I'm trying to gather a dataset with a pair of T265 and D435. Everything is set except that I experience a very bad motion blur in my D435 RGB images (IR images are fine). I tried several suggestions from the forums including setting auto-exposure off and tried testing on very high and very low FPS. But none of them worked. I'm getting very desperate in this project and I'll appreciate if you can help me with this.
Thanks very much.

MartyG-RealSense, 2020-05-01:
Although the D435 camera model's IR imager component has a fast "global shutter", its RGB imager component is a slower "rolling" type shutter, and these are vulnerable to motion blur / smear when dealing with motion that is faster than human walking pace.  

The discussion in the link below may provide some useful insights about RGB motion blur on D435.

https://support.intelrealsense.com/hc/en-us/community/posts/360036210733-D435-control-RGB-module-speed

The first thing I would recommend trying is disabling the 'Auto Exposure Priority' option to enforce a constant FPS and see if that helps with the blur.  On the RealSense Viewer, you can test this by toggling the 'Auto-Exposure Priority' box off in the "RGB" section of the options side panel.

In a well-lit scene such as strong sunlight, manually setting exposure to around 1 ms may also help to reduce motion artifacts.

MartyG-RealSense, 2020-05-17:
Hi @smtabatabaie   Do you require further assistance with this question please, or is it okay to close this case?   Thanks!

closed 2020-05-24 by MartyG-RealSense