realsenseai/realsense-ros #1188: Dosen't work with rs_t265.launch

Geonhee-LEE, 2020-05-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |    Linux (Ubuntu 18)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  v2.34.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  ROBOT }                          | 

### Issue Description

Hi, I want to use the tracking camera T265.
So that use it I installed the librealsense(v2.34.0) and realsense_ros(v2.2.13) pkg. I could check the working well using Realsense-viewer. But When I launch with rs_t265.launch, it occurred the error like below:

`07/05 15:19:53,915 ERROR [139789561546496] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_NO_DEVICE
 07/05 15:20:03,916 WARNING [139789561546496] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: No such device, err. num: 19
 07/05 15:20:03,916 ERROR [139789561546496] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_NO_DEVICE
 07/05 15:20:03,916 WARNING [139789561546496] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: No such device, err. num: 19
 07/05 15:20:03,916 ERROR [139789561546496] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_NO_DEVICE
 07/05 15:20:13,917 WARNING [139789561546496] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: No such device, err. num: 19
 07/05 15:20:13,917 ERROR [139789561546496] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_NO_DEVICE
 07/05 15:20:13,917 WARNING [139789561546496] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: No such device, err. num: 19
 07/05 15:20:13,917 ERROR [139789561546496] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_NO_DEVICE
 07/05 15:20:23,917 WARNING [139789561546496] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: No such device, err. num: 19
 07/05 15:20:23,917 ERROR [139789561546496] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_NO_DEVICE
 07/05 15:20:23,918 WARNING [139789561546496] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: No such device, err. num: 19
 07/05 15:20:23,918 ERROR [139789561546496] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_NO_DEVICE
 07/05 15:20:33,918 WARNING [139789561546496] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: No such device, err. num: 19
 07/05 15:20:33,918 ERROR [139789561546496] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_NO_DEVICE
 07/05 15:20:33,918 WARNING [139789561546496] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: No such device, err. num: 19
 07/05 15:20:33,919 ERROR [139789561546496] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_NO_DEVICE`

Please help me.
regards.

closed 2020-08-25 by Geonhee-LEE

SalahSoliman, 2021-02-09:
I get the same issue, how did you solve it?

Geonhee-LEE, 2021-02-09:
@SalahSoliman  Hi, the problem is USB connection

SalahSoliman, 2021-02-10:
Thank you for the quick response, so it’s only the usb cable was causing the problem? 😮