realsenseai/realsense-ros #1191: roslaunch realsense2_camera rs_camera.launch error

Eslamrady42, 2020-05-08:
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud /////// I have error with this command:
 eslam@root:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to /home/eslam/.ros/log/41c06a78-910c-11ea-acda-64006afb8a06/roslaunch-root-3896.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://root:40451/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
[... 72 lines cut ...]
[ INFO] [1588929272.673530542]: Expected frequency for depth = 30.00000
[ INFO] [1588929272.695807088]: Expected frequency for infra1 = 30.00000
[ INFO] [1588929272.707848730]: Expected frequency for infra2 = 30.00000
[ INFO] [1588929272.722153323]: Expected frequency for color = 30.00000
[ INFO] [1588929272.737794460]: setupStreams...
[ INFO] [1588929272.738547885]: insert Depth to Stereo Module
[ INFO] [1588929272.738594438]: insert Color to RGB Camera
[ INFO] [1588929272.738618780]: insert Infrared to Stereo Module
[ INFO] [1588929272.738642169]: insert Infrared to Stereo Module
[ INFO] [1588929272.744466230]: SELECTED BASE:Depth, 0
[ INFO] [1588929272.801163125]: RealSense Node Is Up!
 08/05 11:14:32,968 ERROR [140683238541056] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 08/05 11:14:33,170 ERROR [140683238541056] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 08/05 11:14:33,439 ERROR [140683238541056] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 08/05 11:14:33,641 ERROR [140683238541056] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 08/05 11:14:39,102 ERROR [140683238541056] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
[camera/realsense2_camera_manager-2] process has died [pid 3924, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/eslam/.ros/log/41c06a78-910c-11ea-acda-64006afb8a06/camera-realsense2_camera_manager-2.log].
log file: /home/eslam/.ros/log/41c06a78-910c-11ea-acda-64006afb8a06/camera-realsense2_camera_manager-2*.log
^C[camera/realsense2_camera-3] killing on exit
[rosout-1] killing on exit
[rosout-1] escalating to SIGTERM
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2020-05-08:
The error **get_device_time_ms() took too long (more then 2 mSecs)** has been a well documented error with SDK 2.34.0.  I would recommend using the previous Librealsense version 2.33.1 if possible until the error is fixed.  I do apologise for the inconvenience.

Eslamrady42, 2020-05-08:
> The error **get_device_time_ms() took too long (more then 2 mSecs)** has been a well documented error with SDK 2.34.0. I would recommend using the previous Librealsense version **2.33.1** if possible until the error is fixed. I do apologise for the inconvenience

So how can I change it to  Librealsense version **2.33.1** by a commend or link ??

MartyG-RealSense, 2020-05-08:
I haven't done this process of installing a specific SDK on Linux myself, so I apologise in advance for any mistakes.  

I would speculate that it could be done in the following way:

1.  Download the zip file of the source code for 2.33.1 from the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1

If you scroll down to the page, you should come to a section titled **Assets** where the source code zip file for 2.33.1 can be downloaded from.

[image]

2.  Follow the instructions for compiling the Librealsense SDK from source in the link below from the '**Prepare Linux Backend and the Dev. Environment**' section onwards.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

[image]

I note that you added an an edit that you tried the 2.33.1 link.  What happened when you did so, please?

Eslamrady42, 2020-05-08:
> I haven't done this process of installing a specific SDK on Linux myself, so I apologise in advance for any mistakes.  
> 
> 
> 
> I would speculate that it could be done in the following way:
> 
> 
> 
> 1.  Download the zip file of the source code for 2.33.1 from the link below.
> 
> 
> 
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1
> 
> 
> 
> If you scroll down to the page, you should come to a section titled **Assets** where the source code zip file for 2.33.1 can be downloaded from.
> 
> 
> 
> [image]
> 
> 
> 
> 2.  Follow the instructions for compiling the Librealsense SDK from source in the link below from the '**Prepare Linux Backend and the Dev. Environment**' section onwards.
> 
> 
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
> 
> 
> 
> [image]
> 
> 
> 
> I note that you added an edit that you tried the 2.33.1 link.  What happened when you did so, please?

Thank you for your efforts in the project and the issues fixing but please write down in SDK 2.34 it’s have a bug until you solve it to avoid any loss in time and efforts .

MartyG-RealSense, 2020-05-08:
There is a fix that has been created for the error but it has not passed review by the developers for implementation into the SDK yet, unfortunately.  

https://github.com/IntelRealSense/librealsense/pull/6150

A new version of the Librealsense SDK should be released soon, so it may be worth @doronhi the RealSense ROS wrapper developer adding a note about the error to the wrapper release notes if the new SDK release does not correct the **get_device_time_ms** error.

Eslamrady42, 2020-05-08:
> There is a fix that has been created for the error but it has not passed review by the developers for implementation into the SDK yet, unfortunately.
> 
> [IntelRealSense/librealsense#6150](https://github.com/IntelRealSense/librealsense/pull/6150)
> 
> A new version of the Librealsense SDK should be released soon, so it may be worth @doronhi the RealSense ROS wrapper developer adding a note about the error to the wrapper release notes if the new SDK release does not correct the **get_device_time_ms** error.

I removed every thing from SDK 2.34 and reinstall SDK 2.33 by your links but I have the same problem and realsense-viwer is on SDK 2.34 after I install SDK 2.33 !!!!

MartyG-RealSense, 2020-05-08:
I'm very sorry that using the previous Librealsense version did not solve your problem.  

Are you able to view the output of camera topics in RViz, please?   That would indicate that the topics are being published correctly despite the error messages.

Eslamrady42, 2020-05-08:
> I'm very sorry that using the previous Librealsense version did not solve your problem.  
> 
> 
> 
> Are you able to view the output of camera topics in RViz, please?   That would indicate that the topics are being published correctly despite the error messages.

Unfortunately it can’t load the camera or run it including the topics ...

MartyG-RealSense, 2020-05-08:
There was a case a month ago who had almost identical errors to yours when their process died.  There was not a successful resolution in that case, unfortunately.

https://github.com/IntelRealSense/realsense-ros/issues/1159#issuecomment-611620704

The logs suggest that you are using the camera in a USB 2 port.  If so, do you have an available USB 3 port that you can try the camera in please to see if it makes a difference?

Eslamrady42, 2020-05-08:
> There was a case a month ago who had almost identical errors to yours when their process died.  There was not a successful resolution in that case, unfortunately. [...]

Definitely it’s not the same error and already I connect it by USB 3 because I read the previous issues with the cameras

MartyG-RealSense, 2020-05-08:
Please try a different launch instruction and see if it publishes a pointcloud:

roslaunch realsense2_camera demo_pointcloud.launch

Eslamrady42, 2020-05-09:
> Please try a different launch instruction and see if it publishes a pointcloud:
> 
> roslaunch realsense2_camera demo_pointcloud.launch

I found the seme error !!

MartyG-RealSense, 2020-05-09:
When you say that you "got the same error", do you mean that you got the same errors as before when using the different point cloud launch instruction that I suggested, please?

Eslamrady42, 2020-05-09:
> When you say that you "got the same error", do you mean that you got the same errors as before when using the different point cloud launch instruction that I suggested, please?
> 
> 

Yes that’s what I mean ..

MartyG-RealSense, 2020-05-09:
Same result if you drop 'filters' from the launch instruction and just do **roslaunch realsense2_camera rs_camera.launch** ?

If that does not work, try deleting your workspace build folder and building again.

Eslamrady42, 2020-05-09:
> Same result if you drop 'filters' from the launch instruction and just do **roslaunch realsense2_camera rs_camera.launch** ?
> 
> If that does not work, try deleting your workspace build folder and building again.

I did all of this and what I revived :

eslam@root:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/eslam/.ros/log/6a288564-91fb-11ea-99da-64006afb8a06/roslaunch-root-5686.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://root:46575/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 158 lines cut ...]
[ INFO] [1589032009.943452924]: HectorSM p_update_factor_free_: 0.400000
[ INFO] [1589032009.943468277]: HectorSM p_update_factor_occupied_: 0.900000
[ INFO] [1589032009.943480630]: HectorSM p_map_update_distance_threshold_: 0.400000 
[ INFO] [1589032009.943493122]: HectorSM p_map_update_angle_threshold_: 0.060000
[ INFO] [1589032009.943504924]: HectorSM p_laser_z_min_value_: -1.000000
[ INFO] [1589032009.943519763]: HectorSM p_laser_z_max_value_: 1.000000
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/eslam/.ros/log/6a288564-91fb-11ea-99da-64006afb8a06/camera-realsense2_camera-3*.log
[ WARN] [1589032029.901492120]: No transform between frames /map and /base_link available after 20.003345 seconds of waiting. This warning only prints once.
^C[hector_geotiff_node-10] killing on exit
[hector_trajectory_server-9] killing on exit
[rviz-8] killing on exit
[hector_mapping-7] killing on exit
[base_link_to_Camera-6] killing on exit
[odom_to_base_link-5] killing on exit
[map_to_odom-4] killing on exit
Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
         at line 122 in /tmp/binarydeb/ros-melodic-class-loader-0.4.1/src/class_loader.cpp
[rosout-1] killing on exit
[rosout-1] escalating to SIGTERM
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
eslam@root:~/catkin_ws$

MartyG-RealSense, 2020-05-09:
I notice that the log referred to T265 this time, and your other new case on the main GitHub also mentioned T265.   Do you have both a T265 and a D435 attached to the computer at the same time?

Eslamrady42, 2020-05-09:
> I notice that the log referred to T265 this time, and your other new case on the main GitHub also mentioned T265. Do you have both a T265 and a D435 attached to the computer at the same time?

Firstly I tried without T265 then I found the old error with types.h after that I connected the both cameras and tried and the previous error what I got .

MartyG-RealSense, 2020-05-09:
Using the T265 and D435 on the same computer together in ROS can cause problems, because if the D435 is recognised by the computer first then it can stop the T265 from working.  A launch file for using the D435 and T265 together was created:

**roslaunch realsense2_camera rs_d400_and_t265.launch**

Eslamrady42, 2020-05-09:
> Using the T265 and D435 on the same computer together in ROS can cause problems, because if the D435 is recognised by the computer first then it can stop the T265 from working. A launch file for using the D435 and T265 together was created:
> 
> **roslaunch realsense2_camera rs_d400_and_t265.launch**

when I tried : **roslaunch realsense2_camera rs_d400_and_t265.launch** : 

eslam@root:~/catkin_ws$ roslaunch realsense2_camera rs_d400_and_t265.launch 
... logging to /home/eslam/.ros/log/36cd1824-9200-11ea-99da-64006afb8a06/roslaunch-root-6349.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://root:33519/

SUMMARY
========

PARAMETERS
 * /d400/realsense2_camera/accel_fps: 0
 * /d400/realsense2_camera/accel_frame_id: d400_accel_frame
 * /d400/realsense2_camera/accel_optical_frame_id: d400_accel_optica...
 * /d400/realsense2_camera/align_depth: True
 * /d400/realsense2_camera/aligned_depth_to_color_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/allow_no_texture_points: False
 * /d400/realsense2_camera/base_frame_id: d400_link
 * /d400/realsense2_camera/calib_odom_file: 
 * /d400/realsense2_camera/clip_distance: -2.0
 * /d400/realsense2_camera/color_fps: 30
 * /d400/realsense2_camera/color_frame_id: d400_color_frame
 * /d400/realsense2_camera/color_height: 480
 * /d400/realsense2_camera/color_optical_frame_id: d400_color_optica...
 * /d400/realsense2_camera/color_width: 640
 * /d400/realsense2_camera/depth_fps: 30
 * /d400/realsense2_camera/depth_frame_id: d400_depth_frame
 * /d400/realsense2_camera/depth_height: 480
 * /d400/realsense2_camera/depth_optical_frame_id: d400_depth_optica...
 * /d400/realsense2_camera/depth_width: 640
 * /d400/realsense2_camera/device_type: d4.5
 * /d400/realsense2_camera/enable_accel: True
 * /d400/realsense2_camera/enable_color: True
 * /d400/realsense2_camera/enable_depth: True
 * /d400/realsense2_camera/enable_fisheye1: True
 * /d400/realsense2_camera/enable_fisheye2: True
 * /d400/realsense2_camera/enable_fisheye: True
 * /d400/realsense2_camera/enable_gyro: True
 * /d400/realsense2_camera/enable_infra1: True
 * /d400/realsense2_camera/enable_infra2: True
 * /d400/realsense2_camera/enable_pointcloud: False
 * /d400/realsense2_camera/enable_sync: False
 * /d400/realsense2_camera/filters: pointcloud
 * /d400/realsense2_camera/fisheye1_frame_id: d400_fisheye1_frame
 * /d400/realsense2_camera/fisheye1_optical_frame_id: d400_fisheye1_opt...
 * /d400/realsense2_camera/fisheye2_frame_id: d400_fisheye2_frame
 * /d400/realsense2_camera/fisheye2_optical_frame_id: d400_fisheye2_opt...
 * /d400/realsense2_camera/fisheye_fps: 30
 * /d400/realsense2_camera/fisheye_frame_id: d400_fisheye_frame
 * /d400/realsense2_camera/fisheye_height: 0
 * /d400/realsense2_camera/fisheye_optical_frame_id: d400_fisheye_opti...
 * /d400/realsense2_camera/fisheye_width: 0
 * /d400/realsense2_camera/gyro_fps: 0
 * /d400/realsense2_camera/gyro_frame_id: d400_gyro_frame
 * /d400/realsense2_camera/gyro_optical_frame_id: d400_gyro_optical...
 * /d400/realsense2_camera/imu_optical_frame_id: d400_imu_optical_...
 * /d400/realsense2_camera/infra1_frame_id: d400_infra1_frame
 * /d400/realsense2_camera/infra1_optical_frame_id: d400_infra1_optic...
[... 265 lines cut ...]
[ INFO] [1589034081.452971771]: HectorSM p_update_factor_free_: 0.400000
[ INFO] [1589034081.452985402]: HectorSM p_update_factor_occupied_: 0.900000
[ INFO] [1589034081.453001650]: HectorSM p_map_update_distance_threshold_: 0.400000 
[ INFO] [1589034081.453019072]: HectorSM p_map_update_angle_threshold_: 0.060000
[ INFO] [1589034081.453032871]: HectorSM p_laser_z_min_value_: -1.000000
[ INFO] [1589034081.453047760]: HectorSM p_laser_z_max_value_: 1.000000
[ WARN] [1589034101.397361197]: No transform between frames /map and /base_link available after 20.005289 seconds of waiting. This warning only prints once.
[rviz-11] process has finished cleanly
log file: /home/eslam/.ros/log/36cd1824-9200-11ea-99da-64006afb8a06/rviz-11*.log
^C[hector_geotiff_node-13] killing on exit
[hector_trajectory_server-12] killing on exit
[hector_mapping-10] killing on exit
[base_link_to_d400_and_t265-9] killing on exit
[odom_to_base_link-8] killing on exit
[map_to_odom-7] killing on exit
[t265_to_d400-6] killing on exit
[t265/realsense2_camera_manager-2] killing on exit
Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
         at line 122 in /tmp/binarydeb/ros-melodic-class-loader-0.4.1/src/class_loader.cpp
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
eslam@root:~/catkin_ws$

MartyG-RealSense, 2020-05-09:
I'm sorry that the **roslaunch realsense2_camera rs_d400_and_t265.launch** launch instruction did not solve your problem.

At this point, because the case now involves the T265 I will have to refer it to a support manager to see what they say, as I am not a T265 engineer.  I apologise very much for the delay this will cause in the meantime.

Eslamrady42, 2020-05-09:
> I'm sorry that the **roslaunch realsense2_camera rs_d400_and_t265.launch** launch instruction did not solve your problem.
>
> At this point, because the case now involves the T265 I will have to refer [...]

but my problem in the both cameras not only T265 !!!

MartyG-RealSense, 2020-05-09:
I am sorry, if a case involves T265 at all then I cannot take it on and would not have been able to start this case if I had known that a T265 was being used.  I understand how frustrating this is for you.  I have referred the case to a support manager above me.

Eslamrady42, 2020-05-09:
> I am sorry, if a case involves T265 at all then I cannot take it on and would not have been able to start this case if I had known that a T265 was being used.  I understand how frustrating this is for you.  I have referred the case to a support manager above me.

Thank you for your efforts and support and I hope he can replay on me as fast as he can because I’m working by this two cameras on my graduation project so I don’t have any time to wait.

MartyG-RealSense, 2020-05-09:
I will emphasise to them that speed of answering is important due to your graduation project.  I have an engineering degree, so I remember what graduation projects are like!

Eslamrady42, 2020-05-09:
> I will emphasise to them that speed of answering is important due to your graduation project. I have an engineering degree, so I remember what graduation projects are like!

that's great , all we can found from this project only errors in the cameras also in objectdetection by the depth camera.
Thank you again and I'm waiting for his response .

MartyG-RealSense, 2020-05-09:
Please do not wait at the computer for a response as I do not know when they will reply.  It might not even be today.  You should receive an email when somebody replies in the same way that you do when I make a comment.  I hope that you can find another aspect of your project to work on in the meantime.

Eslamrady42, 2020-05-09:
> Please do not wait at the computer for a response as I do not know when they will reply. It might not even be today. You should receive an email when somebody replies in the same way that you do [...]

yes, I know I'll leave it .
**I hope you have a nice day ** ^^.

MartyG-RealSense, 2020-05-09:
Thanks very much for your understanding.  I wish the very best for you too.   Good luck!   :)

doronhi, 2020-05-10:
Hi @Eslamrady42 ,
Regarding the "get_device_time_ms() took too long" error, as @MartyG-RealSense already said it was introduced in v2.34.0 and already have a PR to fix it.
The error message did not exist in v2.34.1 (I downloaded [librealsense-2.33.1.zip](https://github.com/IntelRealSense/librealsense/archive/v2.33.1.zip) file and double checked.)
Therefor, although the log of realsense2_camera shows librealsense2 v2.33.1, the version you built was not it. I really don't know how that happened.
Also, if realsense-viewer shows version 2.34.1 it means that the system is not clean of that version.
When you manage to clean your machine and have librealsense2 v2.33.1 only you cannot see the   "get_device_time_ms() took too long" error.

Assuming v.2.33.1, using D435 camera alone, running `roslaunch realsense2_camera rs_camera.launch` without any other nodes such - do you see any issues? are the topics published and can be subscribed with RViz?

Can you check T265 alone, without D435 or any other node (Using `roslaunch realsense2_camera rs_t265.launch`)?

Eslamrady42, 2020-05-10:
> Hi @Eslamrady42 ,
>
> Regarding the "get_device_time_ms() took too long" error, as @MartyG-RealSense already said it was introduced in v2.34.0 and already have a PR to fix it.
>
> The error message did [...]

First of all thank you for your fast response, secondly I don’t know how to clean up my machine  from SDK 3.34 if you have a method or link to follow up I’ll do it know, Also I tried what did you said then I found another error while I connect T265 only: ( https://github.com/IntelRealSense/librealsense/issues/6374 ).

zustwangmingjun, 2020-05-22:
> When I tried a different launch instruction, this is OK. it publishes a pointcloud:
> roslaunch realsense2_camera demo_pointcloud.launch

>But I tried roslaunch realsense2_camera rs_camera.launch, It has the same problem.

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
[... 60 lines cut ...]
[ INFO] [1590129935.003552664]: Setting Dynamic reconfig parameters.
[ INFO] [1590129935.068597028]: Done Setting Dynamic reconfig parameters.
[ INFO] [1590129935.082793683]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1590129935.083118051]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1590129935.083377088]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
 22/05 14:45:35,084 WARNING [140263851534080] (backend-v4l2.cpp:1223) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO] [1590129935.102782492]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1590129935.102870406]: setupPublishers...
[ INFO] [1590129935.105807247]: Expected frequency for depth = 30.00000
[ INFO] [1590129935.151810458]: Expected frequency for infra1 = 30.00000
[ INFO] [1590129935.186884937]: Expected frequency for infra2 = 30.00000
[ INFO] [1590129935.220902832]: Expected frequency for color = 30.00000
[ INFO] [1590129935.264991666]: setupStreams...
[ INFO] [1590129935.266220585]: insert Depth to Stereo Module
[ INFO] [1590129935.266305668]: insert Color to RGB Camera
[ INFO] [1590129935.266348786]: insert Infrared to Stereo Module
[ INFO] [1590129935.266378507]: insert Infrared to Stereo Module
[ INFO] [1590129935.299557591]: SELECTED BASE:Depth, 0
[ INFO] [1590129935.367030602]: RealSense Node Is Up!
 22/05 14:45:35,416 WARNING [140263607387904] (ds5-timestamp.cpp:73) UVC metadata payloads not available. Please refer to the installation chapter for details.
[... cut ...] )
 22/05 14:45:35,619 ERROR [140263766882048] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 22/05 14:45:35,718 WARNING [140263620728576] (ds5-timestamp.cpp:73) UVC metadata payloads not available. Please refer to the installation chapter for details.
 22/05 14:45:35,822 ERROR [140263766882048] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 22/05 14:45:36,074 ERROR [140263766882048] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)

doronhi, 2020-05-24:
@Eslamrady42 , Sorry for the late response. Hopefully I'll be more available in the next couple of days.
My previous message, I see now, was very confusing and I mixed up the versions. Let me clarify:
We want to use librealsense2 version 2.33.1 as version v2.34.1 introduced a bug.

You referred to [Librealsense issue 6374](https://github.com/IntelRealSense/librealsense/issues/6374). In the log there I can see "Running with LibRealSense v2.33.1"
. In the [above comment](https://github.com/IntelRealSense/realsense-ros/issues/1191#issuecomment-626184014) I see "Running with LibRealSense v2.34.0" in the log.
So, I am confused. What version of librealsense2 did you end up with?

The method you showed to remove librealsense2 seems good. If you built from source you also need to remove this version : `make uninstall`
Make sure "realsense-viewer", "rs-enumerate-device" etc. aren't available before installing version 2.33.1.
Again, once you manage that, let's check sanity with D435 alone. Is this working now? Can you see the relevant information using RViz?

You've create another issue for the T265 error ("SLAM_ERROR"). I don't know how to help with that one yet but it is a good thing to try and separate the issues.
I'd like to treat T265 in another issue as this one is getting too long and confusing for me.
Assuming you successfully downgraded to v2.33.1, are there still issues with **D435**?

doronhi, 2020-05-24:
@zustwangmingjun 
For having point cloud using rs_camera.launch you have to specify the flag "filters:=pointcloud". I can see in the log you attached that no filter was enabled.
Also, the warning "(backend-v4l2.cpp:1223) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!" suggest you did not apply kernel patches (install librealsense2-dkms).
Also, if the message "get_device_time_ms() took too long" appears too often, it's possible that you will also need to downgrade to librealsense2 2.33.1 for the point cloud to be available.

Finally, I will appreciate it if you opened a new, specific issue, with details regarding your machine and OS, if needs further assistance.
Thanks.
Doron

truhoang-ubtrobot, 2020-06-03:
I'm running into the launch issue with similar error message as OP. The launch file worked before but I recently updated to `librealsense 2.35.0` and the launch file no longer works. I narrowed the error down to the `nodelet.launch.xml` specfically in the `RealSenseNodeFactory` class. The `realsense2_camera_manager` failed to launch due to segfault.

```
Thread 25 "nodelet" received signal SIGSEGV, Segmentation fault.
```

I enabled `gdb` and ran a backtrace: 

```shell
    this=<optimized out>) at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/types.h:415
#34 librealsense::sensor_base::get_stream_profiles (this=<optimized out>, tag=<optimized out>)
    at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/sensor.cpp:196
#35 0x00007fffda494e6c in rs2_get_stream_profiles (sensor=0x7fffa4091570, error=0x7fffa8b2c238)
    at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/rs.cpp:306
#36 0x00007fffdb4f1a4b in rs2::sensor::get_stream_profiles (this=0x7fffa4090f28)
    at /usr/include/librealsense2/hpp/rs_sensor.hpp:228
#37 0x00007fffdb4d6594 in realsense2_camera::BaseRealSenseNode::enable_devices (this=0x7fffa401e130)
    at /home/truhoang/1-Project_Code/ubtech_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:903
#38 0x00007fffdb4c9aa5 in realsense2_camera::BaseRealSenseNode::publishTopics (this=0x7fffa401e130)
    at /home/truhoang/1-Project_Code/ubtech_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:191
#39 0x00007fffdb490d1f in realsense2_camera::RealSenseNodeFactory::StartDevice (this=0x55555624c940)
    at /home/truhoang/1-Project_Code/ubtech_ws/src/realsense-ros/realsense2_camera/src/realsense_node_factory.cpp:320
```

@doronhi @MartyG-RealSense Does this provide any clue to you?

truhoang-ubtrobot, 2020-06-03:
Downgraded `librealsense` to `v2.33.1` and relaunch shows no segfault error. But new firmware installed using `realsense-viewer` (`05.12.05.00`) shows the following error:

```shell
[ INFO] [1591148280.396089253]:  
 02/06 18:38:00,406 ERROR [140402921432832] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 02/06 18:38:00,427 WARNING [140403035809536] (rs.cpp:283) null pointer passed for argument "device"
[ WARN] [1591148280.427877974]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1591148280.427915510]: The requested device with device name containing d435 is NOT found. Will Try again.
```

On `realsense-viewer` the camera is shown connected to a USB 3.2 port

doronhi, 2020-06-03:
@truhoang-ubtrobot ,
Thanks for the gdb input. I'll try to reproduce and check.
Could you please open this as a new issue as this is a different issue plus this issue is already very long?
Second, could you specify there your type of hardware and OS?

truhoang-ubtrobot, 2020-06-03:
@doronhi please see #1228

RealSenseSupport, 2020-08-20:
@Eslamrady42 

Does this issue still occur with the latest SDK and ROS updates?

*If we don’t hear from you in 7 days, this issue will be closed.

Eslamrady42, 2020-08-21:
Thank you for your late reply but all of the problems solved by changing the Ubuntu 18 to Ubuntu 16 and ROS melodic to ROS Kinetic..