realsenseai/realsense-ros #1192: D435 keep faling and resetting

gachiemchiep, 2020-05-09:
Hello guys. 

Currently my camera is having the following error when using in ROS. 

```bash
[ WARN] [1589005487.127092164]: Hardware Notification:REC error,1.58901e+12,Error,Hardware Error
[ WARN] [1589005488.127736227]: Hardware Notification:REC error,1.58901e+12,Error,Hardware Error
[ WARN] [1589005489.128154373]: Hardware Notification:REC error,1.58901e+12,Error,Hardware Error
[ WARN] [1589005490.128476922]: Hardware Notification:REC error,1.58901e+12,Error,Hardware Error
[ WARN] [1589005491.129013577]: Hardware Notification:REC error,1.58901e+12,Error,Hardware Error
 09/05 15:24:51,538 WARNING [139851282233088] (backend-v4l2.cpp:973) Frames didn't arrived within 5 seconds
 09/05 15:24:51,538 WARNING [139851290625792] (backend-v4l2.cpp:973) Frames didn't arrived within 5 seconds
[ WARN] [1589005492.129417712]: Hardware Notification:REC error,1.58901e+12,Error,Hardware Error
[ WARN] [1589005493.130044461]: Hardware Notification:Depth stream start failure,1.58901e+12,Error,Hardware Error
[ERROR] [1589005493.130176210]: Hardware Reset is needed. use option: initial_reset:=true
 09/05 15:24:54,420 ERROR [139851282233088] (types.h:304) xioctl(VIDIOC_DQBUF) failed for fd: 26 Last Error: No such device
 09/05 15:24:54,420 ERROR [139851290625792] (types.h:304) xioctl(VIDIOC_DQBUF) failed for fd: 23 Last Error: No such device
 09/05 15:24:54,420 ERROR [139851299018496] (types.h:304) xioctl(VIDIOC_DQBUF) failed for fd: 18 Last Error: No such device
 09/05 15:24:54,421 ERROR [139851282233088] (backend-v4l2.cpp:1295) xioctl(VIDIOC_DQBUF) failed for fd: 26 Last Error: No such device
 09/05 15:24:54,421 ERROR [139851290625792] (backend-v4l2.cpp:1295) xioctl(VIDIOC_DQBUF) failed for fd: 23 Last Error: No such device
 09/05 15:24:54,421 ERROR [139851299018496] (backend-v4l2.cpp:1295) xioctl(VIDIOC_DQBUF) failed for fd: 18 Last Error: No such device
[ WARN] [1589005494.421388477]: Hardware Notification:xioctl(VIDIOC_DQBUF) failed for fd: 26 Last Error: No such device,1.58901e+12,Error,Unknown Error
[ WARN] [1589005494.421507599]: Hardware Notification:xioctl(VIDIOC_DQBUF) failed for fd: 18 Last Error: No such device,1.58901e+12,Error,Unknown Error
[ WARN] [1589005494.421616167]: Hardware Notification:xioctl(VIDIOC_DQBUF) failed for fd: 23 Last Error: No such device,1.58901e+12,Error,Unknown Error
 09/05 15:24:54,451 ERROR [139851424843520] (types.h:304) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/05 15:24:54,451 ERROR [139851424843520] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/05 15:24:54,454 ERROR [139851401365248] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/05 15:24:54,454 ERROR [139851401365248] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/05 15:24:55,451 ERROR [139851424843520] (types.h:304) Cannot open '/dev/video0 Last Error: No such file or directory
 09/05 15:24:55,452 ERROR [139851424843520] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: Cannot open '/dev/video0 Last Error: No such file or directory
 09/05 15:24:55,454 ERROR [139851401365248] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/05 15:24:55,455 ERROR [139851401365248] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/05 15:24:56,452 ERROR [139851424843520] (types.h:304) Cannot open '/dev/video0 Last Error: No such file or directory
 09/05 15:24:56,452 ERROR [139851424843520] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: Cannot open '/dev/video0 Last Error: No such file or directory
 09/05 15:24:56,455 ERROR [139851401365248] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/05 15:24:56,455 ERROR [139851401365248] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/05 15:24:56,560 ERROR [139851273840384] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/05 15:24:56,560 ERROR [139851273840384] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[ERROR] [1589005496.765641346]: The device has been disconnected!
 09/05 15:24:56,897 ERROR [139851541288704] (types.h:304) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 09/05 15:24:56,974 ERROR [139851541288704] (types.h:304) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 09/05 15:24:56,975 ERROR [139851541288704] (types.h:304) xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
[ INFO] [1589005496.978613267]: Checking new devices...
[ INFO] [1589005496.978667786]:  
[ INFO] [1589005496.996141091]: Device with serial number 819612072507 was found.

```

I tried to switch into realsense-viewer. This time, the error message look different but the camera still isn't usable

```bash
:~/workspace/catkin_ws$ realsense-viewer 
 09/05 15:25:53,255 WARNING [140462392334080] (sensor.cpp:365) Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Y12I Y8I  Z16  ]
 09/05 15:25:53,261 WARNING [140462392334080] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 09/05 15:25:53,261 WARNING [140462392334080] (sensor.cpp:365) Unregistered Media formats : [ RW16 ]; Supported: [ RW16 YUYV ]
 09/05 15:25:59,210 WARNING [140462232938240] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.

```

Do you guys have any idea about this ?

MartyG-RealSense, 2020-05-09:
It looks like a kernel related issue.  You may have more success if you build Librealsense using the "bypass" method.  It requires an internet connection to install but is not dependent on Linux versions, kernel versions or patches and so can help to correct problems that occur when using the usual Librealsense installation process.

A typical CMake build command that uses the bypass, launched from the **Librealsense > Build** folder, would look like this;

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

gachiemchiep, 2020-05-09:
Hello @MartyG-RealSense  
What is the role of DFORCE_RSUSB_BACKEND? 

Today, I switch the old realsense with a newer one (same D435 model, but larger serial number) and the above problem is gone.  ....

MartyG-RealSense, 2020-05-09:
The link below discusses the advantages and disadvantages of the backend bypass method of installation.

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

gachiemchiep, 2020-05-11:
Hello @MartyG-RealSense . 
Thanks i will take a look

closed 2020-05-11 by gachiemchiep