realsenseai/realsense-ros #1193: Error while starting camera

malapatiravi, 2020-05-09:
Camera D415
OS: Ubuntu 16.04
Kernel: 4.15.0-99-generic
librealsense: 2.33.1

Followed all the instructions from 
https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

Once i build catkin workspace and run the following command
roslaunch realsense2_camera rs_camera.launch I get the following error
```
[ INFO] [1589062675.203426114]: Setting Dynamic reconfig parameters.
 09/05 22:18:00,232 WARNING [140563834771200] (types.cpp:49) get_xu(id=11) failed! Last Error: Input/output error
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  get_xu(id=11) failed! Last Error: Input/output error
```
Please let me know if i am doing something wrong. 
I tried with librealsens installation using the following config 
`cmake ../ -DBUILD_EXAMPLES=true -DENFORCE_METADATA=false -DBUILD_PYTHON_BINDINGS=true -DFORCE_LIBUVC=false -DBUILD_WITH_CUDA=true -DIMPORT_DEPTH_CAM_FW=false -DBUILD_WITH_TM2=false -DCMAKE_BUILD_TYPE=Release` 
and also with default configuration 
`cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false` taken from 
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

```
roslaunch realsense2_camera rs_camera.launch
... logging to /home/dummy/.ros/log/edc66228-9242-11ea-9fc0-b42e99a4904a/roslaunch-everestlabs-3967.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dummy:40015/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
[... 72 lines cut ...]
[ INFO] [1589062675.202093979]: Device FW version: 05.12.03.00
[ INFO] [1589062675.202178175]: Device Product ID: 0x0AD3
[ INFO] [1589062675.202277370]: Enable PointCloud: Off
[ INFO] [1589062675.202351669]: Align Depth: Off
[ INFO] [1589062675.202419555]: Sync Mode: Off
[ INFO] [1589062675.202563389]: Device Sensors: 
[ INFO] [1589062675.202732118]: Stereo Module was found.
[ INFO] [1589062675.202836333]: RGB Camera was found.
[ INFO] [1589062675.202978703]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589062675.203044224]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589062675.203080384]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589062675.203115366]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589062675.203179032]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589062675.203252909]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589062675.203361948]: num_filters: 0
[ INFO] [1589062675.203426114]: Setting Dynamic reconfig parameters.
 09/05 22:18:00,232 WARNING [140563834771200] (types.cpp:49) get_xu(id=11) failed! Last Error: Input/output error
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  get_xu(id=11) failed! Last Error: Input/output error
[camera/realsense2_camera_manager-2] process has died [pid 4007, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/everestlabs/.ros/log/edc66228-9242-11ea-9fc0-b42e99a4904a/camera-realsense2_camera_manager-2.log].
log file: /home/everestlabs/.ros/log/edc66228-9242-11ea-9fc0-b42e99a4904a/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/everestlabs/.ros/log/edc66228-9242-11ea-9fc0-b42e99a4904a/camera-realsense2_camera-3*.log

```

MartyG-RealSense, 2020-05-10:
After the catkin workspace was set up, following the Travis setup guidelines, was the ***- catkin_make install** instruction run to perform the catkin install process, please?

- catkin_make clean
  - catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
  **- catkin_make install**

malapatiravi, 2020-05-10:
@MartyG-RealSense 
catkin_make install completed without any issues. 
Here is the sample log
```
-- Installing: /home/dummy/catkin_ws/install/share/common-lisp/ros/realsense2_camera/msg/Extrinsics.lisp
-- Installing: /home/dummy/catkin_ws/install/share/common-lisp/ros/realsense2_camera/msg/IMUInfo.lisp
-- Installing: /home/dummy/catkin_ws/install/share/common-lisp/ros/realsense2_camera/msg/_package.lisp
-- Up-to-date: /home/dummy/catkin_ws/install/share/gennodejs/ros/realsense2_camera
-- Installing: /home/dummy/catkin_ws/install/share/gennodejs/ros/realsense2_camera/_index.js
-- Up-to-date: /home/dummy/catkin_ws/install/share/gennodejs/ros/realsense2_camera/msg
-- Installing: /home/dummy/catkin_ws/install/share/gennodejs/ros/realsense2_camera/msg/Extrinsics.js
-- Installing: /home/dummy/catkin_ws/install/share/gennodejs/ros/realsense2_camera/msg/IMUInfo.js
-- Installing: /home/dummy/catkin_ws/install/share/gennodejs/ros/realsense2_camera/msg/_index.js
Listing /home/dummy/catkin_ws/devel/lib/python2.7/dist-packages/realsense2_camera ...
Listing /home/dummy/catkin_ws/devel/lib/python2.7/dist-packages/realsense2_camera/msg ...
Compiling /home/dummy/catkin_ws/devel/lib/python2.7/dist-packages/realsense2_camera/msg/_Extrinsics.py ...
Compiling /home/dummy/catkin_ws/devel/lib/python2.7/dist-packages/realsense2_camera/msg/_IMUInfo.py ...
Compiling /home/dummy/catkin_ws/devel/lib/python2.7/dist-packages/realsense2_camera/msg/__init__.py ...
-- Up-to-date: /home/dummy/catkin_ws/install/lib/python2.7/dist-packages/realsense2_camera
-- Up-to-date: /home/dummy/catkin_ws/install/lib/python2.7/dist-packages/realsense2_camera/__init__.py
-- Up-to-date: /home/dummy/catkin_ws/install/lib/python2.7/dist-packages/realsense2_camera/__init__.pyc
-- Up-to-date: /home/dummy/catkin_ws/install/lib/python2.7/dist-packages/realsense2_camera/msg
-- Installing: /home/dummy/catkin_ws/install/lib/python2.7/dist-packages/realsense2_camera/msg/_IMUInfo.pyc
-- Installing: /home/dummy/catkin_ws/install/lib/python2.7/dist-packages/realsense2_camera/msg/__init__.py
-- Installing: /home/dummy/catkin_ws/install/lib/python2.7/dist-packages/realsense2_camera/msg/_Extrinsics.py
-- Installing: /home/dummy/catkin_ws/install/lib/python2.7/dist-packages/realsense2_camera/msg/_Extrinsics.pyc
-- Installing: /home/dummy/catkin_ws/install/lib/python2.7/dist-packages/realsense2_camera/msg/_IMUInfo.py
-- Installing: /home/dummy/catkin_ws/install/lib/python2.7/dist-packages/realsense2_camera/msg/__init__.pyc
-- Installing: /home/dummy/catkin_ws/install/lib/pkgconfig/realsense2_camera.pc
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/cmake/realsense2_camera-msg-extras.cmake
-- Installing: /home/dummy/catkin_ws/install/share/realsense2_camera/cmake/realsense2_cameraConfig.cmake
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/cmake/realsense2_cameraConfig-version.cmake
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/package.xml
-- Installing: /home/dummy/catkin_ws/install/lib/librealsense2_camera.so
-- Set runtime path of "/home/dummy/catkin_ws/install/lib/librealsense2_camera.so" to ""
-- Up-to-date: /home/dummy/catkin_ws/install/include/realsense2_camera
-- Up-to-date: /home/dummy/catkin_ws/install/include/realsense2_camera/base_realsense_node.h
-- Up-to-date: /home/dummy/catkin_ws/install/include/realsense2_camera/constants.h
-- Up-to-date: /home/dummy/catkin_ws/install/include/realsense2_camera/t265_realsense_node.h
-- Up-to-date: /home/dummy/catkin_ws/install/include/realsense2_camera/realsense_node_factory.h
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/rs_t265.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/rs_from_file.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/rs_aligned_depth.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/rs_d400_and_t265.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/demo_t265.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/rs_rgbd.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/opensource_tracking.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/rs_rtabmap.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/includes
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/includes/nodelet.launch.xml
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/rs_multiple_devices.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/rs_d435_camera_with_model.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/demo_pointcloud.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/launch/rs_camera.launch
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/rviz
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/rviz/pointcloud.rviz
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/rviz/t265.rviz
-- Up-to-date: /home/dummy/catkin_ws/install/share/realsense2_camera/nodelet_plugins.xml

```

MartyG-RealSense, 2020-05-10:
Thank you very much for the confirmation.  And you are running the roslaunch instruction whilst in the **catkin_ws** directory, yes?

malapatiravi, 2020-05-10:
> Thank you very much for the confirmation. And you are running the roslaunch instruction whilst in the **catkin_ws** directory, yes?
Yes it was from catkin_ws directory

MartyG-RealSense, 2020-05-10:
Thank you.   I see that you have the most recent camera firmware (5.12.3) installed, so there is no need to suggest that you update the firmware.

The input/output error makes me think that the problem might be USB related.  What computer hardware are you running the program on please, and are you using a mains electricity powered USB hub or plugging the camera directly into the computer.

malapatiravi, 2020-05-10:
> Thank you. I see that you have the most recent camera firmware (5.12.3) installed, so there is no need to suggest that you update the firmware.
>
> The input/output error makes me think that the [...]

Please find the hw information. The cameras are directly plugged to the computer. 
```
*-usb
             description: USB controller
             product: Cannon Lake PCH USB 3.1 xHCI Host Controller
             vendor: Intel Corporation
             physical id: 14
             bus info: pci@0000:00:14.0
             version: 10
             width: 64 bits
             clock: 33MHz
             capabilities: pm msi xhci bus_master cap_list
             configuration: driver=xhci_hcd latency=0
             resources: irq:122 memory:55320000-5532ffff
           *-usbhost:0
                product: xHCI Host Controller
                vendor: Linux 4.15.0-99-generic xhci-hcd
                physical id: 0
                bus info: usb@1
                logical name: usb1
                version: 4.15
                capabilities: usb-2.00
                configuration: driver=hub slots=16 speed=480Mbit/s
              *-usb:0
                   description: Keyboard
                   product: USB Receiver
                   vendor: Logitech
                   physical id: 3
                   bus info: usb@1:3
                   version: 29.01
                   capabilities: usb-2.00
                   configuration: driver=usbhid maxpower=98mA speed=12Mbit/s
              *-usb:1
                   description: USB hub
                   product: USB 2.0 Hub
                   vendor: Terminus Technology Inc.
                   physical id: 5
                   bus info: usb@1:5
                   version: 1.00
                   capabilities: usb-2.00
                   configuration: driver=hub maxpower=100mA slots=2 speed=480Mbit/s
              *-usb:2
                   description: USB hub
                   product: USB3.0 Hub
                   vendor: Realtek
                   physical id: 7
                   bus info: usb@1:7
                   version: 1.26
                   capabilities: usb-2.00
                   configuration: driver=hub slots=4 speed=480Mbit/s
              *-usb:3
                   description: Keyboard
                   product: ITE Device(8595)
                   vendor: ITE Tech. Inc.
                   physical id: d
                   bus info: usb@1:d
                   version: 0.03
                   capabilities: usb-2.00
                   configuration: driver=usbhid maxpower=100mA speed=12Mbit/s
              *-usb:4
                   description: Bluetooth wireless interface
                   vendor: Intel Corp.
                   physical id: e
                   bus info: usb@1:e
                   version: 0.02
                   capabilities: bluetooth usb-2.00
                   configuration: driver=btusb maxpower=100mA speed=12Mbit/s
           *-usbhost:1
                product: xHCI Host Controller
                vendor: Linux 4.15.0-99-generic xhci-hcd
                physical id: 1
                bus info: usb@2
                logical name: usb2
                version: 4.15
                capabilities: usb-3.10
                configuration: driver=hub slots=10 speed=10000Mbit/s
              *-usb:0 UNCLAIMED
                   description: Video
                   product: Intel(R) RealSense(TM) Depth Camera 415
                   vendor: Intel(R) RealSense(TM) Depth Camera 415
                   physical id: 5
                   bus info: usb@2:5
                   version: 50.c3
                   serial: 846623023430
                   capabilities: usb-3.20
                   configuration: maxpower=440mA speed=5000Mbit/s
              *-usb:1
                   description: USB hub
                   product: USB3.0 Hub
                   vendor: Realtek
                   physical id: 7
                   bus info: usb@2:7
                   version: 1.26
                   capabilities: usb-3.00
                   configuration: driver=hub slots=4 speed=5000Mbit/s
              *-usb:2 UNCLAIMED
                   description: Video
                   product: Intel(R) RealSense(TM) Depth Camera 415
                   vendor: Intel(R) RealSense(TM) Depth Camera 415
                   physical id: 8
                   bus info: usb@2:8
                   version: 50.c3
                   serial: 850623021790
                   capabilities: usb-3.20
                   configuration: maxpower=440mA speed=5000Mbit/s

```

MartyG-RealSense, 2020-05-10:
I was thinking of the type of computer you have when I asked about the hardware.  Do you have PC / Mac / Android / Raspberry Pi / Jetson etc.  Are you using the camera in a USB 2 or USB 3 port.  That kind of thing.  Thanks.

malapatiravi, 2020-05-10:
> I was thinking of the type of computer you have when I asked about the hardware. Do you have PC / Mac / Android / Raspberry Pi / Jetson etc. Are you using the camera in a USB 2 or USB 3 port. That kind of thing. Thanks.

its PC running Ubuntu.

MartyG-RealSense, 2020-05-10:
I could only find one reference case that is very similar to the exact error with input/output that you are experiencing.

https://github.com/IntelRealSense/librealsense/issues/2029

The eventual advice in that case was to create an app that resets the camera when the error occurs and then try re-running the launch file.  

https://github.com/IntelRealSense/librealsense/issues/2029#issuecomment-414712195

You could probably test it without this script though by unplugging the camera after the error occurs, waiting a couple of seconds and then plugging it back in and running the roslaunch command again.

malapatiravi, 2020-05-10:
> without this script though

```
rs2::pipeline pipe;
rs2::pipeline_profile profile = pipe.start();
auto dev = profile.get_device();
dev.hardware_reset();
```
Running the above script and restarting the system seems work.
Thanks for the help.

MartyG-RealSense, 2020-05-11:
That's great news!  Thanks so much for the update.  :)

MartyG-RealSense, 2020-05-24:
@malapatiravi   Do you require any further assistance or can this case be closed, please?   Thanks!

closed 2020-05-24 by malapatiravi