realsenseai/realsense-ros #1194: ros depth scale unit

chillcloud-dev, 2020-05-11:
Hello realsense team,

previously I am using librealsense to access all necessary camera streams. It works perfectly. But since the realsense is blocked that way and other applications can not access them I am using your ROS wrapper to publish RGBD information. 

Now there is one issue regarding the depth information. i need to obtain the depth unit scale. I don't want to hard-coded. It is around 0.001 m.

Previously i was getting this value by running 
`  auto depth_units = rs2::context().query_devices().front().query_sensors().front().get_option(RS2_OPTION_DEPTH_UNITS);
`

but since i am using ROS i can not find this value anywhere. 

Could you tell me where i find this depth scale ? 

Thanks

MartyG-RealSense, 2020-05-11:
I researched this subject carefully but could not find a single simple, clear answer.  

https://github.com/IntelRealSense/realsense-ros/issues/714

Apparently the complication is in the RealSense camera scale (1mm) versus the convention used by ROS (uint16).

https://github.com/IntelRealSense/realsense-ros/issues/714#issuecomment-479907272

@doronhi the RealSense ROS wrapper developer offers a Python script for ROS in this same discussion.  I think it's for printing the depth value in mm.

https://github.com/IntelRealSense/realsense-ros/issues/714#issuecomment-561036465

Andyshen555, 2020-05-27:
I'd like to know if there is a way to change the depth scale when recording through the launch file? Or is there some other methods? Previously I change it to 0.0001 in the realsense viewer, which works great for me.

MartyG-RealSense, 2020-05-27:
Hi @Andyshen555  The RealSense ROS wrapper developer explains reasons in the link below why the depth scale cannot be changed in RealSense ROS.

https://github.com/IntelRealSense/realsense-ros/issues/277#issuecomment-525676873

Andyshen555, 2020-05-27:
Got it, thx Marty.

MartyG-RealSense, 2020-06-05:
Case closed due to no further comments received and answer provided.

closed 2020-06-05 by MartyG-RealSense