realsenseai/realsense-ros #1195: Disable/Enable stream and topic publication at runtime ?

doisyg, 2020-05-11:
Hello,
Is there any service or dynamic parameters that I missed that will allow disabling / re enabling the publications of the sensors topics (video streams and pointcloud) at runtime. This would be very convenient to save CPU and in order to have the sensor active only when needed.
If not, any pointers of where to start ?

MartyG-RealSense, 2020-05-11:
If you need to be able to toggle topics on and off at will, one approach that was taken to this problem was to use a ROS node called **mux** to create two publishers, one that sends a topic's data and one that doesn't, and switch between them.

https://answers.ros.org/question/257561/stop-publishing-a-specific-topic/

http://wiki.ros.org/topic_tools/mux

doisyg, 2020-05-11:
Well the goal is mainly to save CPU resources and the camera lifetime by effectively disabling the sensors (and the laser emitter which wears over time I believe), so it has to be done at the driver level.
I currently unsubscribe from the topics when I don't use them

MartyG-RealSense, 2020-05-11:
In the case of the Laser Emitter, if you set "Laser Power" to '0' then it turns the emitter off.

https://github.com/IntelRealSense/librealsense/issues/5464#issuecomment-569487713

pmirabel, 2020-05-18:
Hello, thanks for the tips.
My need is to enable / disable rgb stream "as in realsense-viewer". Do you guys have some hints to achieve that ?

I can see in **realsense_node_factory.cpp**:
```
auto pipe = std::make_shared<rs2::pipeline>();
rs2::config cfg;
cfg.enable_device_from_file(rosbag_filename.c_str(), false);
cfg.enable_all_streams();
```

That make me think that right now, ros_wrapper do not give any possiblity of en/disable stream after initialization. 
Any hints?

MartyG-RealSense, 2020-05-18:
I have responded to your question on the main GitHub.  After receiving the answer there, do you still wish to continue this ROS GitHub conversation please?   Thanks!

doisyg, 2020-05-18:
@MartyG-RealSense , we still need need a way to enable/diable streams from the ROS wrapper in a similar way as in realsense-viewer, so this should stay open until a PR adds this feature

MartyG-RealSense, 2020-05-18:
I gave the subject further consideration.  Once the realsense2_camera node has been started with roslaunch,  I wonder if you could stop it by killing the process with:

**rosnode kill realsense2_camera**

Then after the camera is stopped by the node kill, you could do the roslaunch instruction again to restart publishing of the camera topics.

doronhi, 2020-05-26:
Currently the wrapper does not support a service or a parameter to enable/disable a sensor. It could be done but it has implications like enabling/disabling dynamic parameter and probably some more consequences I don't foresee right now.
As already mentioned, saving CPU can be done by unsubscribing the topics.
Avoiding overheating can be done by disabling the emitter.
As for the camera lifetime, I don't know that it is related.
In order to prioritize this feature request I would like to be convinced that it has some practical need, that cannot be obtained otherwise.
Thanks,
Doron

pmirabel, 2020-06-02:
Hello and thank you all.

> As already mentioned, saving CPU can be done by unsubscribing  topics.
> Avoiding overheating can be done by disabling the emitter.
> As for the camera lifetime, I don't know that it is related.

1. I was not aware of how librealsense works accurately at the time of my last answer. I have dig into the code and made some tests. I was not aware that RGB module can be used by software A while Depth module is used by software B at the same time.

2. Regarding the CPU usage, I made some optimisation to my ROS stack and that's indeed way better.

@doisyg do you have any usecase where it could be usefull ?

MartyG-RealSense, 2020-06-18:
Hi @pmirabel   Do you wish to continue with this case or can it be closed, please?   Thanks!

doisyg, 2020-06-18:
> In order to prioritize this feature request I would like to be convinced that it has some practical need, that cannot be obtained otherwise.

The use case is quite simple: disabling the camera streams in a simple way when there are not in use (for instance on a docked robot), and reenabling them quickly (killing/restarting the whole node is too slow/messy), in the same way it is done on the realsense-viewer tool.

> As already mentioned, saving CPU can be done by unsubscribing the topics.
> Avoiding overheating can be done by disabling the emitter.

Yes, this would work, but the unsubscribtion has to be done on the (multiple) subcribers side and an unsubscribtion mechanism has to be implemented for every subscriber. Moreover, this is not trully disabling the camera streams and the USB usage as in with the realsense-viewer tool, just the publication of the data on the ROS side.

MartyG-RealSense, 2020-06-18:
@doisyg   Thanks for your further comments!

@doronhi  Could you provide input on the new comments of @doisyg above please?   Thanks!

doronhi, 2020-06-23:
I agree that it could be done, the way it is done in the realsense-viewer. It's possible that it's not hard to implement. I'm not even objecting this feature and I see it's beauty and value. I merely try to prioritize it as our hands are full.
Considering that one can unsubscribe, and maybe it's even a better practice then cutting the pipe beneath the feet of the subscribing nodes, I don't feel that this is feature is urgent. As such I'll happily consider a PR but for me to bring it to the top and implement it, I expect it will take a while.

MartyG-RealSense, 2020-07-05:
@doisyg   As moving this issue forward will require someone to program some code to disable / enable stream and create a Pull Request (PR) to propose merging it into the ROS wrapper, I will close this issue as there is no way to know when / if anybody will do it in future.

If someone reading this wants to create a PR and is able to, please creare a new issue on this GitHub about it and create a matching PR for your code contribution using the link below.   Thanks everyone!

https://github.com/IntelRealSense/realsense-ros/pulls

closed 2020-07-05 by MartyG-RealSense