realsenseai/realsense-ros #1196: Create 2D map

intel-user, 2020-05-13:
Hi every one, I'm trying SLAM with D435i camera following this tutorial [https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i]. I used this command: 
**`rosrun map_server map_saver map:=/rtabmap/proj_map –f my_map_1`**
to create 2D map like tutorial, terminal display something like this 
[image] then nothing happened. 
I also wanna run rtabmap_ros with intel d435i like this
[image]
Can you give me the instruction for these?
Thank you!!!

MartyG-RealSense, 2020-05-13:
Hi @intel-user   I only see the **rosrun** statement on your terminal.  The step before that would be to launch realsense2_camera with the **relaunch** statement, otherwise nothing can happen.  Did you do this please?

**roslaunch realsense2_camera opensource_tracking.launch**

intel-user, 2020-05-13:
Hi @MartyG-RealSense, I already install 4 components and launch realsense2_camera but it still nothing happen.
[image]

MartyG-RealSense, 2020-05-13:
Thanks for the image above.  Your on screen log is full of error warning messages.  It looks as though you are experiencing the same **synthetic stream** related errors that were reported in other cases in April.

https://github.com/IntelRealSense/realsense-ros/issues/1161
https://github.com/IntelRealSense/realsense-ros/issues/1181

To check whether the errors are occurring because your computer is having difficulty coping with the amount of data that is being processed, you could try reducing the FPS speed by modifying the frame rate in the launch file to be '15':

<arg name="depth_fps"           default="15"/>
  <arg name="infra_fps"           default="15"/>
  <arg name="color_fps"           default="15"/>

intel-user, 2020-05-14:
hi @MartyG-RealSense, thanks for your reply.
I already reduce FPS to be 15 but is seem nothing change, I can't create 2D map
[image]

By the way, can you instruct for me how to use rtabmap display robot's path like this?
[image]
Thanks for your help!

MartyG-RealSense, 2020-05-14:
The fact that you are receiving a 3D point cloud image would seem to indicate that the topics are being successfully published despite the errors in the log (which I would expect to be able to happen so long as the process has not "died").  I can see that you are successfully receiving a color image in RViz too.

The way that the terminal keeps offering advice about -h (help guidance) and suggestions for instruction structuring suggests to me that RViz is having trouble understanding the **rosrun map_server** instruction that you inputted, even though it is the same instruction as the one in Intel's D435i SLAM guide.

Looking at other examples on this map_saver command on the internet, everyone else seems to put the **rtabmap/proj** command on the end of the statement instead of the middle like the D435i SLAM guide does.  Try this please:

**rosrun map_server map_saver -f my_map_1 map:=/rtabmap/proj_map**

MartyG-RealSense, 2020-05-30:
Do you require further assistance please or can this case be closed?   Thanks!

closed 2020-06-05 by MartyG-RealSense