realsenseai/realsense-ros #1197: Read auto-exposure values

piotrpolatowski, 2020-05-13:
Hello,
are you planning to allow querying the sensor? 
I'm interested in reading the exposure time for enabled auto exposure, like https://github.com/IntelRealSense/librealsense/issues/3717
This would be very convenient to let realsense adjust the exposure time and later collect series of images with same settings. We want to compare images over time and dynamic exposure causes us problems, whereas picking right exposure to a new environment seems like a complicated task which is solved on the device.

Best regards
Piotr

MartyG-RealSense, 2020-05-13:
I wonder if a solution might already exist that includes the **dynamic reconfigure** system for changing parameters dynamically during runtime instead of in the launch file.

RealSense camera parameters, including those of the RGB and stereo cameras, can be dynamically affected in ROS using the Dynamic Reconfigure (**rqt_reconfigure**) interface.

https://github.com/IntelRealSense/realsense-ros#set-camera-controls-using-dynamic-reconfigure-params

A **rosparam get /** instruction may therefore be able to read the current settings of Dynamic Reconfigure.

https://github.com/IntelRealSense/realsense-ros/issues/609#issuecomment-463438354

piotrpolatowski, 2020-05-14:
I have tested **dynamic_reconfigure** on `REALSENSE_ROS_VERSION=2.2.8` and it always returns the same exposure value. 

My impression is that the sensor properties are read on the node start, but later are not updated.
https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp#L189

MartyG-RealSense, 2020-05-14:
It may be the auto-exposure being enabled that is preventing the exposure reading from being updated.  The "real" exposure value is in the camera metadata.  The link below has an explanation from a member of the RealSense team about the mechanics of the exposure system.

https://github.com/IntelRealSense/librealsense/issues/1624#issuecomment-389085387

On Windows 10, metadata support is built in.  For Linux users, the support has to be patched in.

https://github.com/IntelRealSense/librealsense/issues/5596#issuecomment-575775702

There is a "bypass" method of installing the Librealsense SDK on a computing device with an internet connection that is not dependent on Linux versions, kernel versions or patches.  Metadata support is included in that installation method, so it is one of the easiest ways to get metadata support.  It can be installed by using a CMake instruction like the one below in the **Build** folder of the Librealsense root directory (Librealsense > Build):

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

piotrpolatowski, 2020-05-14:
Indeed, the librealsense allows for reading the exposure from the device. 

What I am wondering is, if this feature could be included in the ROS driver, for example get the real exposure value using **dynamic reconfigure**. We use vanilla ROS node and would prefer to keep it this way.

MartyG-RealSense, 2020-05-14:
As I am on the RealSense support team and not the developer team (@doronhi is the main RealSense ROS developer), I will label this case as a Feature Request.  Hopefully @doronhi can also provide advice on your request, as I have name-tagged him to this discussion.

doronhi, 2020-05-27:
I agree that this is a nice enhancement. I'll appreciate a PR on the subject or look into it myself as time and priorities permit.

piotrpolatowski, 2020-05-31:
@doronhi I could prepare a PR, only if this was supported by hardware. With the latest firmware `5.12.3`, depth works well, but RGB always return same value, similarly as reported in 2018 https://github.com/IntelRealSense/librealsense/issues/2549.

Do you know if RGB will support reading exposure metadata from frame?

doronhi, 2020-06-01:
I checked and it is a know limitation with RGB sensor. The automatic and manual exposure settings are separate and I don't know of a way to obtain the value of the calculated auto-exposure value for the D4xx series.

piotrpolatowski, 2020-06-01:
Thanks a lot. As for frame exposure time for depth, it could be published to an independent topic next to camera_info. 
I could prepare a PR if you find it useful. We could also document there, that RGB is not currently supported and save few more questions in the future.

doronhi, 2020-06-01:
I wonder if it would be possible to occasionally update the dynamic parameter instead of adding another topic. Seems more intuitive to me.

MartyG-RealSense, 2020-06-14:
Hi @piotrpolatowski   Do you wish to continue with this case please, or can it be closed?  Thanks!

MartyG-RealSense, 2020-07-05:
Case closed due to no further comments received.   @piotrpolatowski   If you do decide to kindly create a PR at a future date and submit it, please create a new issue on this GitHub to go with it.  Thanks!

closed 2020-07-05 by MartyG-RealSense