realsenseai/realsense-ros #1199: Mean Intensity Set Point for Auto Exposure

manomitbal, 2020-05-14:
Hi,

I am using version 2.30 for librealsense and was wondering if there was an exposed parameter to tune the Mean Intensity Set Point to determine the range of intensities to determine auto exposure value as shown in librealsense viewer screenshot below.

[image]

I have noticed that the slider is not movable and has a set static value in the viewer, but my intention is to procure and set these values through realsense-ros, is there a way to do that through the current setup? 

Thanks!

MartyG-RealSense, 2020-05-14:
The approach that I would recommend to set values that are part of **Advanced Mode** is to define a visual preset file (also known as a **json**) that sets the Mean Intensity Set Point to the desired value.  Then you could try calling the json file from your ROS launch file.

https://github.com/IntelRealSense/realsense-ros/issues/411#issuecomment-477463559

I believe that the json command for setting auto-exposure set point would be **aux-param-autoexposure-setpoint**.  Here is an example:

**"aux-param-autoexposure-setpoint": "400"**

An easy way to define a custom json file is to set the values up in the RealSense Viewer program and then use the json export option on the toolbar at the top of the Viewer's options side-panel.

RealSenseCustomerSupport, 2020-05-15:
----------------------------------------------

as Marty recommended, you can export the setting to json file
and even there is no slide bar for this setting
you can still edit its value by click the edit icon at right side  to enter text edit mode to tweak its value manually

manomitbal, 2020-05-15:
This worked! Thanks guys!

closed 2020-05-15 by manomitbal

MartyG-RealSense, 2020-05-15:
Great news - thanks for letting us know!   :)

ghhhgfgh, 2025-12-03:
Hello. Can I know what is the parameters name of “aux-param-autoexposure-setpoint” in ros2 humble? I cannot see the parameter in ros2 parameter list. Thank you