realsenseai/realsense-ros #1200: Failed to load nodelet '/camera/driver` of type `realsense_camera/R200Nodelet`

jagauthier, 2020-05-15:
Greetings,

I've followed these instructions [](http://wiki.ros.org/realsense_camera/Tutorials/Building_librealsense_from_Sources)
with few deviatations.  

I'm using the system librealsense (which I built from source).  
I'm using Melodic in a debian system, with an Intel R200 camera.
So, the version of librealsense I am using is actually 1.12, from the legacy branch.

Once I've built the ROS package, I tried to launch it:
`roslaunch realsense_camera r200_nodelet_default.launch`

I get the following output:

started roslaunch server http://192.168.80.159:36060/
```

SUMMARY
========

PARAMETERS
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: R200
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/fisheye_frame_id: camera_fisheye_frame
 * /camera/driver/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/driver/imu_frame_id: camera_imu_frame
 * /camera/driver/imu_optical_frame_id: camera_imu_optica...
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir2_optical_frame_id: camera_ir2_optica...
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/ir_optical_frame_id: camera_ir_optical...
 * /camera/driver/mode: manual
 * /camera/driver/serial_no:
 * /camera/driver/usb_port_id:
 * /rosdistro: sys.path: ['/opt/...
 * /rosversion: sys.path: ['/opt/...

NODES
  /camera/
    driver (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [27709]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cdb92a66-96b0-11ea-b655-b2188dd1b44b
process[rosout-1]: started with pid [27720]
started core service [/rosout]
process[camera/nodelet_manager-2]: started with pid [27727]
process[camera/driver-3]: started with pid [27728]
[ INFO] [1589549672.217063629]: Initializing nodelet with 4 worker threads.
[FATAL] [1589549674.847983927]: Failed to load nodelet '/camera/driver` of type `realsense_camera/R200Nodelet` to manager `nodelet_manager'
[camera/nodelet_manager-2] process has died [pid 27727, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/uav122/.ros/log/cdb92a66-96b0-11ea-b655-b2188dd1b44b/camera-nodelet_manager-2.log].
log file: /home/uav122/.ros/log/cdb92a66-96b0-11ea-b655-b2188dd1b44b/camera-nodelet_manager-2*.log
[camera/driver-3] process has died [pid 27728, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense_camera/R200Nodelet nodelet_manager depth:=depth color:=color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/uav122/.ros/log/cdb92a66-96b0-11ea-b655-b2188dd1b44b/camera-driver-3.log].
log file: /home/uav122/.ros/log/cdb92a66-96b0-11ea-b655-b2188dd1b44b/camera-driver-3*.log

```

I've been fighting with this for awhile, and I'm not sure where to take my next troubleshooting steps.  This output, nor the logs provide any further information about why this is exiting.

MartyG-RealSense, 2020-05-15:
I wonder if the problem is with incompatibility between realsense_camera and Melodic.  Even the current RealSense ROS wrapper for Librealensense did not receive official Melodic support until recently.   The realsense_camera web-page indicates that it supports the earlier **Kinetic** ROS version.

[image]

http://wiki.ros.org/realsense_camera

Here is the same indicator for reasense2_camera, showing its support for both Kinetic and Melodic:

[image]

jagauthier, 2020-05-15:
Hm, I'll try Kinetic!  I didn't catch that

jagauthier, 2020-05-15:
Hey, thanks for catching that!  I built Kinetic, and deployed it to my application and sure enough.. I am getting somewhere!

closed 2020-05-15 by jagauthier

MartyG-RealSense, 2020-05-15:
Awesome news - thanks for the update.   Good luck!   :)