realsenseai/realsense-ros #1201: RPI 4 Buster D435i Not Working

Zarbokk, 2020-05-17:
Hello,
I am currently not making the depth camera d435i work with my RPI4 buster.
In Short: Realsense-Viewer Works, Ros doesnt. on RPI4 Raspberrian Buster with d435i. t265 also works

What did i do?
I installed the realsense Library, with the realsense-viewer from here: [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md](url)
This works fine. I can see the depth image. 

Then second I installed ROS-Melodic from source for RPI Buster.
additionally i installed realsense-ros library, Which compiled.

After Starting the camera, the Topics shown do not publish anything and the error on the bottom is shown.

Additionally the t265 camera works perfectly, which is even more weird.
And on my main computer the camera also works well.

What did i try to solve the error:

- reinstall everything
- Updating driver of d435i
- Changing settings in the launch file of the camera.
- Using usb3 or usb2 on the PI

My current thought is maybe the power supply? which i cant easily try out currently. 
Does Anyone have an idea?

`[ INFO] [1589718601.728351020]: Initializing nodelet with 4 worker threads.
[ INFO] [1589718601.883100065]: RealSense ROS v2.2.13
[ INFO] [1589718601.883341077]: Running with LibRealSense v2.34.0
[ INFO] [1589718601.949496831]:  
 17/05 13:30:02,459 WARNING [2874897424] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed.
Error type: No data to return (-21).
 17/05 13:30:02,459 WARNING [2874897424] (ds5-motion.cpp:464) IMU Calibration is not available, see the previous message
[ INFO] [1589718602.512191210]: Device with serial number 947122070852 was found.

[ INFO] [1589718602.512447667]: Device with physical ID 2-1-8 was found.
[ INFO] [1589718602.513614581]: Device with name Intel RealSense D435I was found.
[ INFO] [1589718602.516032427]: Device with port number 2-1 was found.
[ INFO] [1589718602.525016605]: getParameters...
[ INFO] [1589718602.685142105]: setupDevice...
[ INFO] [1589718602.685333323]: JSON file is not provided
[ INFO] [1589718602.685426783]: ROS Node Namespace: d435i
[ INFO] [1589718602.685520151]: Device Name: Intel RealSense D435I
[ INFO] [1589718602.685606464]: Device Serial No: 947122070852
[ INFO] [1589718602.685684758]: Device physical port: 2-1-8
[ INFO] [1589718602.686149228]: Device FW version: 05.12.03.00
[ INFO] [1589718602.686272410]: Device Product ID: 0x0B3A
[ INFO] [1589718602.686883635]: Enable PointCloud: On
[ INFO] [1589718602.687507064]: Align Depth: On
[ INFO] [1589718602.688049457]: Sync Mode: On
[ INFO] [1589718602.688356579]: Device Sensors: 
[ INFO] [1589718602.689077412]: Stereo Module was found.
[ INFO] [1589718602.689281092]: RGB Camera was found.
[ INFO] [1589718602.689431422]: Motion Module was found.
[ INFO] [1589718602.690289622]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589718602.690433359]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589718602.690547949]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589718602.690654650]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1589718602.691613422]: Add Filter: hole_filling
[ INFO] [1589718602.694550569]: Add Filter: pointcloud
[ INFO] [1589718602.697273851]: num_filters: 2
[ INFO] [1589718602.697567455]: Setting Dynamic reconfig parameters.
[ INFO] [1589718608.078880810]: Done Setting Dynamic reconfig parameters.
[ INFO] [1589718608.130671599]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1589718608.134263285]: setupPublishers...
[ INFO] [1589718608.146659839]: Expected frequency for depth = 30.00000
[ INFO] [1589718608.162238459]: setupStreams...
[ INFO] [1589718608.214207780]: insert Depth to Stereo Module
[ INFO] [1589718608.214597492]: insert Gyro to Motion Module
[ INFO] [1589718608.214791006]: insert Accel to Motion Module
[ INFO] [1589718608.416249880]: SELECTED BASE:Depth, 0
[ INFO] [1589718608.440968934]: RealSense Node Is Up!
 17/05 13:30:08,546 WARNING [2839528464] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ WARN] [1589718609.089747698]: Hardware Notification:REC error,1.58972e+12,Error,Hardware Error
 17/05 13:30:12,618 ERROR [2715804688] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
`

MartyG-RealSense, 2020-05-17:
Hi @Zarbokk  To begin, could you tell me the **roslaunch** instruction that you are using please?   Thanks!

Zarbokk, 2020-05-17:
Sure 
In serial_no i added the serial number. 
When using the t265 i use a different one.

```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="d435i"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="true"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="640"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default="hole_filling"/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye"           value="$(arg enable_fisheye)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
      <arg name="depth_height"             value="$(arg depth_height)"/>
      <arg name="enable_depth"             value="$(arg enable_depth)"/>

      <arg name="color_width"              value="$(arg color_width)"/>
      <arg name="color_height"             value="$(arg color_height)"/>
      <arg name="enable_color"             value="$(arg enable_color)"/>

      <arg name="infra_width"              value="$(arg infra_width)"/>
      <arg name="infra_height"             value="$(arg infra_height)"/>
      <arg name="enable_infra1"            value="$(arg enable_infra1)"/>
      <arg name="enable_infra2"            value="$(arg enable_infra2)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>
</launch>
```

MartyG-RealSense, 2020-05-17:
Sorry, I meant the roslaunch instruction.  Like **roslaunch realsense2_camera rs_camera.launch**.  The reason that I ask is that there have been a couple of ROS cases recently that reported having problems with publishing a point cloud in RViz with the **pointcloud** filter:

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**

For example, the RGBD point cloud example in the wrapper documentation:

https://github.com/IntelRealSense/realsense-ros#rgbd-point-cloud

Zarbokk, 2020-05-17:
Maybe I understand that wrong, but exactly that is what the roslaunch file is doing right?
It sets the pointcloud on true.

To clarify I use the above launchfile with:
**roslaunch realsense2_camera rs_camera.launch**

Which should be equal with your example of:
**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**

What I tried first, was the classical way of starting the camera with the provided launch file after installation:
**roslaunch realsense2_camera rs_camera.launch**
This results in the error: 
Time is out of Dual 32-bit range

Which is also an issue from here:
Duplicate of #1178

MartyG-RealSense, 2020-05-17:
Experiencing errors such as these is not necessarily an obstacle to topics being successfully published and able to be visualized in RViz so long as the errors only appear once after launch and are not being generated continuously.  If they appear once or twice at launch, it may just be because the camera is still settling down.

Errors are more likely to occur on "low power" devices such as single-board computers than they are on laptop and desktop computers, hence why it is common that a program may work fine on a full computer but have errors when run on the kind of low-power single board used for robotics projects, such as Pi and Jetson.

Using a mains electricity powered USB hub can provide extra power stability on the USB port because the power supplied by the mains power socket can be more reliable than that available from a port on the board, or an unpowered USB hub attached to one of the computer's ports.  It is a good idea to budget up to 2W of power for a camera, and a mains powered hub can provide 12W to share among the devices attached to it.  

If you would prefer to try for a solution that does not require a powered hub though, the first small step I would recommend is that if your program does not require the D435i and the T265 to work together then the code that queries a serial number could be removed as a serial number is not needed for a single device, just to distinguish between multiple cameras being used at the same time.

If you prefer to keep some form of identification, then the ROS wrapper provides an alternative to serial number based ID that was developed in response to problems with using serial numbers in ROS.  You can instead identify devices by **usb_port_id**

https://github.com/IntelRealSense/realsense-ros#launch-parameters

Zarbokk, 2020-05-17:
Firstly, I dont want to use multiple Cameras, but only the d435i

> Experiencing errors such as these is not necessarily an obstacle to topics being successfully published and able to be visualized in RViz so long as the errors only appear once after launch and are [...]
I cant get any topic information. Gyro/accelerometer or any point cloud/image is not getting published.

> Errors are more likely to occur on "low power" devices such as single-board computers than they are on laptop and desktop computers, hence why it is common that a program may work fine on a full [...]

Do you think the missing power is the problem?
I can only test that in a few days, since I am at home. 

I dont understand the Serial number point.

MartyG-RealSense, 2020-05-17:
Mains powered USB hubs have a history of often making RealSense streams more stable and USB related errors disappear, though they are not a guaranteed fix in every case.  For example, there was a recent case where a Jetson Xavier board had a powered hub but could not work with more than one camera plugged into the hub.  Investigations by a RealSense user suggested that the failure was related to serial numbers rather than power supply.  The user offered a script workaround.

https://github.com/IntelRealSense/librealsense/issues/5828#issuecomment-622993107

Zarbokk, 2020-05-18:
When not using the serial number im back at the duplicate of:
Duplicate of #1178
With the 32 bit error: Time is out of Dual 32-bit range

MartyG-RealSense, 2020-05-18:
It is difficult to advise on this particular error, as @doronhi the RealSense ROs developer has suggested that it occurs due to something in ROS that is outside of Librealsense.

Some ROS users have fixed this error by cleaning their catkin workspace.  

If you just wanted to clean up the **realsense2_camera** packagae you could try this command after cd'ing to the catkin workspace folder:

**catkin_make clean --pkg realsense2_camera**

There is a **Catkin Tools** tool for aiding the catkin clean process:

https://catkin-tools.readthedocs.io/en/latest/

Do not try it though if you do not feel confident in carrying out the procedure.

closed 2020-06-14 by MartyG-RealSense