realsenseai/realsense-ros #1205: Plans to integrate On-Chip Calibration into ROS wrapper

YoshuaNava, 2020-05-19:
Dear developers,
RealSense cameras are awesome devices, with great support and potential. Nonetheless, one of the pain points when building robots with the RealSense cameras is that their inner extrinsic parameters need to be calibrated from time to time to guarantee that the depth estimation is at its best.
Even though the dynamic calibration GUI is user friendly and nicely designed, it relies on OpenGL for rendering. If the user intent is to use these cameras on a mobile robot, and at some point needs to run calibration, then he/she will need to:
1. Have GUI components and libraries installed on the robot computer, and stream the UI to the operator PC.
2. Run the dynamic calibration and perform indirect rendering so that it can be visualized in an external PC, through an SSH connection. This is in some cases quite difficult to achieve, if the operator PC and the robot computers have different graphic cards and/or driver versions.

It would be really useful, if the calibration of D400 devices could be run on robot platforms, without the need of launching a GUI. Do you have any plans to integrate this functionality into the RealSense ROS wrapper?

MartyG-RealSense, 2020-05-19:
Hi @YoshuaNava   If maintaining calibration is important to your projects then the new RealSense L515 lidar depth camera may be useful to you, as it retains its depth accuracy throughout its lifespan without requiring calibration.

Another approach to professional-level calibration is to use the $1500 OEM Calibration Target system available from the official RealSense Store.  The OEM Dynamic Calibrator for the 400 Series cameras can calibrate both intrinsics and extrinsics on multiple cameras and allows remote connection to cameras through a Linux server.  

For mobile robots, an implementation of this system could conceivably be to place the calibration target board at a robot home position calibrated towards the board so that the robots can be recalibrated when not in active service.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

Also, although On-Chip Calibration can be accessed through the GUI of the RealSense Viewer, it can also be invoked through API scripting for **Python**, **C++** or **LabVIEW**.  Please read downward from the point in the Self Calibration White Paper linked to below to find details of these script-based activations.  

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-appendix-c-on-chip-calibration-python-api

The YouTube tutorial video in the link below introduces how to call C++ and Python scripts from ROS.

https://www.youtube.com/watch?v=a3ldvpnOyog&feature=youtu.be

YoshuaNava, 2020-05-19:
Hi @MartyG-RealSense,

Thank you for the thorough and prompt answer.

Regarding the Calibration Target, it solves a completely different use case than the On-Chip calibration. Then you are relying on infrastructure, and you need a complex setup. For my application it is not ideal.

Thank you for the reference to the On-Chip Calibration API, it is really nice.

Do you know if Intel aims to integrate this into ROS at some point? It would simplify the usage of the cameras quite a bit. 
I ask first in this repo because I think this could benefit other users, and it is better if new features are considered upstream before you implement them in your own fork and the software starts diverging. 
If you guide me through the RealSense-ROS workflow, I would also be willing to help with ideas or implementations for integrating the calibration API in the node.

Best,
Yoshua

MartyG-RealSense, 2020-05-19:
The RealSense ROS wrapper is not developed in sync with the release schedule of the Librealsense SDK, so wrapper updates that support new features in the SDK tend to come at some point after the SDK version's release.  

I do not have information about the ROS wrapper's development roadmap, as I am on the Support team.  The wrapper developer @doronhi may be able to advise you about whether built-in support for the on-chip calibration is planned in the near future.

RealSense ROS community members are welcome to submit their own code for consideration for integration into the wrapper through a "pull request".

https://github.com/IntelRealSense/realsense-ros/pulls

Thanks!

doronhi, 2020-05-20:
Hi @YoshuaNava ,
I am not familiar with "dynamic calibration GUI". I expect it's possible to create a calibration tool that connect to the camera using the realsense2_camera wrapper for input and add some features to the wrapper to enable a service for writing the results back to the camera. 

Or, do you find the on-chip calibration as exist in RealSense Viewer appropriate for your needs? In that case, is enabling it in the ROS wrapper as a service will be considered a solution the needs to describe?

I prefer the second solution as it simpler and requires much less attention from the end user if any at all.

YoshuaNava, 2020-05-20:
Hi @doronhi 
By dynamic calibration GUI I refer to this application: https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide#section-4-5-2-running-dynamic-calibrator-with-targeted-calibration

Enabling the On-Chip calibration in the ROS wrapper as a service (the second option) would be really useful, more than the first one.

Would it be OK if we introduce this change? If so, we should probably define a user workflow for running it. 
E.g. the RealSense viewer on-chip calibration option asks the user for his approval of a given calibration. Would we keep this level of user interaction?

Best,
Yoshua

doronhi, 2020-05-21:
Hi @YoshuaNava ,
I encourage you to introduce this change.
I think asking the user for approval is not necessary and, as you say, requires too much interaction and possibly even some GUI. I think we should introduce 3 services: 
1. on-chip calibration. 
2. undo last on-chip calibration.
3. reset calibration to the original calibration as it was when the driver (realsense2_camera node) started running.

I think that way the user may decide to accept or reject the calibration based on available ros-viewers or other parameters of his choice.
What do you think?

YoshuaNava, 2020-06-01:
@doronhi I will start working on this during this week. I will get back to you with any questions that may arise.

YoshuaNava, 2020-06-05:
@doronhi I now have a sensor to start the implementation and run tests.

I have a few questions:

a) Is it possible to save the current calibration to some standard location when the ROS wrapper is started? E.g. $HOME/.ros/ 
Which functions of the librealsense API can be used for this?

b) Which format is the preferred one for backing up an existing calibration?

c) How often does the on-chip calibration reach a good result? 
  - How much confidence do we have in the calibration score?

Thank you.

Best,
Yoshua

YoshuaNava, 2020-06-22:
Dear @doronhi,

I submitted a PR to the librealsense repo contaning a self calibration example in C++: https://github.com/IntelRealSense/librealsense/pull/6600

This afternoon I will submit another PR to this repo. I'll keep you updated.

Best,
Yoshua

MartyG-RealSense, 2020-08-08:
Hi @YoshuaNava   Your kindly submitted pull has a "failing check".  Can you go to the pull page and take a look at it please?   Thanks!

MartyG-RealSense, 2020-09-03:
Hi once again @YoshuaNava Can you follow up on the "failing check" of your pull mentioned above please? Thanks!

YoshuaNava, 2020-09-03:
Dear @MartyG-RealSense 
Sorry for the delay in responding.

My other PR for the realsense-ros was closed, as the on chip-calibration was integrated into the Python examples.

Are the maintainers still positive about my PR to librealsense? If that's the case I'll address the failing check ASAP.

Once again, sorry for going radio silent.

Best,
Yoshua Nava

MartyG-RealSense, 2020-09-03:
Hi @YoshuaNava   Thanks for the response!  The pull https://travis-ci.org/github/IntelRealSense/librealsense/builds/698688707 still has 1 failing check related to Python and NodeJS:

[image]

It looks as though a reviewer has not been assigned to the pull yet, so there are no developer comments.

MartyG-RealSense, 2020-09-18:
Hi @YoshuaNava   There appears to be no new information  available regarding your pull.  Please let us know what you decide to do with the pull.   Thanks very much.

YoshuaNava, 2020-09-24:
@MartyG-RealSense 

Sorry, I have been a bit busy with other tasks. I will resume this, but I can't give an estimate atm.

MartyG-RealSense, 2020-09-24:
Thanks for the update.   It may be better to close the pull for now and re-submit it when you are able to work on it again.   Thanks!

YoshuaNava, 2020-09-24:
Ok. Thanks

MartyG-RealSense, 2020-10-08:
Closing this ticket due to long-term pull project inactivity.  The pull itself remains open.

closed 2020-10-08 by MartyG-RealSense