realsenseai/realsense-ros #1210: i can't get orientation data by /camera/imu

jacobye1, 2020-05-27:
camera:D435i
 i set the "rs_camera.launch" <arg name="unite_imu_method"          default="linear_interpolation"/>
but i don't know how can i get orientation data,the orientation data is full with 0.
---
header: 
  seq: 1318122
  stamp: 
    secs: 1590548261
    nsecs:  57455301
  frame_id: "camera_imu_optical_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00174532923847
  y: -0.00174532923847
  z: 0.0
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: -0.225066577601
  y: -8.52246474629
  z: 4.16831254368
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
---
header: 
  seq: 1318123
  stamp: 
    secs: 1590548261
    nsecs:  59951305
  frame_id: "camera_imu_optical_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00349065847695
  y: -0.00174532923847
  z: 0.0
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: -0.217745052508
  y: -8.5219783783
  z: 4.16001829373

Can you tell me how to get this orientation data?

doronhi, 2020-05-27:
The node does not publish orientation. The IMU on the D435i provides linear acceleration and angular velocity. The fields are only part of the standard ROS Imu message and the (-1) in the orientation_covariance field mark it invalid.
To obtain orientation you a node that integrate the values together. Preferably filtering noise and taking images into consideration. Basically, a SLAM node.

jacobye1, 2020-05-27:
> The node does not publish orientation. The IMU on the D435i provides linear acceleration and angular velocity. The fields are only part of the standard ROS Imu message and the (-1) in the [...]

Okay, that means I need to write a node if I need orientation?I want send the imu data which is from /camera/imu to node 'robot_pose_ekf'.Do you have some idea?

doronhi, 2020-05-27:
You most definitely don't have to write your own node. I am not a specialist on that area and am not familiar with the latest packages available so I preferred you do your own research rather then put you on a wrong track.
For start you can check out the file "opensource_tracking.launch" (part of the realsense2_camera package). It's an example I wrote once but I'm pretty sure it needs corrections and one can do better.

jacobye1, 2020-05-27:
> You most definitely don't have to write your own node. I am not a specialist on that area and am not familiar with the latest packages available so I preferred you do your own research rather then [...]

well,i will find a packages if it exist .I have read the file"opensource_tracking.launch",but now i cannot launch this node，so can you tell me what the data which be sended by node imu_filter_madgwick?

doronhi, 2020-05-27:
Sounds right but it's been a while...
Why can't you launch the node? Can you find help in this [SLAM-with-D435i wiki page](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i)?

jacobye1, 2020-05-27:
> Sounds right but it's been a while...
> Why can't you launch the node? Can you find help in this [SLAM-with-D435i wiki page](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i)?

ooops,the node can work now!!!!
i echo the /imu/data topic,maybe my guess is right.
---
header: 
  seq: 45807
  stamp: 
    secs: 1590570299
    nsecs:  90885162
  frame_id: "camera_imu_optical_frame"
orientation: 
  x: 0.70889838521
  y: 0.255775364355
  z: -0.241723329327
  w: 0.611237985154
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.267035365105
  y: -1.11352002621
  z: -2.75238418579
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: 3.07096815484
  y: -10.9999166212
  z: 0.338300882929
linear_acceleration_covariance: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
---
header: 
  seq: 45705
  stamp: 
    secs: 1590570298
    nsecs: 853271961
  frame_id: "camera_imu_optical_frame"
orientation: 
  x: 0.681965451241
  y: 0.138087855767
  z: -0.160967818981
  w: 0.699960162192
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 1.39451801777
  y: 0.438077628613
  z: 6.03883934021
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: -1.98080642958
  y: 29.2136165684
  z: -0.173528339196
linear_acceleration_covariance: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
---

RealSenseSupport, 2020-05-29:
Hi @jacobye1 Great! Do you need further help on this?
If we don't hear from you in 7 days, this issue will be closed.

closed 2020-06-05 by RealSenseSupport

MrOCW, 2021-01-05:
Hi @doronhi , I am a beginner in ROS and am using the D435i. I am currently using the command roslaunch realsense2_camera rs_rgbd.launch to determine the x y z coordinates of an object with respect to the camera's axes. (x right, y down, z forward) I have successfully obtained said coordinates, however, using this coordinates, I need to transform the camera's frame to match the object's frame so i can localize the camera in the world coordinate. May I know how i can do so without using another launch file as the current rs_rgbd.launch is required for me to obtain the object's coordinate with respect to the camera's axes.

doronhi, 2021-01-12:
I am not sure I can help but, sorry, I didn't understand the question. 
Did you mean you want to avoid using the rs_rgbd.launch file?
Could you try to rephrase the question?

MrOCW, 2021-01-12:
> I am not sure I can help but, sorry, I didn't understand the question.
> Did you mean you want to avoid using the rs_rgbd.launch file?
> Could you try to rephrase the question?

Sorry issue has been solved. Thanks for the reply!

jacobye1, 2021-01-12:
> I am not sure I can help but, sorry, I didn't understand the question.
> Did you mean you want to avoid using the rs_rgbd.launch file?
> Could you try to rephrase the question?

But I haven't been in work this project for a long time. At that time, I used rs_ rgbd.launch  File is to get IMU data to do Kalman filtering to help me correct the odom of gmapping. Later, I used a separate IMU sensor