realsenseai/realsense-ros #1211: Unstable inclusion of two cameras Realsense D435 and D435i

amburkoff, 2020-05-27:
Computer: NVIDIA Jetson AGX Xavier - Jetpack 4.3 [L4T 32.3.1]
D435 firmware: 05.12.03.00
D435i firmware: 05.12.03.00
Two cables https://www.amazon.com/TetherPro-USB-C-1-8m-High-Visibility-Orange/dp/B0794BL8CH/ref=sr_1_15?dchild=1&keywords=TetherPro+1.8&qid=1590582032&sr=8-15 inserted into two ports type-c in Jetson directly without an additional usb hub.
Ubuntu 18.04
Enable jetson_clocks.sh
LibRealSense  v2.33.1
RealSense ROS v2.2.13

Both camera devices are visible
[realsense_list.txt](https://github.com/IntelRealSense/realsense-ros/files/4688670/realsense_list.txt)

Launched a startup file 
[realsense_multiple.txt](https://github.com/IntelRealSense/realsense-ros/files/4688707/realsense_multiple.txt)

[realsense_error2.txt](https://github.com/IntelRealSense/realsense-ros/files/4688682/realsense_error2.txt) - the first camera started, and the second camera did not start.

[realsense_error3.txt](https://github.com/IntelRealSense/realsense-ros/files/4688681/realsense_error3.txt) - The first camera did not start, and the second camera started.

[realsense_error4.txt](https://github.com/IntelRealSense/realsense-ros/files/4688680/realsense_error4.txt) - the first camera started, and the second camera did not start.

[realsense_error5.txt](https://github.com/IntelRealSense/realsense-ros/files/4688679/realsense_error5.txt) - launch file launch failed.

[realsense_error6.txt](https://github.com/IntelRealSense/realsense-ros/files/4688678/realsense_error6.txt) - cameras did not start.

Pulled out and again inserted usb cable on both cameras

[realsense_error7.txt](https://github.com/IntelRealSense/realsense-ros/files/4688677/realsense_error7.txt) - both cameras started up

[realsense_error8.txt](https://github.com/IntelRealSense/realsense-ros/files/4688676/realsense_error8.txt) - both cameras started up

[realsense_error9.txt](https://github.com/IntelRealSense/realsense-ros/files/4688675/realsense_error9.txt)  - the one camera started, and the second camera did not start.
[realsense_list9.txt](https://github.com/IntelRealSense/realsense-ros/files/4688669/realsense_list9.txt)

[realsense_error10.txt](https://github.com/IntelRealSense/realsense-ros/files/4688674/realsense_error10.txt)  - the one camera started, and the second camera did not

They left one camera connected, the second camera was physically disabled.

[realsense_error11.txt](https://github.com/IntelRealSense/realsense-ros/files/4688673/realsense_error11.txt) - launch file launch failed.

[realsense_error12.txt](https://github.com/IntelRealSense/realsense-ros/files/4688672/realsense_error12.txt) - it works, but it gives errors in the terminal

[realsense_error13.txt](https://github.com/IntelRealSense/realsense-ros/files/4688671/realsense_error13.txt) - it works, but it gives errors in the terminal

[realsense_list14.txt](https://github.com/IntelRealSense/realsense-ros/files/4688847/realsense_list14.txt) - The camera started working, worked for a while, then an error occurred "27/05 15:44:51,024 ERROR [545645384064] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 131" and the camera stopped sending an image
[realsense_error14.txt](https://github.com/IntelRealSense/realsense-ros/files/4688849/realsense_error14.txt)

MartyG-RealSense, 2020-05-27:
The RealSense developer team are working on upcoming improvements to multiple camera use that should hopefully help with some of the multiple-camera problems that Jetson users have encountered.

https://github.com/IntelRealSense/librealsense/pull/6467

dorodnic, 2020-05-27:
It may. ROS aspect of this should be covered in #5800 once its completed and merged
(ROS nodes are usually separate processes hence requiring more software synchronization)

MartyG-RealSense, 2020-06-05:
Hi @amburkoff   Do you require further assistance with this case please or can it be closed?   Thanks!

RealSenseSupport, 2021-01-19:
Closing this issue. Thank you.

closed 2021-01-19 by RealSenseSupport