realsenseai/realsense-ros #1212: Realsense R200 IR camera, no data published

jagauthier, 2020-05-27:
Greetings,

  I am using some Visual SLAM methods and discovered that the IR cameras are not being published.
```
[ INFO] [1590590160.190992206]: Initializing nodelet with 4 worker threads.
Intel RealSense F200_camera ; 2.60.0.0
Intel RealSense LR200_camera ; 2.0.71.18
Intel RealSense R200_camera ; 1.0.72.06
Intel RealSense SR300_camera ; 3.10.10.0
Intel RealSense ZR300_adapter ; 1.29.0.0
Intel RealSense ZR300_camera ; 2.0.71.28
Intel RealSense ZR300_motion_module ; 1.25.0.0
[ INFO] [1590590160.512758259]: /camera/driver - Detected the following camera:
                                - Serial No: 2381007980, USB Port ID: 2-4, Name: Intel RealSense R200, Camera FW: 1.0.71.06
[ WARN] [1590590160.512861648]: /camera/driver - Detected unvalidated firmware:
                                - 2381007980's current camera firmware is 1.0.71.06, Validated camera firmware is 1.0.72.06
[ INFO] [1590590160.512937724]: /camera/driver - Connecting to camera with Serial No: 2381007980, USB Port ID: 2-4
[ INFO] [1590590160.778515794]: /camera/driver - Setting static camera options
[ INFO] [1590590160.821594288]: /camera/driver - Enabling Depth in manual mode
[ INFO] [1590590160.822421062]: /camera/driver - Enabling Color in manual mode
[ INFO] [1590590160.823581440]: /camera/driver - Starting camera
[ INFO] [1590590160.847988706]: /camera/driver - Publishing camera transforms (/tf_static)
[ INFO] [1590590160.848530401]: /camera/driver - Setting dynamic camera options (r200_dc_preset=5)
[ INFO] [1590590161.518739867]: /camera/driver - Initializing Depth Control Preset to 5
[ INFO] [1590590165.250638045]: /camera/driver - Setting dynamic camera options
```

```
$ rostopic list|grep camera
/camera/color/camera_info
/camera/color/image_raw
/camera/depth/camera_info
/camera/depth/image_raw
/camera/depth/points
/camera/driver/parameter_descriptions
/camera/driver/parameter_updates
/camera/ir/camera_info
/camera/ir/image_raw
/camera/ir2/camera_info
/camera/ir2/image_raw

```

Everything is there.  However, nothing returns from:
`rostopic echo /camera/ir/camera_info`

and rviz says there is no image returned from either IR or IR2.

This also just hangs
`rostopic echo /camera/ir/image_raw`

MartyG-RealSense, 2020-05-27:
Hi @jagauthier   Your description does not make clear whether you have **Librealsense1** installed (the legacy version of Librealsense that is compatible with the camera models you are using) and most importantly, the **realsense_camera** RealSense ROS node.

Usually, the streams from a RealSense camera would be published in ROS by doing a roslaunch of the realsense_camera node.  The color streams may be an exception to this if they can be accessed as though the cameras are RGB webcams.

**Librealsense 1 (Legacy Librealsense)**
https://github.com/IntelRealSense/librealsense/tree/v1.12.1

**realsense_camera**
http://wiki.ros.org/realsense_camera

jagauthier, 2020-05-27:
I am using the legacy version of realsense, in order to support the R200.   I am also using the **realsense_camera** ROS node.

I omitted some of the output that would have helped clarify that.
```

$ roslaunch realsense_camera r200_nodelet_default.launch
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.17.0.2:41803/

SUMMARY
========

PARAMETERS
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: R200
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/fisheye_frame_id: camera_fisheye_frame
 * /camera/driver/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/driver/imu_frame_id: camera_imu_frame
 * /camera/driver/imu_optical_frame_id: camera_imu_optica...
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir2_optical_frame_id: camera_ir2_optica...
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/ir_optical_frame_id: camera_ir_optical...
 * /camera/driver/mode: manual
 * /camera/driver/serial_no:
 * /camera/driver/usb_port_id:
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    driver (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)

```

I hope that clarifies anything.  Both color and depth cameras are accessible through the topics, but the IR cameras are not.

MartyG-RealSense, 2020-05-27:
I have not used legacy Librealsense / realsense_camera, so I apologise in advance for any errors.

If you have multiple cameras attached to the computer at the same time then you can use this launch statement:

**roslaunch realsense_camera r200_nodelet_multiple_cameras.launch**

You could also try launching your R200 in RGBD mode with:

**roslaunch realsense_camera r200_nodelet_rgbd.launch**

jagauthier, 2020-05-27:
Thanks. I checked into the RBGD one, but unfortunately it is incomplete.  The launch references files that exist, and causes an exception.
I pulled down the current git tree and those files aren't there either. So who knows.

Not sure why the IR cameras are not accessible.  I guess I could try to publish them using the v4l module and their respective /dev/video* files

MartyG-RealSense, 2020-05-27:
May I confirm that the R200 cameras are being used in a USB 3 port, as that era of cameras do not work with USB 2.

jagauthier, 2020-05-27:
It is connected to an Intel Aero Compute board, on the USB3 interface.  I am able to access the IR cameras using models this device in Simulink.  Just seems not within ROS.

I am able to access them with the built in SDK example applications like cpp-headless, and cpp-capture

MartyG-RealSense, 2020-05-27:
The Intel Aero Compute Board typically had a flavor of Linux called Yocto Project installed for the purposes of controlling an Intel Ready To Fly drone with it.  I believe it was also possible to install the full Ubuntu on it too though and that its hardware specification was comparable to the original Up Board.

The Aero web-page now provides links to setting up Ubuntu 16.04 and ROS Kinetic on the board, suggesting that it should be able to run ROS.  You could check the way in which you installed ROS against the instructions in that guide if you have not read the guide already, to see if there are any special Aero-specific installation steps required.

https://github.com/intel-aero/meta-intel-aero/wiki

jagauthier, 2020-05-28:
So, as it turns out I was missing a ROS package. I had to install `'ros-kinetic-rgbd-launch'`.
After that RBGD was able to launch, and also the IR cameras have data.  Thank you for the help!

closed 2020-05-28 by jagauthier

MartyG-RealSense, 2020-05-28:
Great news - thanks so much for the update!   :)