realsenseai/realsense-ros #1213: How to set the optimal parameter value for D430 depth camera

dingbaiyin, 2020-05-28:
My project is using Intel D430 to collect images, and the target is pigs. The distance between the camera and the ground is within two meters. Image acquisition is performed 24 hours. During the daytime, the influence of the sun's rays causes black holes to appear in the depth image, and sometimes there are many small black spots on the depth image. At present, the image is collected in the automatic exposure mode, and the other parameters are the default values. At present, the filter is being installed to reduce the impact of sunlight and light on the depth image. How can I adjust each parameter to achieve the best depth image effect?
Sorry i didn't upload the picture successfully

MartyG-RealSense, 2020-05-28:
Hi @dingbaiyin   If you have the IR Emitter enabled and are projecting a dot pattern from the camera then you could try setting the "Laser Power" option to a low value such as '1' in the launch file to see whether it is the cause of the many small black dots.  This is because visibility of the IR dot pattern is tied to the Laser Power value.  A greater value increases visibility of the dots, a lower pattern reduces their visibility, and setting Laser Power to '0' turns off the projector (which can make an image darker, which is why a minimum of '1' is preferable).

 **<rosparam>
/camera/stereo_module/laser_power: 1
</rosparam>**  

The dot pattern is useful for helping the camera to analyse the depth detail of flat objects with low texture or no texture.  Given that the average pig's skin is neither flat nor as low-detailed as a wall or door, having the dot pattern visible may not be necessary.

You may be able to enhance the image with post-processing filters, such as the **Hole Filling Filter**.  These can be defined in the roslaunch instruction and then configured in a rosrun **dynamic reconfigure** statement immediately afterwards.

 The subject of setting filters in ROS is discussed in the link below:

https://github.com/IntelRealSense/realsense-ros/issues/742

For night-time conditions, some RealSense users add an LED illuminator.

https://github.com/IntelRealSense/librealsense/issues/2000

dingbaiyin, 2020-05-29:
Thank you very much for your reply, I will test it according to your suggestion。

dingbaiyin, 2020-06-01:
> 
> 
> Hi @dingbaiyin If you have the IR Emitter enabled and are projecting a dot pattern from the camera then you could try setting the "Laser Power" option to a low value such as '1' in the launch file to see whether it is the cause of the many small black dots. This is because visibility of the IR dot pattern is tied to the Laser Power value. A greater value increases visibility of the dots, a lower pattern reduces their visibility, and setting Laser Power to '0' turns off the projector (which can make an image darker, which is why a minimum of '1' is preferable).
> 
> ` <rosparam> /camera/stereo_module/laser_power: 1 </rosparam>`
> 
> The dot pattern is useful for helping the camera to analyse the depth detail of flat objects with low texture or no texture. Given that the average pig's skin is neither flat nor as low-detailed as a wall or door, having the dot pattern visible may not be necessary.
> 
> If you believe that the dot pattern is necessary, you could investigate using an external LED pattern projector instead of the laser-based projector that is built into the camera. LED projectors help to avoid noise on the image called **laser speckle**. Intel has published a white-paper guide document about using projectors with RealSense 400 Series cameras.
> 
> https://dev.intelrealsense.com/docs/projectors
> 
> You may be able to enhance the image with post-processing filters, such as the **Hole Filling Filter**. These can be defined in the roslaunch instruction and then configured in a rosrun **dynamic reconfigure** statement immediately afterwards.
> 
> The subject of setting filters in ROS is discussed in the link below:
> 
> #742
> 
> For night-time conditions, some RealSense users add an LED illuminator to the location where the camera is. The

dingbaiyin, 2020-06-01:
For the Intel D4XX camera in use, are there any recommended parameters that need to be adjusted and the effects of their functions? Are there parameters recommended by Controls and Advanced Controls?

MartyG-RealSense, 2020-06-01:
Intel has an excellent camera tuning guide at the link below.

https://www.intel.com/content/www/us/en/support/articles/000027833/emerging-technologies/intel-realsense-technology.html

The majority of Advanced Mode functions do not have documentation.  This is because they interact with each other in complex ways and so Intel controls them with 'machine learning' algorithms.  RealSense users are free to carry out trial and error experimentation with the settings to see how it affects the image though.

dingbaiyin, 2020-06-01:
Thanks for your reply. I still have a few image problems, I will re-initiate a request for help

MartyG-RealSense, 2020-06-18:
@dingbaiyin   Did you receive the help that you requested?  If you did, may this case be closed please?   Thanks!