realsenseai/realsense-ros #1214: ROS 1 Noetic Support

NathanDodson, 2020-05-28:
Hi there! My team uses RealSense cameras for our projects in ROS 1, so we were wondering if there is a planned port to ROS 1 Noetic Ninjemys. We are wanting to update our system to the Noetic for the long term support of the system. Let me know if there's a planned port!

Thanks!

BriceRenaudeau, 2020-05-29:
Hi,
I already use D435 cameras en Noetic.
You can build from source. :+1: 
And TF publisher/Depth point cloud seems to work with real cameras.

Unfortunately, there is some trouble with the URDF :
In `roslaunch realsense2_description view_d435_model.launch `
The launch uses the deprecated `[state_publisher]`
But even after fixing it, the TF tree is still not published, as if `arg use_nominal_extrinsics` is not used.

MartyG-RealSense, 2020-05-29:
Hi @BriceRenaudeau    Thanks so much for helping @NathanDodson and creating the pull https://github.com/IntelRealSense/realsense-ros/pull/1217 with a fix.  I appreciate it a lot!

reinzor, 2020-06-15:
Noetic release would be nice :)

MartyG-RealSense, 2020-06-18:
ROS wrapper **2.2.14** has now been released.   It does not contain official Noetic support.  You are welcome to try it though to see if it improves your Noetic experience, as the release contains URDF fixes.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.14

reinzor, 2020-06-18:
Thanks, we will use this release tag from source until it has been released into the Noetic debs!

MartyG-RealSense, 2020-07-05:
Case closed due to no further comments.

closed 2020-07-05 by MartyG-RealSense

chfritz, 2020-07-06:
Can I ask why this ticket was closed? The given explanation ("due to no further comments") does not seem to suggest that there are any plans to support Noetic. This is hard to believe but would be very concerning. So can I ask which ticket, if not this, Noetic support is tracked/discussed? Asked differently, what are the plans for supporting Noetic?

MartyG-RealSense, 2020-07-06:
If there are no further comments on a subject received after a few weeks, and no compelling reason to keep the discussion open if it is silent, then it qualifies to be closed. 

In this case, the discussion was closed because (a) librealsense is already useable with Noetic, and (b) URDF fixes for Noetic were added in the new 2.2.14 wrapper release.

https://github.com/IntelRealSense/realsense-ros/tree/development/realsense2_description/urdf

You can therefore be assured that the wrapper will receive Noetic updates, either officially or through user created "pull" submissions.

MartyG-RealSense, 2020-11-19:
Hi everyone, RealSense ROS wrapper 2.2.20 has added Noetic support.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.20