realsenseai/realsense-ros #1216: ROS 1 Noetic TF not published from URDF model

BriceRenaudeau, 2020-05-29:
Hi,
We use D435 cameras en Noetic.
The package builds from source. +1
And TF publisher/Depth point cloud seems to work with real cameras.

Unfortunately, there is some trouble with the URDF :
In roslaunch realsense2_description view_d435_model.launch
The launch uses the deprecated [state_publisher]
But even after fixing it, the TF tree is still not published.

MartyG-RealSense, 2020-05-31:
Hi @BriceRenaudeau  There is a Noetic branch for **robot_state_publisher** in the link below.  Is this of any use to you, please?

https://github.com/ros/robot_state_publisher

BriceRenaudeau, 2020-06-01:
HI, 
Yes, I was talking about `state_publisher` that has been deprecated and removed.
In the PR I changed it to `robot_state_publisher`.
Thanks for your concerns.

MartyG-RealSense, 2020-07-18:
Closed due to no further comments received.

closed 2020-07-18 by MartyG-RealSense