realsenseai/realsense-ros #1219: SR305 Incomplete bulk usb transfer

humberthumbert, 2020-05-30:
Hi, I am currently working on ubuntu 18.04 and trying to use roslaunch wrapper to use RealSense SR305.
When I run `roslaunch realsense2_camera rs_camera.launch ` after installation from [realsense-ros](https://github.com/IntelRealSense/realsense-ros#installation-instructions), I constantly get the following warning and error:

> [ INFO] [1590848584.522511572]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1590848584.522526979]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1590848584.522541681]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1590848584.522555239]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1590848584.522571069]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1590848584.522584296]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1590848584.522619732]: num_filters: 0
> [ INFO] [1590848584.522636072]: Setting Dynamic reconfig parameters.
> [ INFO] [1590848589.762034789]: Done Setting Dynamic reconfig parameters.
> [ INFO] [1590848589.794032466]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
> [ INFO] [1590848589.794497784]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
> [ WARN] [1590848589.794567827]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Width: 640, Height: 480, FPS: 30
> [ INFO] [1590848589.829309186]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
> [ INFO] [1590848589.829501801]: setupPublishers...
> [ INFO] [1590848589.831627534]: Expected frequency for depth = 30.00000
> [ INFO] [1590848589.851689771]: Expected frequency for infra1 = 30.00000
> [ INFO] [1590848589.865956279]: Expected frequency for color = 30.00000
> [ INFO] [1590848589.879857892]: setupStreams...
> [ INFO] [1590848589.910789552]: insert Depth to Coded-Light Depth Sensor
> [ INFO] [1590848589.910861225]: insert Color to RGB Camera
> [ INFO] [1590848589.910887796]: insert Infrared to Coded-Light Depth Sensor
> [ INFO] [1590848590.048271315]: SELECTED BASE:Depth, 0
> [ INFO] [1590848590.052181727]: RealSense Node Is Up!
>  30/05 22:23:10,550 WARNING [139878173636352] (types.cpp:49) Incomplete bulk usb transfer!
> [ERROR] [1590848590.550575927]: Incomplete bulk usb transfer!
> 

I have also tried to run the camera with realsense-viewer, however, what I got is following:

> INFO [140417743763200] (backend-v4l2.cpp:454) Cannot access /sys/class/video4linux

I have seen a closed issue mentioned about this situation, yet there is no follow up. So any thoughts would be appreciated. Thanks!

MartyG-RealSense, 2020-05-30:
Past cases of the **Cannot access /sys/class/video4linux** error with RealSense Viewer have been solved by doing a fresh reinstall of Ubuntu to clear any conflicts that might be occurring with previously installed files, such as different kernel versions.  

This is not convenient for everyone to do though, especially if they also have ROS packages and various other programs / files on their existing Ubuntu installation.  One RealSense user who was in this position said that they fixed the problem using the instruction below:

**sudo apt-get install realsense-uvcvideo**

humberthumbert, 2020-05-30:
Thank you for responding so quickly. But, there is no package called as uvbvideo.

> Unable to locate package realsense-uvcvideo

Also, do you know whether the bulk usb transfer error could be caused by the same reason as video4linux?

MartyG-RealSense, 2020-05-30:
Which method did you install Librealsense with please?  Did you build it from source or install a package?

Also, which computing device are you using the SR305 with please?   (PC, Mac, Jetson, Raspberry Pi, etc)

I would not immediately associate the bulk transfer usb error with video4linux - I would like to do further diagnosis first.

humberthumbert, 2020-05-30:
I installed `sudo apt-get install ros-melodic-realsense2-camera`. In the instruction, it said 

> This will install both realsense2_camera and its dependents, including librealsense2 library.

And I am working on a Lenovo Y510P where I have ubuntu and window dual boot.

MartyG-RealSense, 2020-05-30:
I am admittedly less familiar with this "Method 1" approach of installing the wrapper and Librealsense together.  I am more used to the "Method 2" approach of installing Librealsense first and then the ROS wrapper secondly.

It looks as though your ROS installation went okay though, since you are able to perform **roslaunch**.  

Was the RealSense Viewer included as part of the Librealsense installation that you created with the ROS installation, or did you install it separately afterwards?

humberthumbert, 2020-05-30:
I install the viewer separately afterward.

MartyG-RealSense, 2020-05-30:
It looks as though Method 1 is installing the ROS wrapper, and probably Librealsense too, using Debian packages instead of compiling them from source code.

 I believe that the Debian package installation also creates the example programs, which should include the RealSense Viewer, and that they are placed in the folder  **/usr/local/bin**

Could you see if you can find the Viewer in  /usr/local/bin and launch that version instead of the one that you downloaded seprately, please?

humberthumbert, 2020-05-30:
Emmm, mine is located in /usr/bin. I have run it. However, it seems like it cannot detect my RealSense camera.
[image]

Actually, I got the same output from roslaunch, saying that "No Realsense devices were found!".

MartyG-RealSense, 2020-05-30:
Is it still undetected if you unplug the camera, wait a couple of seconds and then plug it back in?

humberthumbert, 2020-05-30:
Previously, there were messages about video4linux, but now they have gone. Yet, the camera is still undetected.

MartyG-RealSense, 2020-05-30:
Is it still undetected if you unplug the micro-size end of the USB cable from the side of the camera, turn it upside down and re-insert it into the camera (USB-C is two-way insertion on the micro end) and then reboot the computer and re-launch the Viewer?

I appreciate your patience during this diagnostic process!

MartyG-RealSense, 2020-05-30:
As you have a dual-boot computer, would it be possible to install the RealSense SDK on Windows if you have not done so already?  The installation process is much easier on Windows and places a launch icon for the Viewer on the Windows desktop when installation completes.  This will help to determine whether the problem is with the camera hardware or with the Ubuntu installation specifically.

you can install Librealsense on Windows by visiting its Releases page with the link below, going to the bottom of the page and left-clicking on the indicated installer file, then downloading and running it.

I appreciate your patience during this diagnostic process!

https://github.com/IntelRealSense/librealsense/releases/tag/v2.34.0

[image]

humberthumbert, 2020-05-30:
Unfortunately, my windows is WIN8. But I have a MacBook on which I have had successfully run it.

MartyG-RealSense, 2020-05-30:
Okay, thank you very much.  That would indicate that the camera hardware and its USB camera are not at fault and that it is an issue related to the Ubuntu installation specifically.

What kernel version are you using, please?

humberthumbert, 2020-05-30:
5.3.0-53-generic

MartyG-RealSense, 2020-05-30:
Okay, thanks.  Librealsense does officially support Kernel 5.3 (and works with 5.4 unofficially).

Do you know how to use CMake, please?  You probably do if you have Librealsense on Mac.

humberthumbert, 2020-05-30:
Cannot say familiar with it, but I think I can have a try.

MartyG-RealSense, 2020-05-30:
When various types of problem occur with an Ubuntu build of Librealsense by traditional means of installation such as Debian packages or source code, there is a "backend" method of installation that can get around the problems and provide a working Librealsense.  This is because it is not dependent on Linux versions, kernel versions or patches.  It does require an internet connection to carry out the installation with this method though.

Basically, the program [https://cmake.org/](url) is used to build Librealsense.  Once you have CMake installed on your computer, you follow these steps:

1.  Go to the Librealsense **Releases** page and obtain the zipped file of the Librealsense source code from the Assets file list.  Unzip the folder.

[image]

2.  Go to the **Librealsense** folder and create a **Build** folder and change directory into it with this instruction:

**mkdir build && cd build**

3.  Use the CMake build instruction below to build Librealsense and all its dependencies without the need to do any patching:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

humberthumbert, 2020-05-30:
Should I delete whatever I have installed before moving on?

MartyG-RealSense, 2020-05-30:
It's a good idea to have a clean-up if possible to reduce the likelihood of a conflict with an older installation.  Again, I greatly appreciate your patience during this process.

humberthumbert, 2020-05-31:
I have successfully installed the RealSense package(librealsense-2.33.1) via CMake. 
I basically follow the [instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) except that I skip `./scripts/setup_udev_rules.sh`. I accidentally skipped that line, and encountered the same problem in [#5598](https://github.com/IntelRealSense/librealsense/issues/5598). Therefore, I manually entered the UEFI and disabled the Secure Boot.
The examples in the package work fine by far. I will move on to the ROS wrapper later.
@MartyG-RealSense Thanks for the advices.

MartyG-RealSense, 2020-05-31:
That's great news!   Thanks so much for updating us on your success and sharing your method so that others in the RealSense community can benefit from the knowledge.   :)

humberthumbert, 2020-05-31:
Well, I have tried ROS wrapper again. The result is still the same. Incomplete bulk USB transfer.
I think I will stop here anyway. Just write my own wrapper to publish the RealSense camera node in ROS : )

closed 2020-05-31 by humberthumbert

MartyG-RealSense, 2020-05-31:
Thanks again, and sorry that you had problems with the ROS wrapper.  Good luck!

swinterbotix, 2020-06-17:
Hi @MartyG-RealSense,

Not sure if I should've opened a new issue or not, but I'm having a similar issue to @humberthumbert in that I'm trying to run the SR305 Camera on Linux and am seeing a warning and error about Incomplete bulk usb transfer!

My installation procedure was:

1. Method 2 on the realsense-ros Github page (specifically installing librealsense 2.33.1 and dkms version 1.3.11)

2. realsense-ros package version 2.2.13 from source

This is all happening on Ubuntu 16.04.6 ROS Kinetic kernel 4.15.0-106-generic.

The firmware version on the camera is 3.27.3.0 and its showing up as Intel RealSense SR300 (not SR305 - though I understand that both cameras are essentially the same except for the housing). 

realsense-viewer works just fine and the ros node also works just fine (I can get both point cloud and rgb data). I'm just wondering why I'm getting the 'Incomplete bulk usb transfer!' message during startup. As an FYI, I only get the message once.

humberthumbert, 2020-06-17:
Same situation here. But since it did not affect the image transfer, I chose to ignore that message.

MartyG-RealSense, 2020-06-17:
@swinterbotix   @humberthumbert   Yes, if a 'warning' or' info' message appears only at launch, or at a rate of about one per minute if it is repeating, then it should be safe to continue if a topic is being correctly published.