realsenseai/realsense-ros #1220: fatal:you need to specify a tag name

Jenny0420, 2020-06-01:
raspberry 4 4G
ubuntu 18.04 LTS
kernel: 5.3.0-1026-raspi2
I need run my relasense d435 on ros melodic
I have download the librealsense v2.33.1
when I run " ./scripts/patch-realsense-ubuntu-lts.sh"
as this picture
[image]

Can anyone please give me some suggestion on how to solve the problem ?

Thank you in advance.

MartyG-RealSense, 2020-06-01:
In case Librealsense is having a conflict with your kernel, I would recommend instead installing librealsense through the RSUSB "backend" method, which requires an internet connection.  It is not dependent on Linux versions, kernel versions or patches, so you do not need to apply the patch script.

1.  Obtain the Librealsense source code.  If you have tried already to install SDK 2.33, you probably already have the unzipped Librealsense folder.

If you do not have it, you can visit the link below, go to the 'Assets' list at the bottom of its page and download the source-code zip file.  Then un-zip it.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1

[image]

2.  In the unzipped Librealsense folder, go to the **Build** directory (Librealsense > Build).

If you do not have a Build folder, use the command below whilst in the Librealsense root folder to create one and navigate to it:

**mkdir build && cd build**

3.  Run the CMake command below whilst in the Build directory to install Librealsense:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

Jenny0420, 2020-06-03:
@MartyG-RealSense
thanks very much,but I have meat a new error
I have already ran "realsense-viewer" successfully
but I need use  realsense with ros
[https://github.com/IntelRealSense/realsense-ros](url)
when I run "roslaunch realsense2_camera rs_camera.launch"
I have got a new error
[error_launch.txt](https://github.com/IntelRealSense/realsense-ros/files/4721862/error_launch.txt)
In this txt,you can see 
" 03/06 06:36:24,315 WARNING [281472082633088] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 12, 
[ WARN] [1591166184.355585657]: Hardware Notification:USB REC overflow,1.59117e+12,Error,Hardware Error"

how can I do.And thanks for adviance

MartyG-RealSense, 2020-06-03:
The **USB REC overflow** error suggests that more data is being generated than the USB can handle.  

You could try adding **enable_infra2:=false** to your roslaunch statement to disable infrared2 and free up USB bandwidth.  Doronhi the RealSense ROS wrapper developer has said "It will have no effect on the depth quality.  It only disable the infra2 images' transmission via the USB port.  They are still taken and the depth is calculated using them inside the device".

Jenny0420, 2020-06-03:
I have already tried your suggesstion ,but it didn't work
[image]

 I can't get the picture in RViz,but I can get the picture when I run "realsense-viewer"

MartyG-RealSense, 2020-06-03:
How often do the **control_transfer returned** messages appear?   Once a minute, or more often?

Jenny0420, 2020-06-03:
@MartyG-RealSense 
thanks for your suggesstion.
I've tried your method, but it doesn't work
[image]
I can't get the picture in RViz,but i can get the picture when i run 'realsense-viewer'

doronhi, 2020-06-03:
@Jenny0420 , Are you familiar with RViz? To see the picture you need to do the following on RViz:
Add->"By Topic" tab -> /camera/color/image_raw/Image

Jenny0420, 2020-06-03:
@doronhi 
sorry
I have chosed Add->By display ->pointcloud(and pointcloud2)
but nothing happend
I should choose "By Topic"
And i find that in depth->image_rect_raw->(camera or depthcloud) the refreash time of camera is very long almost 1.5s
by raspberrypi 4 4G

Jenny0420, 2020-06-03:
with rapberrypi4 4gMem
when i only turn on the RGB camera
4CPU used almost 90%
4GMem used 1.48G
[image]
but when i turn on the depthcloud 
4CPU used almost 95%
4GMem used 3.6G
[image]
and the RViz will be broken

MartyG-RealSense, 2020-06-18:
@Jenny0420  Do you still require assistance with this case please or can it now be closed?  Thanks!

closed 2020-06-19 by Jenny0420

MartyG-RealSense, 2020-06-19:
Thanks so much for the update!

JHPHELAN, 2022-01-23:
@doronhi @MartyG-RealSense 
Has the "tag" issue been resolved for ARM devices e.g. Raspberry Pi and Jetson Nano?
Having to "backend" with RSUSB instead of V4L causes severe performance degradation as referenced here:
[https://github.com/IntelRealSense/realsense-ros/issues/2068#issuecomment-924590313](url)
With ROS2 being the future and the ARM platforms so popular, it seems a straightforward "sudo apt install ..." should be a priority project.  Am I wrong?  I last tried on LibRealSense v2.50.
Appreciate, as always, your prompt and thoughtful responses!
JHP

MartyG-RealSense, 2022-01-24:
My understanding is that if libralsense is built from packages on Jetson instead of source code - as referenced by Doronhi in https://github.com/IntelRealSense/realsense-ros/issues/1110#issuecomment-596517559 - then the build will also be based on the RSUSB backend.  So building from the dedicated Jetson packages instead of source code may not be a solution if RSUSB gives you performance degradation.

The Jetson installation instructions do though offer the option of a non-RSUSB build from source code with the **V4L Native Backend**, which has a dedicated patch script **patch-realsense-ubuntu-L4T.sh** instead of the 'patch-realsense-ubuntu-lts.sh' one used on x86 / x64 non-RSUSB source code builds. 

In regard to the Doronhi comment that you linked to at https://github.com/IntelRealSense/realsense-ros/issues/2068#issuecomment-924590313 - Intel made updates to the RSUSB backend in librealsense 2.50.0.  I saw a recent report though from a RealSense user that said that they had RSUSB problems with 2.50.0 that disappeared if they used the previous SDK version 2.49.0.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend