realsenseai/realsense-ros #1221: A few depth image problems of D430  camera module

dingbaiyin, 2020-06-01:
My company uses the D430 camera module. The camera is fixed on the aluminum metal frame by manual installation. As shown in picture 1, we encountered several image problems during the use of D430
1. In automatic exposure mode, when the light is weak at night, there will be small black spots on the edge of the target object, as shown in Figure 2
2. The depth image is ok, but the 3D point cloud image will have pits, as shown in Figures 3 and 4
3. After using the calibrated depth camera for a period of time, the acquired depth image has a lot of overall deviation in the Z-axis direction, and can be restored after re-calibration
What are the causes of these 3 problems and how to avoid them?
4. What are the areas of D430 installation that need improvement and attention, I hope you can give some suggestions

[image]
[image]
[image]

[image]

dingbaiyin, 2020-06-01:
this is figure 3，that is the depth map with the most 3D point cloud image above
[image]

MartyG-RealSense, 2020-06-01:
1.  You could try using the "High Density" visual preset, as it prioritises a high 'fill rate' and so can fill more holes.  Using "High Accuracy" preset will increase the risk of holes occurring.  "Medium Density" provides a balance between accuracy and fill-rate.

You could also try increasing the "Laser Power" value.  This may result in increased camera operating temperature but can reduce sparseness in the depth image

Another option would be to apply a Hole Filling post-processing filter to the depth image.

2.  The black areas in Figure 3 look like the kind of holes that a Hole Filling filter would help to close up.

3.  Assuming that there is not severe vibration passing through the mounting point, and the camera is unable to be physically knocked by a human due to its mount position, a mis-calibration of the camera after a period of time might occur if the D430 module is exposed to high temperature.

4.  If there is excessive temperature and it is generated by the camera board and not by an external temperature source, adding an improved 'heat sink' component to the design to draw heat away from the board more effectively may reduce the likelihood of a temperature-caused mis-calibration.

Page 83 of the current edition of the data sheet document for the 400 Series cameras provides design guidance about a heat sink.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

Also, it is worth mentioning that care should be taken when handling the depth module during installation, as mis-calibration can occur if the module is physically flexed when being handled, requiring a re-calibration to correct it.

Regarding the flex cable that connects the Vision Processor D4 board and the depth module: there is also a rigid inter-connector board that is available.  As far as I know it does not offer a performance advantage over cable, but I mention it in case you want to incorporate it into your product design.

https://www.framos.com/en/d400-rigid-50-pin-interposer-20737

dingbaiyin, 2020-06-02:
Thank you ，I read the  intel-realsense-d400-series-product-family-datasheet .This is very useful for us.
I find that ASIC Temperature is higher than 42 degrees most of the time and it is difficult to drop below 42 degrees Celsius
I still have a few questions:
1. When we use a Hole Filling post-processing filter to the depth image, the speckle of the depth image is obviously reduced, so will the image effect of the corresponding 3D point cloud be improved or not?
2. From the depth image, the image of the pig is quite complete, but from the 3D point cloud image, the back of the pig is not flat and smooth, and even a large depression appears as shown in the picture. What causes this and how to avoid it. I'm sorry this question I mentioned earlier but still don't understand
depth image:
[image]
point cloud image :
[image]
this is two imgae 
The point cloud image does not look very clear. This is the Baidu cloud link address of two pictures, which can be downloaded and viewed

MartyG-RealSense, 2020-06-02:
1.  Yes, the Hole-Filling Filter works in 3D point cloud mode as well as 2D mode.

2.  It looks as though there is a lot of waviness in the point cloud image.The links below on reducing waviness and "ghost noise" may be a useful reference.

https://github.com/IntelRealSense/librealsense/issues/4553

https://github.com/IntelRealSense/librealsense/issues/1375

dingbaiyin, 2020-06-02:
Great, I will try.Thank you !

MartyG-RealSense, 2020-06-18:
Hi @dingbaiyin  do you still require assistance with this case please, or can it be closed?   Thanks!

MartyG-RealSense, 2020-06-25:
Case closed due to no further comments received.

closed 2020-06-25 by MartyG-RealSense

bestaps, 2021-02-08:
@dingbaiyin where did you get the cable from? we are looking for suppliers for the same.

MartyG-RealSense, 2021-02-08:
Hi @bestaps   If you are looking for a pre-made **interposer cable** to connect the Vision Processor D4 to the Depth Module board, or the rigid board connector version, the Intel authorized company FRAMOS can supply them:

**FRAMOS**

**Cable**
Different lengths can be selected using the **Cable length (m)** drop-down menu.

https://www.framos.com/en/hsf50-22273

**Board**
https://www.framos.com/en/d400-rigid-50-pin-interposer-20737

Other suppliers for the rigid board can be found by googoling for **intel interposer 956827**