realsenseai/realsense-ros #1222: Need help; Launch file modify?

Andyshen555, 2020-06-01:
I have installed the ros wrapper from command line on Ubuntu18.04 and ROS M, how and where should I modify the launch files?

MartyG-RealSense, 2020-06-01:
Hi @Andyshen555   You can create or modify launch files using a **text editor** program of your own preference.  

Addmitedly I am not sure where the RealSense ROS launch files are stored though.  I would speculate that they are in the vicinity of the catkin workspace - catkin_ws/src/ - where the RealSense ROS wrapper is installed to.

Andyshen555, 2020-06-01:
I used to build the wrapper from source, so I know how to modify those launch files. But now I used the sudo apt get to intall realsense2, so where do I have access to the launch files? Theres nothing in the src now. Do I need to download the source code and build from that to change launchfile?

Andyshen555, 2020-06-02:
Found it, it's in the default folder and can be found using dpkg -L

closed 2020-06-02 by Andyshen555

MartyG-RealSense, 2020-06-02:
Glad you found it - thanks for the update and details of how you found it!

mintnguyenn, 2022-07-06:
hi @Andyshen555 can you show me how to find the launch file? Many thanks

Andyshen555, 2022-07-06:
It's in /etc/ros/ros_ver/share/realsense2/launch for Linux by default