realsenseai/realsense-ros #1223: Starting and stopping streaming D435 streaming

nawara72, 2020-06-02:
Hi

Is there a way to stop and start the D435 streaming after running  
roslaunch realsense2_camera rs_camera.launch.

Is stopping the publishing of topics the same as stopping streaming?
 I apologize if this issue is the same as https://github.com/IntelRealSense/realsense-ros/issues/1195.
but wasn't sure if that issue was just about stopping publishing or about stopping streaming too.

I couldn't find a way to do that, so I then added a stop_streaming function in base_realsense_node.cpp

void BaseRealSenseNode::stopStreaming()  
{
	for (auto it = _sensors.begin(); it != _sensors.end(); it++)
	{
		auto& sens = _sensors[it->first];
		try
		{
			sens.stop();
			ROS_INFO_STREAM("RealSense Stopped Streaming!");
		}
		catch (const rs2::wrong_api_call_sequence_error & ex)
		{
			ROS_DEBUG_STREAM("toggleSensors: " << ex.what());
		}
	}
}

Then instead of publishing by the roslaunch command above i did the following in my code
        ros::NodeHandle nh;
	ros::NodeHandle nh_private;
	nh_private = ros::NodeHandle("~");

	rs2::context ctx;
	auto list = ctx.query_devices(); // Get a snapshot of currently connected devices
	if (list.size() == 0)
		throw std::runtime_error("No device detected. Is it plugged in?");
	rs2::device dev = list.front();

	realsense2_camera::BaseRealSenseNode brs(nh,
		nh_private, dev, "817612070719");

       brs.publishTopics();
	int count = 0;
	for (int count = 0; count < 50000; ++count) {
		if (count == 200000)
			brs.stopStreaming();
		if (count == 40000)
			brs.startStreaming();
}
However this seems to crash my program nearly every time I call the stop_streaming function.

I would appreciate any advice/help in this matter.

Thanks
Nawar

MartyG-RealSense, 2020-06-02:
Hi @nawara72   As far as I know, the proper way to stop and start streaming in ROS is to unsubscribe to stop and resubscribe to start again.

nawara72, 2020-06-02:
Hi Marty 

Thanks for your quick reply but not sure how stopping the subscription will stop the publishing and stop the streaming?

Are you sure of that? I am very new to ROS but I thought publishing is independent of the subscription/listening?
Thanks

MartyG-RealSense, 2020-06-02:
You may have a point.  My apologies, it's late in my time zone!  Maybe stop data from the **publisher** then without destroying it.  That goes back to my mention in https://github.com/IntelRealSense/realsense-ros/issues/1195 of the method of using **Mux** to have two publishers (one sending, the other not) that you toggle between to stop and start data flow without continually destroying and recreating the publisher.

https://github.com/IntelRealSense/realsense-ros/issues/1195#issuecomment-626750400

nawara72, 2020-06-02:
I saw that option, but I am more interested in D435 stopping streaming cause planning to run for many days and wanting to make sure it doesn't heat up :-)

MartyG-RealSense, 2020-06-02:
The 400 Series cameras are designed to run indefinitely as long as they remain within temperature tolerances.  As long as they do, problems that occur in a long-run are likely because of environmental factors such as lighting conditions, or non-camera factors such as glitches in the USB port / cable, computer hardware or the computer's OS.

Further advice about temperature and the camera's built-in safety mechanism can be found here:

https://github.com/IntelRealSense/librealsense/issues/5209#issuecomment-563358823

In the May 2020 edition of the data sheet document for the 400 Series cameras, temperature guidance was changed.   It now recommends a maximum **outer casing** temperature of 50 degrees during operation, whereas in previous editions it recommended a maximum **internal operating temperature** of 35 degrees C.

MartyG-RealSense, 2020-06-18:
@nawara72  Do you require further assistance with this case please, or can it be closed?   Thanks!

nawara72, 2020-06-19:
Closing it, thanks for all the help

closed 2020-06-19 by nawara72