realsenseai/realsense-ros #1224: D435i IMU problems?

Andyshen555, 2020-06-02:
Those librealsense guys cannot solve, so I come to realsense-ros to try my luck.

When I start whatever node trying to use IMU (eg. opensource_tracking.launch), it gives out:
`[ WARN] [1591132803.683814185]: Warning: Some linear velocity entries in parameter imu0_config are listed true, but an sensor_msgs/Imu contains no information about linear velocities`
`[ WARN] [1591132809.978068997]: Could not obtain transform from camera_imu_optical_frame to camera_link. Error was "camera_link" passed to lookupTransform argument target_frame does not exist. `
`[ WARN] [1591132810.196124894]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds!`

doronhi, 2020-06-03:
Could you post the realsense2_camera's log? If there are any issues they may appear there.
Also, could please specify machine and OS you are using.
Or, if you have an issue open with those inputs could you add a link?

MartyG-RealSense, 2020-06-03:
Hi @doronhi, thanks for your help!  The case of @Andyshen555 starts part way through another RealSense user's case, so I have linked below to the comment where their section of the discussion starts.

https://github.com/IntelRealSense/librealsense/issues/6008#issuecomment-637265899

It is a very long case.  The short version of it was that @Andyshen555 seemed to resolve the problems with bad IMU values but still had errors being generated in ROS, as described in this link:

https://github.com/IntelRealSense/librealsense/issues/6008#issuecomment-637816974

Andyshen555, 2020-06-03:
I am not sure what cause the problemes, since everything was working fine a day ago. Yesterday, the imu problem occurred, I thought it was the hardware failure. Through the trouble shooting, all kings of errors appeared like camera connection lost half way, cannot stream infra1, cannot stream depth, but those came ramdomly and always came with the resources temporily unavailible. However, the imu error is always there. I will post the error log  6 hours later. BTW, I'm running ubuntu 18.04 on a thinkpad T470 and D435i. I currently have a D435 and a D435i availible, if they are needed for eliminating some possible causes.

Andyshen555, 2020-06-03:
@doronhi [https://github.com/IntelRealSense/librealsense/issues/6008#issuecomment-637336965](url) This is exactly the log showing up now for opensource_tracking.launch. Also, I think I might have some other thing wrong with my system or computer. When I try rs_camera.launch disabling the RBG, I receive and only receive `03/06 09:42:39,507 WARNING [139879671510784] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61` See full log for rs_camera.launch below.

Andyshen555, 2020-06-03:
logging to /home/andy1234/.ros/log/133b2c26-a5a0-11ea-921d-d46d6d655e19/roslaunch-andyThinkpad-6868.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://andyThinkpad:37085/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 720
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 1280
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: False
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 720
 * /camera/realsense2_camera/infra_width: 1280
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
[... 63 lines cut ...]
[ INFO] [1591191754.005945027]: Setting Dynamic reconfig parameters.
[ INFO] [1591191758.961412757]: Done Setting Dynamic reconfig parameters.
[ INFO] [1591191759.005465922]: depth stream is enabled - width: 1280, height: 720, fps: 30, Format: Z16
[ INFO] [1591191759.006287925]: infra1 stream is enabled - width: 1280, height: 720, fps: 30, Format: Y8
[ INFO] [1591191759.007011818]: infra2 stream is enabled - width: 1280, height: 720, fps: 30, Format: Y8
[ INFO] [1591191759.007054367]: setupPublishers...
[ INFO] [1591191759.010440154]: Expected frequency for depth = 30.00000
[ INFO] [1591191759.051943439]: Expected frequency for infra1 = 30.00000
[ INFO] [1591191759.077006096]: Expected frequency for infra2 = 30.00000
[ INFO] [1591191759.105237897]: setupStreams...
[ INFO] [1591191759.156399792]: insert Depth to Stereo Module
[ INFO] [1591191759.156555432]: insert Infrared to Stereo Module
[ INFO] [1591191759.156623601]: insert Infrared to Stereo Module
[ INFO] [1591191759.352641376]: SELECTED BASE:Depth, 0
[ INFO] [1591191759.378513274]: RealSense Node Is Up!
 03/06 09:42:39,507 WARNING [139879671510784] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/06 09:42:39,708 WARNING [139879671510784] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/06 09:42:39,889 WARNING [139879671510784] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

closed 2020-06-25 by MartyG-RealSense