realsenseai/realsense-ros #1226: Unstable inclusion of two cameras Realsense D435 and D435i on NUC

amburkoff, 2020-06-03:
Computer: NUC7I7BNKQ 7567U
D435 firmware: 05.12.05
D435 firmware: 05.12.05
Original usb cables realsense inserted into two ports type-A in NUC directly without an additional usb hub.
Ubuntu 18.04
LibRealSense v2.33.1
RealSense ROS v2.35.0
Autosuspend off

Two cameras start through the realsense-viewer, but do not start through realsense-ros:
https://drive.google.com/file/d/1ZbFjwMvPRBPl20jS1e1LB7Te0ponmE3k/view?usp=sharing

Terminal log:
[list_2.txt](https://github.com/IntelRealSense/realsense-ros/files/4723647/list_2.txt)
[test_2.txt](https://github.com/IntelRealSense/realsense-ros/files/4723648/test_2.txt)

MartyG-RealSense, 2020-06-03:
Hi @amburkoff   Have you seen Intel's guide to setting up two D435 cameras in a multi-camera arrangement in ROS, please?

https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

amburkoff, 2020-06-09:
Computer: NUC8i7BEKQA2 8559U
D435 firmware: 05.12.05
D435 firmware: 05.12.05
Original usb cables realsense inserted into two ports type-A in NUC directly without an additional usb hub.
Ubuntu 18.04
LibRealSense v2.33.1
RealSense ROS v2.2.13

Run through rs_multiple_devices.launch with explicit camera id.

Log files:
https://drive.google.com/drive/folders/1DZAl5mJTWlz4PRkT5dTUgJFvc8KFAk_A

Tests are numbered in order.

The list.txt file contains the result of displaying the status of your devices before launching the launch file.

The test.txt file contains the result of displaying of output to the rs_multiple_devices.launch with explicit camera id.

Test Results Table:
https://docs.google.com/spreadsheets/d/1o2Mcmai09DVle932Xk3pWytPIPFHgxRilik4Ucp-s-I/edit?usp=sharing

MartyG-RealSense, 2020-06-09:
Hi @amburkoff   Thanks very much for sharing your carefully documented test results!  

If you are launching the cameras with **rs_multiple_devices.launch** and are using serial numbers to identify the two separate cameras, then an alternative approach in ROS that may give more stable device identification may be to identify by USB port ID instead of serial number with the **usb_port_id** instruction.

https://github.com/IntelRealSense/realsense-ros#launch-parameters

RealSenseSupport, 2021-01-19:
Closing this case for now. Please feel free to let us know if the problem persists using the latest release of SDK/ROS.
Thank you.

closed 2021-01-19 by RealSenseSupport