realsenseai/realsense-ros #1227: Unstable inclusion of one cameras Realsense D435I on Jetson Xavier

amburkoff, 2020-06-03:
Computer: NVIDIA Jetson AGX Xavier - Jetpack 4.3 [L4T 32.3.1]
D435I firmware: 05.12.03
One cables https://www.amazon.com/TetherPro-USB-C-1-8m-High-Visibility-Orange/dp/B0794BL8CH/ref=sr_1_15?dchild=1&keywords=TetherPro+1.8&qid=1590582032&sr=8-15 inserted into one ports type-c in Jetson directly without an additional usb hub.
Ubuntu 18.04
Enable jetson_clocks.sh
LibRealSense v2.33.1
RealSense ROS v2.2.13
Autosuspend off

Realsense stopped working and after restarting did not work, the log was saved only after restarting:
[list_1.txt](https://github.com/IntelRealSense/realsense-ros/files/4724049/list_1.txt)
[test_1.txt](https://github.com/IntelRealSense/realsense-ros/files/4724050/test_1.txt)

MartyG-RealSense, 2020-06-03:
Hi @amburkoff   After the camera stopped working, have you tried a "hard reboot" please (shutting the computer down completely instead of restarting, and then booting with a press of the the power button)?

MartyG-RealSense, 2020-06-25:
Hi @amburkoff   Do you require further assistane with this case please?  Thanks!

RealSenseSupport, 2021-01-19:
Closing this case for now. Please feel free to let us know if the problem persists using the latest release of SDK/ROS.
Thank you.

closed 2021-01-19 by RealSenseSupport