realsenseai/realsense-ros #1228: RealSenseNodeFactory segfault during launch

truhoang-ubtrobot, 2020-06-03:
Per @doronhi request, I'm creating a new issue for this error. Originally posted in #1191.

truhoang-ubtrobot, 2020-06-03:
Downgraded `librealsense` to `v2.33.1` and relaunch shows no segfault error. But new firmware installed using `realsense-viewer` (`05.12.05.00`) shows the following error:

```shell
[ INFO] [1591148280.396089253]:  
 02/06 18:38:00,406 ERROR [140402921432832] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 02/06 18:38:00,427 WARNING [140403035809536] (rs.cpp:283) null pointer passed for argument "device"
[ WARN] [1591148280.427877974]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1591148280.427915510]: The requested device with device name containing d435 is NOT found. Will Try again.
```

On `realsense-viewer` the camera is shown connected to a USB 3.2 port

truhoang-ubtrobot, 2020-06-03:
> Downgraded `librealsense` to `v2.33.1` and relaunch shows no segfault error. But new firmware installed using `realsense-viewer` (`05.12.05.00`) shows the following error:
>
> ```shell
> [ INFO] [...]

Downgraded the firmware to `5.12.3` and the above error goes away

chaitudarz, 2020-07-01:
hello @truhoang-ubtrobot could you tell me how to downgrade the firmware to 5.12.3, as I have the same error [https://github.com/IntelRealSense/realsense-ros/issues/1228#issuecomment-638309946](url).

I have used the method 1 from this page [https://github.com/IntelRealSense/realsense-ros](url). 

Regards,
Chaitanya

truhoang-ubtrobot, 2020-07-01:
@chaitudarz 
1. You can download the 5.12.3 firmware file from here: https://dev.intelrealsense.com/docs/firmware-releases
2. Plug your realsense camera into your PC
3. Open `realsense-viewer`
4. Click on `More`->`Update Firmware` and select the downloaded `.bin` file
5. Follow the prompts

RealSenseSupport, 2020-07-07:
Hi,

Are you still experiencing this? We are unable to reproduce this issue with the latest F/W and RealSenseViewer.

Thanks

chaitudarz, 2020-07-08:
@truhoang-ubtrobot, 

Thanks for the reply. Now it's working totally fine. :)

@RealSenseSupport 

Nope, Now I am not facing any more issues. 

Thank you :)

truhoang-ubtrobot, 2020-07-10:
@RealSenseSupport 
New `realsense-ros` release (2.2.14) fixes the issue mentioned.

truhoang-ubtrobot, 2020-07-10:
Previous comment refers to L515. Tested with D435i and the same issue occurs

RealSenseSupport, 2020-09-01:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

truhoang-ubtrobot, 2020-09-01:
Hi yes, I still have the same issue

RealSenseSupport, 2020-09-03:
This is occurring on the latest version of RealSenseROS+LibRealSense?
And could you please specify which hardware you are attaching the camera to... NUC, JETSON, RasPi, etc?
And the exact command line you are using to launch and the full screen output leading up to the error.
Much Thanks

RealSenseSupport, 2020-09-24:
@truhoang-ubtrobot Seems your problem is similar to https://github.com/IntelRealSense/realsense-ros/issues/1408#issuecomment-698128999 Could you please check if you have 99-realsense-libusb.rules under /etc/udev/rules.d/?

closed 2021-09-11 by MartyG-RealSense